realsenseai/librealsense #13821: D435i MUST sudo realsense-viewer or No Frames Received on RGB Camera except YUYV

sunflowerhcy, 2025-03-08:
- Module: NVIDIA Jetson AGX Xavier
- JetPack: 5.1.4
- RealSense: D435i
- RealSense SDK: 2.55.1
- Firmware: 5.16.0.1

Issue:

- sudo realsense-viewer, everything work fine.
- realsense-viewer, RGB Camera echo No Frames Received **except YUYV**.

Workaround:

- Reinstall RealSense SDK and flash Firmware.
- Check eudv rules.
- Check similar Issues and try everything....

Can anyone help?

MartyG-RealSense, 2025-03-08:
Hi @sunflowerhcy  Which method did you use to install the SDK on your Jetson, please?

happydrones, 2025-03-08:
Hello, I also encountered an issue related to permission. This is my issue (#13812). I tried to launch the realsense-viewer without sudo privileges, and it was able to retrieve the depth frame and RGB frame.

sunflowerhcy, 2025-03-09:
> Hi [@sunflowerhcy](https://github.com/sunflowerhcy) Which method did you use to install the SDK on your Jetson, please?

Install SDK by build the source code,.

sunflowerhcy, 2025-03-09:
> Hello, I also encountered an issue related to permission. This is my issue ([#13812](https://github.com/IntelRealSense/librealsense/issues/13812)). I tried to launch the realsense-viewer without sudo privileges, and it was able to retrieve the depth frame and RGB frame.

I followed your post [#13812], but I still got the issue. Did you fix the problem? If yes, pls provide more info, thx.

sunflowerhcy, 2025-03-10:
> > I followed your post [https://github.com/[/issues/13812](https://github.com/IntelRealSense/librealsense/issues/13812)], but I still got the issue. Did you fix the problem? If yes, pls provide more info, thx.
> 
> I'm also a beginner in using the D435i for a project, but I can provide the full steps I have followed below.
> 
> > * guidence
> >   Clone the Librealsense repository:
> >   `git clone https://github.com/IntelRealSense/librealsense.git`
> >   Navigate to the cloned directory and create a build directory:
> > 
> > ```
> > cd librealsense
> > mkdir build
> > cd build
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Run CMake with the specified options:
> > cmake .. -DBUILD_EXAMPLES=true \
> > -DBUILD_PYTHON_BINDINGS=bool:true\
> > -DPYTHON_EXECUTABLE=/usr/bin/python3.10 {the_path_of_your_python_interpreter}\
> > -DBUILD_WITH_CUDA=true \
> > -DCUDA_ARCHITECTURES=87 {for_agx_orin}\
> > -DCHECK_FOR_UPDATES=false
> >     
> >   
> > Compile the project using make:
> > **`make clean && make && sudo make install`**

I made my install script as follow:

function realsense_sdk_build {
    cd ${FOLDER}
    cd librealsense-${SDKREALSENSEVERSION}

    echo "**  Build SDK"
    ./scripts/setup_udev_rules.sh
    echo -e "${sudopasswd}" | sudo -S udevadm control --reload-rules
    echo -e "${sudopasswd}" | sudo -S udevadm trigger
    rm -rf build
    mkdir build
    cd build
    **cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3) -DCHECK_FOR_UPDATES=false**
    echo -e "${sudopasswd}" | sudo -S make uninstall
    make clean
    make -j$(nproc)
    echo -e "${sudopasswd}" | sudo -S make install
}

Just miss -DCUDA_ARCHITECTURES, I will rebuild SDK later and reply.
Thank u for ur help.

closed 2025-03-31 by MartyG-RealSense