realsenseai/librealsense #13822: Intel Realsense D405 still not detected on WSL2 (Ubuntu 22.04.5 LTS) even after attaching USB

msiauw2, 2025-03-09:
I've connected an Intel Realsense D405 to my Windows laptop and I attached the USB device to my Ubuntu 22.04.5 WSL terminal (it is successfully connected when I type lsusb).

However, even after this, when I try to open realsense-viewer, my device is not detected. 

When I type rs-enumerate-devices, I get "No device detected. Is it plugged in?"

When I try to open the RealSense viewer, I get this:

 09/03 12:13:35,393 INFO [140095690869952] (backend-v4l2.cpp:646) Cannot access /sys/class/video4linux
 09/03 12:13:36,419 INFO [140095690869952] (backend-v4l2.cpp:646) Cannot access /sys/class/video4linux
 09/03 12:13:36,420 INFO [140095690869952] (context.cpp:128) Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)

Any help would be greatly appreciated. I'm new to all of this, so please let me know if more information about my system is needed to adequately help me. Thank you.

MartyG-RealSense, 2025-03-10:
Hi @msiauw2  Is the D405 camera detected if you launch programs in **sudo** admin permissions mode with `sudo realsense-viewer` or `sudo rs-enumerate-devices`

MartyG-RealSense, 2025-03-10:
Did you install the librealsense SDK from source code or from DKMS packages on WLS2, please?  WLS2 users usually have more success with a source code build of the SDK, as demonstrated at  https://github.com/IntelRealSense/librealsense/issues/12810

msiauw2, 2025-03-10:
@MartyG-RealSense I have been following the installation instructions from https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

When I tried installing from source code using the libuvc build script as you explained in #12810, I get this public key-related error:

 The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
Get:10 http://security.ubuntu.com/ubuntu jammy-security/universe Translation-en [207 kB]
Ign:3 http://your.repository.url focal InRelease
Err:3 http://your.repository.url focal InRelease
  Could not resolve 'your.repository.url'
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-/dists/repo/InRelease  403  Forbidden [IP: 13.33.4.78 443]
E: The repository 'https://librealsense.intel.com/Debian/apt- repo InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: https://librealsense.intel.com/Debian/apt-repo jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
E: The repository 'https://librealsense.intel.com/Debian/apt-repo jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MartyG-RealSense, 2025-03-10:
A RealSense user at https://github.com/IntelRealSense/librealsense/issues/13736 describes how they succeeded with a libuvc source code installation of librealsense on WSL2 after attempts at a package installation and their initial try to use libuv failed for them.

msiauw2, 2025-03-10:
Hello, I'm still not able to do the installation detailed at #13736 because of the NO_PUBKEY issue I posted above. Do you know how to resolve this specific issue? For reference, here it is again:

The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
Get:10 http://security.ubuntu.com/ubuntu jammy-security/universe Translation-en [207 kB]
Ign:3 http://your.repository.url/ focal InRelease
Err:3 http://your.repository.url/ focal InRelease
Could not resolve 'your.repository.url'
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-/dists/repo/InRelease 403 Forbidden [IP: 13.33.4.78 443]
E: The repository 'https://librealsense.intel.com/Debian/apt- repo InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: https://librealsense.intel.com/Debian/apt-repo jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
E: The repository 'https://librealsense.intel.com/Debian/apt-repo jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MartyG-RealSense, 2025-03-11:
Are you able to use realsense-viewer on Windows directly (without using WSL2) if you download and run the Windows version of realsense-viewer?  You can do so with the following procedure.

1.  Visit the SDK 'Releases' page, scroll down to the 'Assets' file list at the bottom of the page and click on the link **Intel.RealSense.Viewer.exe** to launch a download of the program in your browser.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

2. Run the program once the download is complete.

msiauw2, 2025-03-11:
Hi, yep, RealSense has been working perfectly on my Windows 11 OS. The software detects the camera and everything. However for my project I need to use Linux, which is why I'm trying to get RealSense working in WSL2.

MartyG-RealSense, 2025-03-11:
There are not many other installation reference guides that I can suggest, unfortunately.  It is difficult to get the librealsense SDK working on WSL2, and if you have the option then it may be better to buy a separate cheap computing board for Ubuntu such as a Raspberry Pi 5 and install librealsense on that.

A RealSense user with Windows 11 at the link below successfully installed the SDK on WSL2, so it may be worth reading that discussion next.

https://stackoverflow.com/questions/75335203/realsense-sdk-2-0-installation-in-ubuntu

msiauw2, 2025-03-11:
Do you know of any users that have received the same NO_PUBKEY error that I did? How have they resolved this issue in the past?

msiauw2, 2025-03-12:
When I try to add the keyserver to my list of repositories as explained in that link, I get this error:

deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main
Ign:1 http://your.repository.url focal InRelease
Hit:2 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:3 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:6 http://security.ubuntu.com/ubuntu jammy-security InRelease
Err:7 https://librealsense.intel.com/Debian/apt- repo InRelease
  403  Forbidden [IP: 13.226.94.33 443]
Hit:8 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
Ign:1 http://your.repository.url focal InRelease
Ign:1 http://your.repository.url focal InRelease
Err:1 http://your.repository.url focal InRelease
  Could not resolve 'your.repository.url'
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-/dists/repo/InRelease  403  Forbidden [IP: 13.226.94.33 443]
E: The repository 'https://librealsense.intel.com/Debian/apt- repo InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MartyG-RealSense, 2025-03-13:
Going back to the beginning of this case and the camera not being detected in realsense-viewer or rs-enumerate-devices, does it make a difference if you unplug and reinsert the camera, and then restart WSL2 and try launching the programs again?

https://www.how2shout.com/how-to/how-to-reboot-wsl-windows-subsystem-linux-in-windows-10.html

msiauw2, 2025-03-13:
Yes, I've already tried unplugging and reinserting and rebooting WSL2 per those instructions

MartyG-RealSense, 2025-03-14:
Going back to the beginning of the case again and researching the **Cannot access /sys/class/video4linux** error this time, I found a case at https://github.com/IntelRealSense/librealsense/issues/13162 where a RealSense user who experienced this error was using Ubuntu on a Windows computer, but taking the approach of setting up a dual-boot (Windows or Ubuntu) instead of WSL2.  

The solution that worked in their particular case was apparently to first uninstall all librealsense packages from their computer with the command below in the Ubuntu terminal and **then** run the libuvc_installation.sh build script.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

closed 2025-03-31 by MartyG-RealSense