realsenseai/librealsense #13824: Do I need to do a distortion correction after getting aligned frames using D455?

cvki, 2025-03-10:
Hello,
 I obtained color and depth aligned frame using the D455 camera. Do I need to further perform distortion processing on the color images? I tried using OpenCV for distortion processing and remapping, and obtained internal parameters for multiple scenes. However, the internal parameters obtained for each scene have a 0.1 scale difference. Is this normal? Can you give me some advice. 
Thank you.

Here is part of my code: 

align_to = rs.stream.color
  align = rs.align(align_to)
  align_frames = align.process(frames)
  frames = align_frames
  frame_num = frames.frame_number
  depth_frame = frames.get_depth_frame()
  color_frame = frames.get_color_frame()

  # process rgb
  color_img = np.asanyarray(color_frame.get_data())
  img_bgr = cv2.cvtColor(color_img, cv2.COLOR_RGB2BGR)
  camera_matrix = np.array([
      [color_intrin.fx, 0, color_intrin.ppx],
      [0, color_intrin.fy, color_intrin.ppy],
      [0, 0, 1]
  ])
  dist_coeffs = np.array(color_intrin.coeffs)
  undistorted_color = cv2.undistort(img_bgr, camera_matrix, dist_coeffs)

  new_camera_matrix, roi = cv2.getOptimalNewCameraMatrix(
                        camera_matrix, dist_coeffs, (undistorted_color.shape[1], undistorted_color.shape[0]), alpha=0)
                    mapx, mapy = cv2.initUndistortRectifyMap(
                        camera_matrix, dist_coeffs, None, new_camera_matrix, (undistorted_color.shape[1], 
  undistorted_color.shape[0]), cv2.CV_32FC1)

  color_img = cv2.remap(undistorted_color, mapx_g, mapy_g, cv2.INTER_LINEAR)

 '''then I will save <color_img>'''

MartyG-RealSense, 2025-03-10:
Hi @cvki  The RealSense camera model's hardware applies a Brown-Conrady distortion model at the point of capture before the data is sent through the USB cable to the computer.  It is not necessary to perform further processing on the image's distortion.

Most RealSense stream formats have distortion applied to them, except for **Y16** because it is used for camera calibration.  

A small number of RealSense users prefer distortion not to be applied and so may use OpenCV's **undistort** function like you did to remove the distortion.

The intrinsic values will vary depending on which resolution is being used, as demonstrated in the image below.  The default color resolution is 1280x720.

[image]

The intrinsic values for each individual camera are unique due to the manufacturing process at the factory.  You can find the intrinsics for your own camera at each resolution by launching the **rs-enumerate-devices** tool in calibration information mode with the command below:

`rs-enumerate-devices -c`

In regard to scale: as you are using rs-align, depth data must be involved in your script.  The default depth scale of the camera is **0.001**.  When RealSense depth data is imported into another tool, the image may be mis-scaled if the tool that the data is imported into is not using the same scale as the original RealSense data.  

It is possible to reconfigure the scale of RealSense depth data to another value, such as 0.1 or 0.001.  https://github.com/IntelRealSense/librealsense/issues/10976#issuecomment-1271236193 has information about changing the depth scale with pyrealsense2 code.

cvki, 2025-03-11:
@MartyG-RealSense Thank you for your careful answer, it has been very helpful to me.

MartyG-RealSense, 2025-03-11:
The intrinsic values that are output can vary between retrieval methods.  If in doubt about which set of values is correct then I recommend relying on the ones from `rs-enumerate-devices -c`

Abrahamh08, 2025-03-23:
Are you sure that's right? deproject_pixel_to_point undistorts before projecting, so I've been using the color stream with the assumption that no correction is applied when I have the image.

MartyG-RealSense, 2025-03-23:
Hi @Abrahamh08  Line 49 of the SDK's rs-utils source code file states that the Inverse Brown-Conrady distortion model (which the color stream uses) undistorts the image instead of distorts it.

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/include/librealsense2/h/rs_types.h#L49

The PR at https://github.com/IntelRealSense/librealsense/pull/13833 that you commented on will be the best place to get advice on this subject directly from an Intel RealSense development team member (OhadMeir) though.

Abrahamh08, 2025-03-25:
My question is the same now as the question in this issue. The original poster asked if it's necessary to apply distortion correction to have an undistorted image. I am confused by what you were saying about the "camera model's hardware applies a Brown-Conrady distortion model at the point of capture before the data is sent through the USB cable to the computer", because I believe the image is going to be distorted, and needs to be undistorted with the inverse-brown conrady model like you were saying for points of interest, or a distorted->undistorted pixel map which is inefficient since it is inverse. In my use case, however, I am interested in undistorted->distorted, so it is good.

Abrahamh08, 2025-03-27:
That makes sense. The Left IR/depth distortion coefficients (didn't check Right IR) are zero. So it makes sense it is rectified beforehand.

MartyG-RealSense, 2025-04-10:
Does anyone require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2025-04-17:
Case closed due to no further comments received.

Zaid-Waghoo, 2025-08-14:
Hello, 

I was looking to get clarification on the same. I did this experiment and these were my findings:

## Title
D455 color `inverse_brown_conrady`: parity with OpenCV `undistortPoints` and vectorized deprojection path

## Setup
- 3× D455, depth aligned to color.
- `intr_for_depth == intr_color`, model = `inverse_brown_conrady`, coeffs `[k1,k2,p1,p2,k3]` from metadata.
- Tested deprojection at ArUco marker centers (Z ≈ 1.06 m).

## Methods tested
1. `rs2_deproject_pixel_to_point(intr_for_depth, [u,v], Z)`
2. Pinhole model with `dist=0` (baseline)
3. `cv2.undistortPoints([(u,v)], K, dist)` → normalized ray → × Z

## Results

| id | u,v (px)       | r (px) | depth (m) | Δ (p0 vs RS) ‖mm‖ | Δ (CV undist vs RS) ‖mm‖ | dX (mm) | dY (mm) | dZ (mm) |
|----|----------------|--------|-----------|-------------------|-------------------------|---------|---------|---------|
| 6  | (633.8, 287.2) | 94.0   | 1.0670    | 0.12              | 0.00                    | 0.00    | 0.00    | 0.00    |
| 9  | (637.0, 112.6) | 268.1  | 1.0610    | 2.97              | 0.00                    | 0.00    | 0.00    | -0.00   |
| 7  | (938.9, 350.3) | 295.7  | 1.0565    | 4.08              | 0.00                    | -0.00   | 0.00    | -0.00   |
| 0  | (333.1, 299.9) | 322.0  | 1.0640    | 5.51              | 0.00                    | -0.00   | -0.00   | 0.00    |

**Summary ‖Δ3D‖ (mm):**
- **Pinhole(dist=0) vs RS** : mean=3.17, median=3.53, max=5.51, N=4  
- **OpenCV undist vs RS**   : mean=0.00, median=0.00, max=0.00, N=4  

## Minimal vectorized OpenCV path (matches RS exactly)
```python
K = np.array([[intr.fx, 0, intr.ppx],
              [0, intr.fy, intr.ppy],
              [0,       0,       1]], np.float32)
dist = np.array(intr.coeffs[:5], np.float32)  # [k1,k2,p1,p2,k3]

# uv: Nx2 pixels, depth_u16 aligned to color, depth_scale in meters/unit
uv = uv.astype(np.float32).reshape(-1,1,2)
rays = cv2.undistortPoints(uv, K, dist, P=None).reshape(-1,2)  # normalized rays
Z = depth_u16[v_inds, u_inds].astype(np.float32) * depth_scale
XYZ = np.column_stack([rays[:,0]*Z, rays[:,1]*Z, Z])
```

Questions
Is exact parity between rs2_deproject_pixel_to_point and OpenCV undistortPoints (when using the intrinsics/distortion from intr_for_depth) expected for D455 color streams labeled inverse_brown_conrady?

Is using OpenCV’s undistortPoints the recommended approach for high-throughput (vectorized) pixel→ray conversion in Python when matching RealSense deprojection behavior?

MartyG-RealSense, 2025-08-15:
Hi @Zaid-Waghoo  It is not usually necessary to perform OpenCV's undistort function on a RealSense image that has been converted to OpenCV, even if rs2_deproject_pixel_to_point() has not been used.

I only know of one other RealSense user at https://github.com/IntelRealSense/librealsense/issues/3880 who has previously performed an undistort on a RealSense frame in OpenCV.

As the name **inverse_brown_conrady** suggests, this distortion model is the inverse of the Brown-Conrady model.  The official RealSense SDK documentation about intrinsics and deprojection at the link below states that "images with Inverse Brown-Conrady Distortion are being undistorted when calling rs2_deproject_pixel_to_point()".

https://dev.realsenseai.com/docs/projection-in-intel-realsense-sdk-20#distortion-models

It sounds as though you are asking whether the undistortion performed when using inverse_brown_conrady and rs2_deproject_pixel_to_point() provides the same results as performing the OpenCV undistort function on a RealSense image.  Is that correct, please?

If that is correct then I honestly do not know if there is exact parity, but as long as you are using the color intrinsics for the aligned image - which your `intr_for_depth == intr_color` instruction seems to make sure of - then there should not be a cause for concern.

MartyG-RealSense, 2025-08-19:
Deprojecting thousands of pixels and performing depth-color alignment are operations that will both be processing-intensive on the computer's CPU.  So you will want to reduce processing wherever you can.

 If you use a workstation that has an Nvidia graphics GPU (such as an Nvidia Jetson computing board or Nvidia-equipped PC) then you could enable the librealsense SDK's **CUDA** support in order to offload alignment processing work from the GPU onto the Nvidia graphics GPU and thus achieve a significant drop in CPU percentage usage.

If depth is aligned to color then yes, RGB color intrinsics should be used.  Using depth intrinsics would cause inaccuracy.  This is because after alignment to color, the origin point of depth changes from the center-line of the left IR sensor of the camera to the center-line of its RGB sensor.

As far as I am aware, depth and color retain their individual distortion models.  The depth-to-color alignment process is just mapping the depth coordinates onto the equivalent RGB color coordinates.

The five distortion coefficients used by RealSense cameras used to all be set to zero - as described at https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916 - but from the D455 model onwards, depth and IR streams have all-zero coefficients but the RGB color stream distortion coefficients are non-zero.  So there could be an issue if you set all the RGB coefficients to zero.

Below is an example of the non-zero color coefficients from a D455.

[image]

Every individual camera has unique intrinsics and extrinsics due to the manufacturing process at the factory.  You can find the values for your own camera by launching the rs-enumerate-devices tool in calibration information mode with the -c flag:

`rs-enumerate-devices -c`