realsenseai/librealsense #13825: Can not pip install https://github.com/IntelRealSense/librealsense/releases/download/v2.56.3/pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta.whl

HaotianHehaha, 2025-03-10:
I can not use pip install pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta.whl 
 from https://github.com/IntelRealSense/librealsense/releases/download/v2.56.3/pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta.whl

I use Ubuntu 24.04LTS and my python version is 3.10.16, but I got
`ERROR: pyrealsense2-2.56.3.7838-cp310-cp310-manylinux1_x86_64_beta.whl is not a supported wheel on this platform.`

MartyG-RealSense, 2025-03-10:
Hi @HaotianHehaha  When you run this command in the terminal, are you in the same directory that the .whl file has been downloaded to?

Also, are you using a PC computer or a non-PC computing board such as Nvidia Jetson or other type of Arm board?  The **manylinux** .whl will not work with Arm-based boards such as Jetson.  For installing pyrealsense2 on Arm devices, it is necessary to use Arm-specific **aarch** .whl packages.  

[image]

These are not available for the 2.56 beta versions though, and the most recent librealsense version that provides them at the time of writing this is **2.55.1**.

HaotianHehaha, 2025-03-10:
Hello, thanks for your advice. I am confirmed that I am in the same directory that the .whl file has been downloaded to. And I am using a PC computer. So I don't know what happened.

HaotianHehaha, 2025-03-10:
I tried, but the command works only if I download pyrealsense2-2.55.1.6486-cp310-cp310-manylinux1_x86_64.whl to my local directory, so the installation process finds the file from my local path.

HaotianHehaha, 2025-03-10:
Amazing! I use librealsense v2.56.3 and pyrealsense2-2.55.1, it can work!

MartyG-RealSense, 2025-03-15:
Hi @HaotianHehaha  Do you require further assistance with this case, please?  Thanks!

HaotianHehaha, 2025-03-15:
No, thanks for your help!

closed 2025-03-15 by HaotianHehaha