realsenseai/librealsense #13829: Color frame didn't arrive within 5000 while depth sensor works fine using pyrealsense2 in Ascend Atlas 200i DK A2

Boatsure, 2025-03-11:
The Ascend Atlas 200i DK A2 is an arm64 edge computing device made by Huawei and an Ubuntu22.04 is flashed.

I installed Librealsense SDK 2.55.1 from source.

Now the rs-enumerate-devices output normally. The realsense-viewer works all fine. I mean the Stereo Module and the RGB Camera can be switched on at the same time perfectly in either 2D or 3D mode.

And I'm using pyrealsense2 for developing. The weird thing is that, the depth pipeline can wait_for_frames() successfully, while the color pipeline always reports _frame didn't arrive within 5000_

Specifically, the codes I'm using are as follows:

```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth or color streams
pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))
print(device_product_line)

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

# depth sensor works fine
# config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# color sensor: frame didn't arrive within 5000
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()

        # depth_frame = frames.get_depth_frame()  # for depth
        color_frame = frames.get_color_frame()  # for color

        # Convert images to numpy arrays
        # depth_image = np.asanyarray(depth_frame.get_data())  # for depth
        color_image = np.asanyarray(color_frame.get_data())  # for color

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense',color_image)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
```
It always reports  
```
frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
But at the same time, I can access the RGB sensor with following code:

```
import cv2

def main():
    cap = cv2.VideoCapture('/dev/video4')

    if not cap.isOpened():
        print("fail to open")
        return
    
    while True:
        ret, frame = cap.read()
        if not ret:
            print("no frame, existing")
            break
        
        cv2.imshow('Camera', frame)
        
        if cv2.waitKey(1) & 0xFF == 27: 
            break
    
    cap.release()
    cv2.destroyAllWindows()

if __name__ == "__main__":
    main()
```
So what's the problem?

MartyG-RealSense, 2025-03-11:
Hi @Boatsure  Does it make a difference if you insert the hardware_reset() Python code below after the `config = rs.config()` line to reset the camera when the script is run?

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

Boatsure, 2025-03-12:
Hello @MartyG-RealSense , thanks for your reply.

While adding the hardware reset code to color visualization has no effect. 

On the contrary, when adding the reset code to the depth visualization version,  there is only one depth frame displayed, and then the error "RuntimeError: Frame didn't arrive within 5000" is reported immediately.

And different D435 and cables are tested, the error has been reproduced stably.

Boatsure, 2025-03-13:
Did you mean remove all the lines related with the config() as follows？

```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth or color streams
pipeline = rs.pipeline()

# Start streaming
pipeline.start()

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()

        color_frame = frames.get_color_frame()  # for color

        # Convert images to numpy arrays
        color_image = np.asanyarray(color_frame.get_data())  # for color

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense',color_image)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
```

I tried in color and depth version code, both of them still reported Frame didn't arrive within 5000 :(

MartyG-RealSense, 2025-03-13:
Does the problem still occur if you reduce the color FPS from 30 to 6 ?

Boatsure, 2025-03-14:
I have tried different resolutions and fps in the first place, 60, 30, 15 or 6, neither of them worked

Boatsure, 2025-03-17:
Unluckily, `frames.keep()` doesn't work

MartyG-RealSense, 2025-03-17:
Going back to the beginning of this issue, I checked your code again and could not see any obvious problems that would cause 'No frames received within 5000' on the color stream only.  But this is typical for this kind of error, which happens with code that should be working but is not.

In regard to why your other RGB script works but the pyrealsense2 RGB code does not, it looks as though the second script is accessing the raw RGB video with ordinary Python code (not pyrealsense2 code) as though the RealSense camera is an ordinary video webcam.  It is not accessing the RealSense-formatted BGR8 stream.

I note that you installed the SDK from source,  What method did you use to perform the source-code installation, please?

Boatsure, 2025-03-18:
The command I used to install the Librealsense SDK is:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/local/miniconda3/bin/python -DPYTHON_INCLUDE_DIR=/usr/local/miniconda3/include/python3.9 -DPYTHON_LIBRARY=/usr/local/miniconda3/lib/libpython3.9.so`

The reason why I specify the python path is that if not, the cmake can not find the python in the system, also w

I'm planning to clear the old build and rebuild the SDK soon. Wait for my good news, hopefully.

MartyG-RealSense, 2025-03-18:
I look forward to your next update after the rebuild.  Good luck!

closed 2025-03-31 by Boatsure

MartyG-RealSense, 2025-03-31:
Okay, that's no problem at all.  Thanks very much for the update!