realsenseai/librealsense #13830: Save cropped area from L515 camera stream window as PLY file using C#

Arulchozhan, 2025-03-11:
Dear Connection Team,

I'm seeking suggestions for resolving an issue I'm facing. Currently, I'm trying to connect a camera to display both depth and color images, including the Region of Interest (ROI), in an image window upon button click. When clicked, I capture and crop the image within the ROI and want to save it as a PLY file, similar to the Depth Quality Tool.

However, the issue is that after saving the file using my UI, it's not a readable file. Is there a direct function in C# to save a PLY file

![Image](https://github.c

[image]

om/user-attachments/assets/60c02d6f-d7ff-42ab-9fe7-a589f1aff578)    

my current Code:
  private void SaveROIAsPly(BitmapSource depthFrame, BitmapSource colorFrame, Rect roi, Rect roi_colour, CameraIntrinsics intrinsics, string filePath)
        {
            if (depthFrame == null || colorFrame == null)
            {
                throw new ArgumentNullException("Depth or color frame is null.");
            }

            // Adjust ROI
            int startX = Math.Max(0, (int)(roi.X - roi.Width / 2));
            int startY = Math.Max(0, (int)(roi.Y - roi.Height / 2));
            int endX = Math.Min(depthFrame.PixelWidth, startX + (int)roi.Width);
            int endY = Math.Min(depthFrame.PixelHeight, startY + (int)roi.Height);

            Int32Rect croppingRect = new Int32Rect(
                startX,
                startY,
                Math.Max(0, endX - startX),
                Math.Max(0, endY - startY)
            );
            int startX_c = Math.Max(0, (int)(roi_colour.X - roi_colour.Width / 2));
            int startY_c = Math.Max(0, (int)(roi_colour.Y - roi_colour.Height / 2));
            int endX_c = Math.Min(colorFrame.PixelWidth, startX + (int)roi_colour.Width);
            int endY_c = Math.Min(colorFrame.PixelHeight, startY + (int)roi_colour.Height);

            Int32Rect croppingRect_c = new Int32Rect(
                startX,
                startY,
                Math.Max(0, endX_c - startX_c),
                Math.Max(0, endY_c - startY_c)
            );

            int bytesPerPixel = (depthFrame.Format.BitsPerPixel + 7) / 8;
            int stride = croppingRect.Width * bytesPerPixel;

            if (stride < 270) // Example validation
            {
                throw new InvalidOperationException($"Invalid stride: {stride}. Cropping dimensions might be incorrect.");
            }

            // Crop bitmaps
            CroppedBitmap croppedDepth = new CroppedBitmap(depthFrame, croppingRect);
            CroppedBitmap croppedColor = new CroppedBitmap(colorFrame, croppingRect);

            // Proceed with point cloud and .ply saving logic
            float[,] pointCloud = DepthToPointCloud(croppedDepth, intrinsics, startX, startY);

            
            CroppedBitmap croppedColorBitmap = new CroppedBitmap(colorFrame, croppingRect);
            byte[,] colorData = ExtractColorData(croppedColorBitmap);
            // byte[,] colorData = ExtractColorData(croppedColorBitmap);
            // byte[,] colorData = ExtractColorData(croppedColor);
            //float[,] colorData = DepthToPointCloud(croppedColor, intrinsics, startX, startY); //ExtractColorData(croppedColor);

            SaveAsPly(filePath, pointCloud, colorData);
        }
private void SaveAsPly(string filePath, float[,] pointCloud, byte[,] colorData)
        {
            // Validate inputs
            if (pointCloud == null || colorData == null)
            {
                throw new ArgumentException("Point cloud and color data dimensions must match.");
            }

            int numPoints = pointCloud.GetLength(0);

            // Create a StringBuilder to construct the .ply file content
            var plyContent = new StringBuilder();

            // Write PLY header
            plyContent.AppendLine("ply");
            plyContent.AppendLine("format ascii 1.0");
            plyContent.AppendLine($"element vertex {numPoints}");
            plyContent.AppendLine("property float x");
            plyContent.AppendLine("property float y");
            plyContent.AppendLine("property float z");
            plyContent.AppendLine("property uchar red");
            plyContent.AppendLine("property uchar green");
            plyContent.AppendLine("property uchar blue");
            plyContent.AppendLine("end_header");

            // Write PLY body (points and colors)
            for (int i = 0; i < numPoints; i++)
            {
                float x = pointCloud[i, 0];
                float y = pointCloud[i, 1];
                float z = pointCloud[i, 2];
                float nx = 0;
                float ny = 0;
                float nz = 0;
                byte r = colorData[i,0];
                byte g = colorData[i,1];
                byte b = colorData[i,2];

                plyContent.AppendLine($"{x} {y} {z} {nx} {ny} {nz} {r} {g} {b}");
            }

            // Save to file
            File.WriteAllText(filePath, plyContent.ToString());
        }

MartyG-RealSense, 2025-03-11:
Hi @Arulchozhan  You can export a point cloud to .ply file in C# by using the instruction **Points.ExportToPLY()**.  https://github.com/IntelRealSense/librealsense/issues/7362 and https://github.com/IntelRealSense/librealsense/issues/7886 provide examples of using this instruction in C# code.

MartyG-RealSense, 2025-03-22:
Hi @Arulchozhan  Do you require further assistance with this case, please?  Thanks!

Arulchozhan, 2025-03-24:
Hello @MartyG-RealSense , 
Thank you for your assistance. I have a few follow-up questions to clarify:

1. I can now save a PLY file using your suggested method, but I've noticed a discrepancy between the displayed image frame and the saved PLY file. Specifically, the PLY file generated from my UI appears out of focus compared to the RealSense UI PLY file. Is there a parameter I need to configure to resolve this issue?

2. How can I save a portion of the live frame as a PLY file? I've tried cropping the frame and saving it as a bitmap, but I'm unable to use this bitmap to create a PLY file.

Thank you for your continued support

MartyG-RealSense, 2025-03-24:
1.  When you refer to a RealSense UI PLY file, do you mean a ply exported from the 3D pointcloud mode of the RealSense Viewer tool, please?  If you are then the Viewer uses a different ply export instruction from ExportToPLY() that is called **SaveToPLY()**.  This instruction provides a range of options for configuring the data of the exported ply, like in the Viewer's menu-driven ply configuration interface.

[image]

The Viewer is based on C++ code though and there is not an available example of implementing the SaveToPLY() instruction with C# code, unfortunately.

2.  Some RealSense users have cropped a ply using bounding box cropping with PCL (Point Cloud Library) code, like the script that a user shared at https://github.com/IntelRealSense/librealsense/issues/1441#issuecomment-377223449

closed 2025-04-10 by MartyG-RealSense