realsenseai/librealsense #13831: Realsense Viewer not showing X,Y and Depth

Akumar201, 2025-03-11:
Hello, I have upgrdaded the realsense-viewer , however when I am hovering over the image after turning on stereo and rgb module , I can not see any values , usually I was able to see that. 
```

realsense-viewer --version

realsense-viewer  version: 2.56.3.0
```

[image]

MartyG-RealSense, 2025-03-11:
Hi @Akumar201  The 3D pointcloud mode of realsense-viewer will display the coordinates beside the mouse cursor if it the cursor is hovered over a coordinate on the image that realsense-viewer considers to be valid.  It does not appear for every coordinate, especially not when hovering over plain black areas that do not contain depth information.

In the '2D' mode of realsense-viewer, the coordinate appears in a fixed position at the bottom corner of the realsense-viewer program window instead of beside the mouse cursor.

When hovering over an area that does clearly have detail on it, does the coordinate appear beside the mouse cursor if you move the camera closer to the objects, please?

[image]

MartyG-RealSense, 2025-03-22:
Hi @Akumar201  Do you require further assistance with this case, please?  Thanks!

Akumar201, 2025-03-22:
I think it's resolved for now. Thank You

closed 2025-03-22 by Akumar201