realsenseai/librealsense #13832: No Device Connected SR300 Ubuntu 22.04.5

MiguelFranco6, 2025-03-11:
I am using a VM, with Ubuntu 22.04.5 LTS, I have installed librealsense SDK 2.54.2 following these steps https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692, I can run any realsense command and it works, like realsense-viewer or rs-enumerate-devices. The problem is that when I try running an example even if it is from the downloaded source code, I always get the same error: 

[image]
 
I have tried different approaches to handle this error, but none worked (https://github.com/IntelRealSense/librealsense/issues/13125 )

Can the kernel version be the problem? Even if it is recognized by realsense commands?
[image]

PS:  I used version 2.54.2 because it is the latest one that detects SR300 and I didn t use version 2.50.0 because the python version is older than mine and didnt want to change the python version

MartyG-RealSense, 2025-03-11:
Hi @MiguelFranco6  Even though it is possible for 2.54.2 to access SR300, it has not been tested (validated) with SR300 and the last SR300-validated version was 2.50.0, as you say.  

Official support for kernel 6.8 was also not introduced until SDK version 2.56.2 beta, though the kernel version does not matter if librealsense was installed from source code with the 'libuvc backend' or 'RSUSB backend' methods.

Is the SR300 able to start if the simple code below is tested?

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
profile = config.resolve(pipeline)
profile = pipeline.start()
```

MiguelFranco6, 2025-03-11:
No, the same error is shown 
profile = config.resolve(pipeline)
RuntimeError: No device connected

MartyG-RealSense, 2025-03-11:
I would recommend next trying to build librealsense again with the https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 instructions but this time build the 2.50.0 source code and folder and include the flag `-DBUILD_PYTHON_BINDINGS=true ` in the CMake build instruction so that librealsense and the pyrealsense2 wrapper are built at the same time and you do not need to install pyrealsense3 separately.

If you have a newer Python 3 version than 2.50.0 has support for then this should not be an issue, as the pyrealsense2 wrapper can work with newer Python 3 versions than librealsense officially supports when the wrapper is built from source code.

MiguelFranco6, 2025-03-11:
Thank you for the fast response, unfortunately it didnt work, same error occurred while realsense-viewer still works. One thing I usually need to do is copy the 99-realsense-libusb.rules file to the rules.d folder, can the error come from this file? I usually copy de rules file from the source code downloaded.

MiguelFranco6, 2025-03-11:
Can t use `udevadm control --reload-rules && udevadm trigger` Permission was denied. Is there any other solutions to these problem?

MartyG-RealSense, 2025-03-11:
It is not likely an issue related to the camera being an SR300, as the same problem with Python non-detection can also occur with newer RealSense 400 Series cameras.

Sometimes attaching the camera to a mains electricity powered USB hub instead of plugging it directly into the computer can help with problems related to detecting and starting the camera, since powering the camera from the mains plug socket instead of the computer's USB port is more reliable.  There are no guarantees though with this method, so it may not be worth buying a powered hub if you do not already have one available to test with.

MiguelFranco6, 2025-03-11:
I don t actually have a powered USB hub, do you know if in ROS the same error persists?

MartyG-RealSense, 2025-03-11:
SR300 has not been used with the RealSense ROS wrapper for years, and the last case of an SR300 not being detected with the ROS wrapper's equivalent **No RealSense devices were found!** error was in 2019, at https://github.com/IntelRealSense/librealsense/issues/3475

So I do not have a picture of how the SR300 would behave now in the present day in the final wrapper versions that supported it (ROS1 - 2.3.2 and ROS2 - 4.51.1).

MiguelFranco6, 2025-03-11:
How would the powered USB hub help? If the camera is well connected and recognized by realsense-viewer and recognized by `lsusb -v`. I only did `pip install pyrealsense2` can the version of pyrealsense2 be the problem?

MartyG-RealSense, 2025-03-11:
As mentioned above, USB connections can be more reliable on a powered hub because they can supply power more stably to the camera.  There is no guarantee they can make a difference though, especially if realsense-viewer is working well on your USB connection, which is why I do not recommend buying one to resolve a detection problem with pyrealsense2 unless you are happy to write off the purchase cost of the hub if it doesn't work.

If you have already installed librealsense 2.54.2 then it is okay to install pyrealsense2 separately afterwards with **pip install pyrealsense2**, The pip install method only supports Python 3.7 to 3.11 though, so it would not be usable with a newer Python 3 version such as 3.12.

MiguelFranco6, 2025-03-11:
I have seen some discussions talking about copying the build output (librealsense2.so and pyrealsense2.so) next to my script. Will this do something? My script is literally on a Desktop folder, should it be elsewhere?

MartyG-RealSense, 2025-03-11:
Putting the script file and the .so file in the same folder together is usually a fix for errors that appear when launching a script - such as **No module named pyrealsense2** - but not the problem of the camera not being detected at all.

MiguelFranco6, 2025-03-11:
Thank you for your help, can t really find a solution to this problem. If I find the solution I will update here.

MiguelFranco6, 2025-03-12:
I guess it really was the pyrealsense2 version, I downgraded it and it started working. @MartyG-RealSense, I will close this because for now it is working

closed 2025-03-12 by MiguelFranco6

MartyG-RealSense, 2025-03-12:
That's great news that it is working now - thanks very much for the update!