realsenseai/librealsense #13835: Install onto Jetson Nano w/o Wifi

Shahab-Ahmed0, 2025-03-11:
I have a Jetson Nano inside of a quadruped robot and I am trying to install pyrealsense on their but it does not not have wifi capabilities. 

I have tried installing the .deb packages onto a usb and then installing them on the jetson but since my laptop is AMD64 and the jetson is ARM64 it does not work.

Any advice is appreciated.

MartyG-RealSense, 2025-03-11:
Hi @Shahab-Ahmed0  If you need to install both the librealsense SDK and pyrealsense2 on your Nano at the same time and do not have an internet connection, building the SDK in 'native' mode (also known as -DFORCE_RSUSB_BACKEND=OFF) may suit your needs, as the installation will be able to be performed without needing internet.

There is a guide for installing librealsense and pyrealsense2 from source code on Nano with CMake at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 though it is quite old now so will need some modification.  

You can download the librealsense source code as a zip file onto your USB stick, copy the file onto your Jetson and extract the zip and install from there.

https://github.com/intelrealsense/librealsense/releases

[image]

1.  Change references to Python 3.6 to whatever Python 3 version you are using.  If you are not sure what number to put for your Python 3 version, changing **python3.6** to simply **python3** should still work.

2. Change the build flag -DFORCE_RSUSB_BACKEND=ON to ` -DFORCE_RSUSB_BACKEND=OFF` so that the installation process is not dependent on an internet connection.

MartyG-RealSense, 2025-03-31:
Case closed due to no further comments received.

closed 2025-03-31 by MartyG-RealSense

Shahab-Ahmed0, 2025-04-07:
Ok I was able to create the build files succesffuly by commenting out the areas in cmake that were downloading pybind11 & put the pybind11 files into the correct place in librealsense. 

I got build files have been writte.

This is progress but what next?

Thanks for all your help.

MartyG-RealSense, 2025-04-08:
You could go to your **librealsense-2.54.1** librealsense source code folder and look in the directory location below to see what pyrealsense2 wrapper files are there.  Can you provide me with an image of the contents of that directory please?

`librealsense-2.54.1 > build > wrappers > Python `

Shahab-Ahmed0, 2025-04-08:
Hi Marty. I was able to get around this by cloning the pybind11 repository, putting it into a repository and then placing it in the pybind11 folder inside of the pyrealsense directory. 

I was able to successfully create build files. However, when I try to make I get this error:

[ 52%] Building CXX object CMakeFiles/realsense2.dir/src/to-string.cpp.o
[ 52%] Building CXX object CMakeFiles/realsense2.dir/third-party/easyloggingpp/src/easylogging++.cc.o
[ 52%] Building CXX object CMakeFiles/realsense2.dir/common/utilities/time/l500/get-mfr-ww.cpp.o
[ 52%] Linking CXX shared library librealsense2.so
[ 53%] Built target realsense2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

I probably won't be back to the lab for a while.

MartyG-RealSense, 2025-04-09:
Possibly the problem is that you are trying to install without wi-fi but the `-DFORCE_RSUSB_BACKEND=ON` flag requires an internet connection in order to install dependencies over the internet.  So removing -DFORCE_RSUSB_BACKEND=ON should enable installation to take place offline.

This adds some complication though, as you will have to install the dependencies manually before performing the SDK installation with CMake.  Jetson instructions for doing so can be found at the link below.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend