realsenseai/librealsense #13837: Why are the frames not processed?

baptiste-lucas, 2025-03-11:
Good evening,

I am really stuck here. I coded a Camera class for my Realsense D435i camera but there is an issue that I am really confused with.   
I spent about two full days on this, I am sure that there is a solution...
Huge huge thank you in advance for your help!!!! 
**Please find the code attached below.**  

### Problem explanation 

When I call capture method (I call it only once) it seems that the parameter nb_waiting_frames is not important, i.e. it does not really iterate on frames (whatever the parameter value I always get the same last frame). However, calling the show_stream method shows that, in the end, the latest frame is what I expect. **I expect capture(4) and show_stream(5,0) to give the same last frame** : the 5th frame when one starts the pipeline. But this is not the case, only show_stream provides the 5th actual frame. The capture method only provides the first one. 
Analogously, **I expect show_stream(5,5) and show_stream(10,0) to give the same last frame**. However, the first gives you the fifth frame of the pipeline, even if you waited for the 5 first frames to be skipped. The second gives you the tenth frame as expected. Why would we have this difference? I tried to find the problem but it's really super weird. I tried to put a time.sleep(0.1) in the for loop of __start_pipeline method but it does not change nothing. I also tried poll_for_frames() in __check_pipeline(), but it does not do anything. I am using a timestamp in __check_pipeline() to see if frames are different.  

#### Example on show_stream()

As a concrete example, I am loading a .bag video (recorded with a Realsense D435i) :
```d435 = Camera(live=False, bag_path="bag_videos/20250121_115352.bag")```
Then, I try show_stream : 
```d435.show_stream(nb_frames=7,nb_waited_frames=0,delay=0.001)```  
I get as result (the plots are fine, I get six frames on matplotlib):
```[ERROR] Failed to get frames: Frame didn't arrive within 5000  
Stream stopped after 6 frames  
```  

This means that the .bag has six frames. Now, I call the same method again but now with waiting frames: ```d435.show_stream(nb_frames=15,nb_waited_frames=5,delay=0.001) ```  
I get as result (the plots are fine again, I get six frames on matplotlib):  
```
[ERROR] Failed to get frames: Frame didn't arrive within 5000  
Stream stopped after 11 frames
```

Thus, you see that it should have skipped the first 5 frames and only plotted the sixth one, not pursuing further since the video contains 6 frames. However, it acted as if the waiting frames represented nothing, plotted the six frames and ended there (thus 5+6=11). You still see that it stopped after these six and did not go further even if the parameter was set to 15.  

#### Example on capture() and frame timestamps

Finally, calling d435.capture with any number for nb_waited_frames always gives the same rgbd data.   
To give you an example of the evolution of frame timestamps, I called the capture method with parameter set to 20. It gave:  
```
1737456842219.504  
A  
1737456842252.8699  
A  
1737456842286.2358  
A  
1737456842286.2358  
A  
1737456842319.6018  
A  
1737456842319.6018  
A  
1737456842352.968  
A  
1737456842352.968  
A  
1737456842386.3345  
A  
1737456842386.3345  
A  
1737456842419.701  
A  
1737456842419.701  
A  
1737456842453.0674  
A  
1737456842453.0674  
A  
1737456842486.434  
A  
1737456842486.434  
A  
1737456842519.8008  
A  
1737456842519.8008  
A  
1737456842553.1677  
A  
1737456842553.1677  
A  
1737456842586.5347  
A   
```  

### Class complete code
```
# import time
import numpy as np                       
import matplotlib.pyplot as plt          
from IPython.display import clear_output 
import pyrealsense2 as rs  

class Camera:
    def __init__(self, live:bool, bag_path:str=None):
        self.__live = live
        self.__pipe = rs.pipeline()
        self.__cfg  = rs.config()
        self._bag_path = bag_path
        self._profile = None
        self._depth_scale = None
        self._last_rgb = None
        self._last_depth = None

        self.check_live()
        self.__prepare_pipeline()

    # check if the camera is live or a .bag file is provided (but not both)
    def check_live(self):
        if not self.__live and self._bag_path is None:
            print("No .bag is provided")
        elif self.__live:
            print("Live camera is used")
            self.check_device()
        if self.__live and self._bag_path is not None:
            print("Live but .bag is provided. Will use .bag file")
            self.__live = False
        elif not self.__live:
            print(f"Using .bag file: {self._bag_path}")
            self.check_device()

    # set up the config
    def __prepare_pipeline(self):
        if self.__live:
            self.__cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
            self.__cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
        else:
            self.__cfg.enable_device_from_file(self._bag_path, repeat_playback=False)

    def __check_pipeline(self):
        try:
            frames = self.__pipe.wait_for_frames(timeout_ms=5000) 
            print(frames.get_timestamp())
            # self.__pipe.poll_for_frames()  # Flush old frames and request a new one
            print("A")
            return frames
        except RuntimeError as e:
            print(f"[ERROR] Failed to get frames: {e}")
            return None
        
    # align RGB and depth frames
    def __align_frames(self):
        frameset = self.__check_pipeline()
        if frameset is None:
            return None
        align = rs.align(rs.stream.color)  
        return align.process(frameset)

    def __prepare_frames(self, aligned:bool=True):
        if aligned:
            return self.__align_frames()
        else:
            return self.__check_pipeline()
        
    def __get_frames(self, aligned:bool=True):
        frameset = self.__prepare_frames(aligned)
        if frameset is None:
            return None, None
        else:
            return frameset.get_color_frame(), frameset.get_depth_frame()
        
    # pipeline start and skipping first frames, waiting for better exposure
    def __start_pipeline(self, nb_waited_frames:int=5):
        self._profile = self.__pipe.start(self.__cfg)
        self._depth_scale = self._profile.get_device().first_depth_sensor().get_depth_scale()
        if nb_waited_frames < 0:
            nb_waited_frames=0
        for _ in range(nb_waited_frames):
            _, _ = self.__get_frames()
        #     time.sleep(0.1)
        # start_time = time.time()
        # while (time.time() - start_time) < (nb_waited_frames / 30):  # Assuming 30 FPS
        #     self.__get_frames()
        
    # capture RGB and depth frames after waiting for better exposure
    def capture(self, nb_waited_frames:int=5,aligned:bool=True):
        self.__start_pipeline(nb_waited_frames)
        self._last_rgb, self._last_depth = self.__get_frames(aligned)
        self.__pipe.stop()

    # show stream (delay in seconds)
    # save RGB and depth frames when 's' is pressed ... need to find how to do it in SSH
    # press 'q' to quit
    def show_stream(self,nb_frames:int=10,nb_waited_frames:int=5,aligned:bool=True, delay:float=0.1): # press a key to stop the stream
        colorizer = rs.colorizer() 
        _, axs = plt.subplots(nrows=1, ncols=2, figsize=(12,4)) # Show 1 row with 2 columns for Depth and RGB frames
        title = ["Depth Image", "RGB Image"]  
        self.__start_pipeline(nb_waited_frames)
        for _ in range(nb_frames):  
            color_frame, depth_frame = self.__get_frames(aligned)
            if color_frame is None and depth_frame is None:
                break
            colorized_streams = []                                # This is what we'll actually display
            if depth_frame:
                colorized_streams.append(np.asanyarray(colorizer.colorize(depth_frame).get_data()))
            if color_frame:
                colorized_streams.append(np.asanyarray(color_frame.get_data()))
            nb_waited_frames+=1
            for i, ax in enumerate(axs.flatten()):                # Iterate over all (Depth and RGB) colorized frames
                if i >= len(colorized_streams): continue          # When getting less frames than expected
                plt.sca(ax)                                       # Set the current Axes and Figure
                plt.imshow(colorized_streams[i])                  # colorized frame to display
                plt.title(title[i])                               # Add title for each subplot
            clear_output(wait=True)                               # Clear any previous frames from the display
            plt.tight_layout()                                    # Adjusts display size to fit frames
            plt.pause(delay)
        self.__pipe.stop()
        print(f"Stream stopped after {nb_waited_frames} frames")
```

MartyG-RealSense, 2025-03-12:
Hi @baptiste-lucas  If you only need a single frame then you could consider using the **save_single_frameset()** instruction, which saves only 1 frame to the bag file.  A pyrealsense2 script at the link below demonstrates its use.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

As you highlighted in your code, when not skipping the first 5 frames there is a risk that the first frame may have bad exposure.  This can be avoided by disabling auto-exposure and using a fixed manual exposure value, as the bad exposure only occurs when auto-exposure is enabled because it needs time over the first few frames to 'settle down' the exposure.

If the camera is being used in a location with consistent lighting all the time, like in an office, then using manual exposure should be okay.

MartyG-RealSense, 2025-03-12:
If **cfg.enable_device_from_file** is used then a script will behave the same as it would if a live camera was being used but will instead use the streams stored in the bag file as the source of the data instead of a camera.

As far as I am aware, a script with cfg.enable_device_from_file that uses a 'for x in range' instruction on the line before wait_for_frames() should skip the first 5 frames of an imported bag file that is being played back in the same way that the first 5 frames of a live camera would be skipped.  So the plotting of the first 5 frames should not usually be necessary.

You should not need to put **timeout_ms=** in the wait_for_frames() brackets.  The value on its own will be fine.

`wait_for_frames(5000)`

baptiste-lucas, 2025-03-13:
Ok I agree with you, but I still have the same problem. And it is independent of the .bag file I am using. It is really strange. I searched on several tutorials but I always have a difference between them. Maybe you can try the code with a .bag that you have to see if this problem is only on my side or not?

MartyG-RealSense, 2025-03-13:
I am not able to test code on my computer.  I do apologize.

Are you only plotting the frames so that you can skip the first five of them, please?  If so then the script would be simpler and have less to go wrong in it if you could strip out the plotting mechanism and re-record the bag using a Python bag recording script with` for x in range(5):` so that the first 5 frames are not included in the bag.

You could also investigate the possibility of disabling auto-exposure and using manual exposure if the lighting level is consistent in the place that bag recording takes place in, so that it would not be necessary to skip frames at all (because the bad-frame phenomenon does not occur with manual exposure).

Or are you only able to use the bag file that has already been recorded?

Looking at your script again, I note that it is not using a bag playback function called **set_real_time**.  It is recommended to set **set_real_time** to false when playing back bag frames, otherwise the playback could be erratic. 
 A Python bag playback script at https://github.com/IntelRealSense/librealsense/issues/6340 demonstrates use of set_real_time.

MartyG-RealSense, 2025-03-22:
Hi @baptiste-lucas  Do you require further assistance with this case, please?  Thanks!

baptiste-lucas, 2025-03-24:
Hello, I did not manage to find a solution. I've tried what you proposed but there is still this strange behaviour with wait_for_frames that does not process the frames as I expected. We can close this subject as I don't have the time to find the actual issue...
Thank you anyway for your help!

closed 2025-03-24 by baptiste-lucas

MartyG-RealSense, 2025-03-24:
You are very welcome.  I'm sorry that a solution was not found.  Thanks very much for the update.