realsenseai/librealsense #13839: realsense-viewer does not show the camera

du-xinyi, 2025-03-12:
When I compile the librealsense in master branch, realsense-viewer can detect the camera. But when change the branch in develop, realsense-viewer can not detect the camera. Besides, all function is normal

du-xinyi, 2025-03-12:
rs-enumerate-devices can find the camera

MartyG-RealSense, 2025-03-12:
Hi @du-xinyi  Did you uninstall the master-branch installation of the librealsense SDK before building the development branch to ensure that there are not conflicts between more than one installation of the librealsense SDK on the same computer, please?

du-xinyi, 2025-03-12:
> 你好[@du-xinyi](https://github.com/du-xinyi) 在构建开发分支之前，您是否卸载了 librealsense SDK 的主分支安装，以确保同一台计算机上多个 librealsense SDK 安装之间不会发生冲突？

yes, I uninstall the compile result when I change the branch

du-xinyi, 2025-03-12:
> Hi [@du-xinyi](https://github.com/du-xinyi) Did you uninstall the master-branch installation of the librealsense SDK before building the development branch to ensure that there are not conflicts between more than one installation of the librealsense SDK on the same computer, please?

Yes,  I uninstall the compile result when I change the branch. My system is ubuntu 22.04

MartyG-RealSense, 2025-03-13:
One of my Intel RealSense colleagues mentions at https://github.com/IntelRealSense/librealsense/discussions/13572#discussioncomment-11985469 that the next non-beta final (master) release of the SDK is being targeted for a release window of end Q1 / start Q2 (i.e end of March / start of April 2025).  So it may be worth waiting for the next master release if you have the option to do so to see if it resolves your realsense-viewer camera detection issue.

closed 2025-03-13 by du-xinyi