realsenseai/librealsense #13845: hwmon command 0x80( 8 3 0 8 ) failed (response -7= HW not ready) during on-chip calibration

vaigen, 2025-03-12:
Hi folks - 

I'm using RS2 SDK 2.56.1 with D405 firmware 5.16.0.1. Running on-chip calibration as pointed out in https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras results in 

`_hwmon command 0x80( 8 3 0 8 ) failed (response -7= HW not ready)_ `

Unplugging/replugging camera or restarting it has no effect on these results. How do we solve this problem?

I've gone through these threads:

https://github.com/IntelRealSense/librealsense/issues/7087
https://github.com/IntelRealSense/librealsense/issues/12014
https://github.com/IntelRealSense/librealsense/issues/13340

and none of them point to a solution. We have 10s of cameras in field where unplugging/using an external computer is not an option. How do we go about calibrating these cameras?

Thanks
Vai

MartyG-RealSense, 2025-03-13:
Hi @vaigen  Using the earlier RealSense SDK version **2.49.0** can sometimes correct this problem with on-chip calibration.  This is because of changes made to the on-chip calibration tool in SDK 2.50.0 onwards.

***********

As mentioned in https://github.com/IntelRealSense/librealsense/issues/13340 an alternative to calibrating the camera is to reset it to its factory-new default calibration in the RealSense Viewer tool using instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

It can be installed on Ubuntu using instructions on page 14 onwards of the tool's user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

*************

If you need to perform calibration, there is an alternative RealSense calibration tool called **Dynamic Calibration** - also mentioned in https://github.com/IntelRealSense/librealsense/issues/13340 - though a drawback of this tool compared to on-chip is that it cannot be controlled with program scripting.

vaigen, 2025-03-14:
@MartyG-RealSense We are in a position where we can't downgrade the SDK easily without losing a lot of bugfixes. This seems like a big gap in a major SDK release which has existed for 2.5 years now and I think we should fix it for all Realsense customers. What are the plans to fix on-chip calibration? or just release another on-chip calibration tool that works for releases 2.5x?

MartyG-RealSense, 2025-03-14:
My understanding is that it's not a bug that affects all users but rather a computer hardware specific issue, with on-chip calibration working better on some PCs than others.  There is no way to know in advance though how on-chip will behave on a specific model of computer / computing device without trying it first on that computer.

May I ask how you are performing on-chip calibration in the field?  For it to work the camera has to be directly facing a paper print-out of a calibration target image or the calibration will fail.  And are the cameras in fixed positions or mounted on a mobile robot?

If the cameras are mobile and can move to a particular location then the OEM version of the Dynamic Calibration tool that is bundled with Intel's $1500 USD OEM Calibration Target product can connect to cameras remotely in the field via an Ubuntu server.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

vaigen, 2025-03-17:
@MartyG-RealSense We can have have a textured surface or paper print out in front of the camera for on-chip calibration no problem, but we can't open up an IPx enclosure to connect RS viewer for this. Plus we have 100s of "devices" in field and this is highly impractical. Since are devices are in a high vibe/shock environment we have to perform on-chip calibration every week or so. We just use our main SoC to do this in-system.

As for the SDK issue, "works on some computers" sounds very much like a bug to me. We couldn't say the same about other  SDK features, right? We can't have the SDK work on some computers and not others. May I ask what attribute of a computer makes this fail? The interface and connections remain the same.

Kindly help resolve this issue with 2.50x SDKs because on-chip calibration is a core feature without which the sensor won't work as intended.

MartyG-RealSense, 2025-03-18:
It is not known what may cause on-chip to work on one computer but experience the hardware error on another computer when using 2.50.0 or newer, except that the computers are typically by different manufacturers.

***********

Have you confirmed that your cameras need to be recalibrated?  Although **severe** vibration can result in the possibility of miscalibration, the depth sensing algorithm of RealSense 400 Series cameras is normally robust against vibration.

Although you cannot obtain a 'health check' score for the calibration whilst the hardware error is occurring, you should be able to tell from the general quality of a camera's depth image how well calibrated the camera is.  An On-Chip calibration can improve the depth image quality (e.g how many holes it has), but the self-calibration mode responsible for improving depth measurement accuracy is **Tare** calibration.

********

Can you confirm please that when attempting on-chip calibration you are using the target image printout from Appendix A of the self-calibration guide and have downloaded and printed the target using one of the PDF files provided on that appendix?

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#appendix-a-example-textured-target

[image]

**********

If you are unable to use realsense-viewer, are you activating on-chip calibration using a programming language script in code such as Python, like the official Python on-chip calibration example program **depth_auto_calibration_example.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py

********

Sometimes, downgrading a camera's firmware driver version to **5.13.0.50** can enable a function to work when it otherwise would not.  Would it be possible to perform a test downgrade to firmware 5.13.0.50 on one of your cameras to test whether it makes a difference to activation of on-chip calibration?

MartyG-RealSense, 2025-03-31:
Hi @vaigen  Do you require further assistance with this case, please?  Thanks!

vaigen, 2025-04-02:
Hi Marty - figured I'd post this. The issue has nothing to do with a specific kind of computer. It's got to do with settings when running on chip calibration.

On chip calibration only runs with depth stream enabled at 256x144 @ 90 fps. This is specified in Tablet 4-3 of the manual.

One of your colleagues supplied a C++ example which was super useful in resolving the discrepancy.

https://github.com/IntelRealSense/librealsense/pull/13902/commits/18332d35bae6e7aacdc1414fcf47ca7c566dcd7d

Thanks, we should mark this as resolved with this solution.

vaigen, 2025-04-02:
Closed with solution above.

closed 2025-04-02 by vaigen

MartyG-RealSense, 2025-04-02:
Thanks very much for the update and the details of the solution!