realsenseai/librealsense #13847: Instructions for installing Unity Wrapper for real sense D405 on Ubuntu

cheabdullahjz, 2025-03-13:
Hi,

Is there any Instructions for installing Unity Wrapper for real sense D405 on Ubuntu .

Thanks

MartyG-RealSense, 2025-03-13:
Hi @cheabdullahjz  Usually the RealSense Unity wrapper is used on Windows, though a small number of RealSense users have attempted an Ubuntu installation.  There are references at https://github.com/IntelRealSense/librealsense/issues/11529#issuecomment-1455982525 and https://github.com/IntelRealSense/librealsense/issues/8120

cheabdullahjz, 2025-03-13:
Hi [MartyG-RealSense](https://github.com/MartyG-RealSense),

So, there is no answer to both references at https://github.com/IntelRealSense/librealsense/issues/11529#issuecomment-1455982525 and https://github.com/IntelRealSense/librealsense/issues/8120

is there any effort on Real Sense side to make this unity wrapper for linux?

from Keijiro twitter. 
The RealSense Unity plugin also works on the Linux Editor. Hint: Copy [http://librealsense2.so](https://t.co/p1yy1ULCVP).2.18.1 to /Assets and rename it to [http://librealsense2.so](https://t.co/p1yy1ULCVP)

MartyG-RealSense, 2025-03-13:
Development of updates for the RealSense Unity wrapper has ceased unfortunately, so no further work will be done on it.

cheabdullahjz, 2025-03-13:
copying librealsense.so to /Assets and using Windows Unity Project file with existing script/class/package for unity wrapper does not work.

Thanks.

MartyG-RealSense, 2025-03-13:
There is not any advice that I can offer about attempting installation of the Unity wrapper on Ubuntu, as it is meant to be used on Windows.  I do apologize.

cheabdullahjz, 2025-03-13:
Hi,

It is ok. thanks for the reply.
I will have to go down the python pathway.

Thanks.

closed 2025-03-13 by cheabdullahjz