realsenseai/librealsense #13848: Detect RGB values from RGBD point cloud

shreeya0411, 2025-03-13:
Hello,

I am currently working on processing point cloud data obtained from an Intel RealSense camera, with the goal of detecting specific colors (orange and green) from the captured RGBD values. However, I am facing challenges in successfully filtering out these specific colors. 

Camera Model: D405
Operating System:  Ubuntu 20.04

I use the command `roslaunch realsense2_camera rs_rgbd.launch` to stream data from my camera. I
Next, using rviz with the topic /camera/depth_registered/points I can see the point cloud being generated. 

Following is the python code I am using for reference 

```
import rospy
from sensor_msgs.msg import PointCloud2
import sensor_msgs.point_cloud2 as pc2
import struct
import numpy as np

# thresholds for y max and min 
LOWER_THRESHOLD = -0.2  # in meters (15 cm below the camera)
UPPER_THRESHOLD = 0.2   # in meters (camera height)

# rgb thresholds for orange and green 
def is_within_color_range(r, g, b):
    if ((155 <= r <= 255) and (0 <= g <= 100) and (0 <= b <= 100)):
         print("Orange Detected!")
         return True
    if ((155 <= r <= 255) and (155 <= g <= 255) and (0 <= b <= 100)):    
         print("Green Detected!)
         return True

def callback(point_cloud_msg):
    points_list = []
    for point in pc2.read_points(point_cloud_msg, field_names=("x", "y", "z", "rgb"), skip_nans=True):
        x, y, z, rgb = point

       # check if point is within y thresholds  
       if y < LOWER_THRESHOLD or y > UPPER_THRESHOLD:
            continue

      # print unpacked rgb value 
       print(f"Unpacked rgb value = {rgb}")

       # unpack the rgb values 
        rgb = struct.unpack('I', struct.pack('f', point[3]))[0] 
        r = (rgb >> 16) & 0xFF
        g = (rgb >> 8) & 0xFF
        b = rgb & 0xFF
        # Print RGB values
        print(f"Parsed RGB values: R={r}, G={g}, B={b}")

        # Check if point color is in the specified ranges
        if is_within_color_range(r, g, b):
            points_list.append([x, y, z, r, g, b])

    # Print the filtered points list 
    print("Filtered Points List: ")
    for point in points_list:
        print(point)

def listener():
    rospy.init_node('pcl_listener', anonymous=True)
    rospy.Subscriber('/camera/depth_registered/points', PointCloud2, callback)
    rospy.spin()

if __name__ == '__main__':
    listener()
```

Output is as follows 

Unpacked rgb value = -3.143129e+38
Point: x=0.106757, y=-0.100026, z=0.5630, R=116, G=128, B=129

I let this run for around 10 minutes and I don't see any orange detected or green detected despite my rgb thresholds being very wide. Following is the image of my setup....

[image]

I also see the bowls in the rviz pointcloud so I know that my camera is detecting them. 

My question now is 
1. As can be observed my unpacked rgb value on average is a very large negative value. In the instance stated above it is -3.143129e+38. Why is this? 
2. Is my method an acceptable way for point cloud processing and rgbd points targeting? Are there any other methods that others have utilised that are more efficient? 

Apologies for the very long message. 

Thank you for your assistance!

MartyG-RealSense, 2025-03-13:
Hi @shreeya0411  I looked up RGB color codes for orange and some references gave 165 as the G value for the color orange.  You could therefore try changing your G range value for orange from <100 to <200 to test whether <100 means that the orange bowls are not detected if its G is 165.

shreeya0411, 2025-03-13:
Hi @MartyG-RealSense Apologies for the repeated message, however, I forgot to mention in the previous comment. Would you have some insight as to why I am receiving a very large negative value for the unpacked RGB value? Again, thanks for your response.

MartyG-RealSense, 2025-03-14:
I had a careful look at your script regarding the issue with a large RGB value.  

It looks as though the unpacked RGB value that is printed originates from the blue-highlighted value below.  My apologies if I am mis-reading the script.

[image]

My knowledge of rospy scripts that obtain data by subscribing to a ROS topic is admittedly limited, so I do not know what is causing this very large negative value.  The only other case that I know of that uses rospy, rs_rgbd.launch and an xyzrgb point cloud is at https://github.com/IntelRealSense/realsense-ros/issues/3180

shreeya0411, 2025-03-15:
Just to update, I wrote a code that streams raw image from the camera, covered the entire camera with the bowl, and then found the average of rgb values of the entire image to help find exact thresholds required for the bowls. I also introduced hue, saturation and value parameters which helped me find the exact colours to target. Best regards.

MartyG-RealSense, 2025-03-15:
It's great to hear that you achieved a solution - thanks very much for the update!