realsenseai/librealsense #13854: E: Unable to locate package librealsense2-dkms oon raspi

saqihussainkalan, 2025-03-16:
i am using 

Operating System & Version | Ubuntu 22.04 (server)
kernel:  ``5.15.0-1073-raspi``
Camera : ``D455``

I have followed the steps [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages](url)

robot@ugv:/home/ws/ugv_ws$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-dkms
robot@ugv:/home/ws/ugv_ws$

MartyG-RealSense, 2025-03-16:
Hi @saqihussainkalan  The DKMS packages on the distribution_linux..md page are not suitable for use on Arm-based computing boards such as Raspberry Pi.  

It is recommended that installation on Pi with Ubuntu is performed by building the librealsense SDK from source code with the **libuvc backend** procedure.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend method provides a pre-made build script called 'libuvc_installation.sh'

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

saqihussainkalan, 2025-03-17:
@MartyG-RealSense Thank you quick reply,

i followed the steps, and its currently in build, i got something strange output do i need to care about,

[image]

[image]

[image]

[image]

saqihussainkalan, 2025-03-17:
and this is after build i run the launch 
`ros2 launch realsense2_camera rs_launch.py`

[image]

saqihussainkalan, 2025-03-17:
@MartyG-RealSense  can you please provide any assistance on this issue?

MartyG-RealSense, 2025-03-17:
Ideally, if you are using librealsense 2.55.1 and wrapper 4.55.1 then camera firmware version **5.16.0.1** should be used.  The 5.13.0.55 firmware version can work with wrapper 4.55.1 (and your launch has completed successfully), but I would normally only recommend using firmware 5.13.0.55 with it if it doesn't work with 5.16.0.1.

If the images are being published normally in RViz without any obvious problems though then the log messages can likely be safely ignored.

saqihussainkalan, 2025-03-24:
i am already using the firmware version i don't provide any nodes ``Firmware Version:5.13.0.55`` if i need to upgrade it to `5.16.0.1` may be ? 

because it only provide one `camera node` command 

[image]

[image]

saqihussainkalan, 2025-03-25:
right now i have only option to use it with raspi 4 

i tried set low resolution it failed again and  have also attached the topics and nodes at the same time 

`robot@ugv:~$ ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/image_raw/compressed
/camera/camera/color/image_raw/compressedDepth
/camera/camera/color/image_raw/theora
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/image_rect_raw/compressed
/camera/camera/depth/image_rect_raw/compressedDepth
/camera/camera/depth/image_rect_raw/theora
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/camera/camera/extrinsics/depth_to_depth
/parameter_events
/rosout
robot@ugv:~$ ros2 node list
/camera/camera
robot@ugv:~$ `

[image]

MartyG-RealSense, 2025-03-25:
When you first installed librealsense from dkms packages at the beginning of this case, did you remove that installation from your Pi 4 before performing the subsequent source code installation, please?  There may be conflicts if you have more than one installation of librealsense on the same computer.

If you did not uninstall the dkms packages or did not wipe the Pi's storage before performing the source code installation, the command below can remove the packages.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

saqihussainkalan, 2025-03-28:
Steps I performed.

1. https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md (For SDK) and it successfully compiled 

and then, i performedthe  next step 

2. Clone the latest ROS2 Intel® RealSense™ wrapper from [here](https://github.com/IntelRealSense/realsense-ros.git) into '~/ros2_ws/src/'

Unfortunately it did not work, and then i removed this from src folder and  tried ``sudo apt install ros-humble-realsense2-*`` this command 

this is what i did

MartyG-RealSense, 2025-03-28:
It can be difficult to get the RealSense ROS wrapper to work on Raspberry Pi.

Have you tried launching with the **ros2 run** method instead of ros2 launch ?

`ros2 run realsense2_camera realsense2_camera_node`

saqihussainkalan, 2025-04-01:
Yes, I have tried this, and it gives the same output 

[image]

MartyG-RealSense, 2025-04-01:
A couple of RealSense users who have had problems with running the ROS wrapper on Raspberry Pi have resolved their issue by writing their own simple custom RealSense ROS node - like the user at https://github.com/IntelRealSense/realsense-ros/issues/2991#issuecomment-1950557027 - though this approach requires programming knowledge and so is not suited for everyone.

Let's try an even lower depth and color stream configuration for D455:

`ros2 launch realsense2_camera rs_launch.py ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x5 rgb_camera.color_profile:=640x480x5`

You could also try upgrading your firmware from 5.130.055 to 5.16.0.1, as we discussed earlier in this case.

saqihussainkalan, 2025-04-02:
I tried this it gives the same issues. It is even providing me the data of all camera topics, but `NO TF found`

Can you guide me on how I can i upgrade firmware from 5.130.055 to 5.16.0.1

MartyG-RealSense, 2025-04-02:
If you have access to the RealSense Viewer tool then you can upgrade the firmware to 5.16.0.1 in it using the instructions below.

1.  Download the firmware .bin file for 5.16.0.1 from the firmware releases page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases

2.  Open the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** option from its drop-down menu.  This will cause a file selector box to pop up.

3.  Navigate to the location where you downloaded the firmware .bin file to and select the file.  The firmware update process should then automatically begin.

*********

If you do not have the RealSense Viewer then the other method of updating the firmware version is to use the **rs-fw-update** tool.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

Or you could perform the update in the Viewer on a Windows computer if you have access to one, as the firmware is stored inside the camera and not on the computer, so the firmware is carried in the camera to whatever computer / computing board you attach it to.

saqihussainkalan, 2025-04-03:
I tried the first method and its showing this 

[image]

saqihussainkalan, 2025-04-03:
All right, now I have upgraded it on a Windows machine. Now my firmware is 5.16.0

MartyG-RealSense, 2025-04-03:
**Connection timed out** suggests to me that the wrapper had not received frames from the camera for 5 seconds (the time after which the pipeline time-outs if it has not received new frames).  

Sometimes streaming stops working if color is enabled.  Does it still time-out if you disable color in the launch?

`ros2 launch realsense2_camera rs_launch.py ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x5 enable_color:=false`

MartyG-RealSense, 2025-04-04:
Have you tried including **initial_reset:=true** in your launch instruction to reset the camera at launch, please?

If you use your custom node script, does the lag reduce if you change 5 FPS back to 15 FPS?

The use of Raspberry Pi 4, which is your only option right now, may be a factor in the low performance that is difficult to work around, unfortunately.

MartyG-RealSense, 2025-04-10:
Hi @saqihussainkalan  Do you require further assistance with this case, please?  Thanks!

saqihussainkalan, 2025-04-11:
> Have you tried including **initial_reset:=true** in your launch instruction to reset the camera at launch, please?
>
> If you use your custom node script, does the lag reduce if you change 5 FPS back [...]

I tried to set it up on 15, but this still presists the same issue (lagging).

MartyG-RealSense, 2025-04-11:
Is your joystick a wireless one that uses infrared signals to communicate?  RealSense cameras can interfere with the operation of wireless IR devices.

saqihussainkalan, 2025-04-16:
My joystick uses a 2.4GHz wireless USB dongle, which works via radio frequency (RF).

MartyG-RealSense, 2025-04-16:
Thanks for the information.  Please try adding `emitter_enabled:=false` to your launch instruction to disable the camera's IR emitter component, just to confirm whether or not the joystick dongle is being interfered with by the camera's IR projections.

saqihussainkalan, 2025-04-16:
i have added the parameters as said. However i want to let you know also this:

if i don't run these commands i am not able to control the robot. 

--Since agetty is managing /dev/ttyS0, i need to disable it. Run the following:---

sudo systemctl stop serial-getty@ttyS0.service

-- verify ---

sudo lsof /dev/ttyS0

Add robot user to the dialout group

sudo usermod -a -G dialout $USER

___Then Ensure Proper Permissions on /dev/ttyS0

sudo chmod 666 /dev/ttyS0

By running these commands the realsesnse launch file throws an error 

[image]

I first launch the realsense with the above command and then run the own custom node in another terminal. 

i don't know where i am doing wrong, if you need my rtabmap.launch.py file i can provide that as well.

MartyG-RealSense, 2025-04-16:
My knowledge of rtabmap is limited, unfortunately.  However, my research of **/dev/ttyS0** suggests that it should not be affecting the RealSense camera if disabled, as this is a serial device address involved in devices such as controller boards.

Do the errors still occur if you add **initial_reset:=true** to your rs_launch.py instruction to reset the camera at launch, please?

MartyG-RealSense, 2025-04-16:
If the lag is gone and the RealSense topics are working normally then it should be safe to ignore the log messages in the terminal, though I appreciate that it can be an annoyance and also make the reading of log files difficult.

saqihussainkalan, 2025-04-16:
Thank you. Just to inform you, when I run it on RViz, there’s no lag at all. However, when using RTAB-Map, there’s a significant amount of lag — even the poses are updated after quite a long delay.

MartyG-RealSense, 2025-04-16:
It is not unusual for RealSense users to report that rtabmap is performing slowly.  A user at https://github.com/introlab/rtabmap_ros/issues/358#issuecomment-545381531 shares their advice for increasing its speed.

saqihussainkalan, 2025-04-24:
Not now, if I need further assistance, I will open it again. for now i am closing it

MartyG-RealSense, 2025-04-24:
Okay, thanks very much for the update.  Please do open it again if you need further assistance.  Thanks again!

closed 2025-07-01 by saqihussainkalan

MartyG-RealSense, 2025-07-01:
You are very welcome.  I'm pleased to hear that it worked for you.  Thanks very much for the update!