realsenseai/librealsense #13857: libusb related errors for D435 cameras

AmosZJH, 2025-03-17:
Issue Description

I have been seeing some libusb errors like below:
libusb: debug [parse_endpoint] skipping descriptor 0x30

如果我能提供其他详细信息，请告诉我。 我只是想知道如何排除故障。

MartyG-RealSense, 2025-03-17:
Hi @AmosZJH  This is a very rare message that has only been reported as occurring when using multiple RealSense cameras.

Are you using more than one D435 camera simultaneously, please?  If you are, did you install the librealsense SDK from source code using the 'libuvc backend' or 'RSUSB backend' installation methods?  Also these methods can work with more than one camera, they are best suited to single-camera programs.

For multiple cameras, it is recommended to install the SDK from DKMS packages (see the link below for Jetson package instructions) or build the SDK from source code with the 'native backend' method.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

The native backend source code installation method involves not using the **-DFORCE_LIBUVC=TRUE** or **-DFORCE_RSUSB_BACKEND=TRUE** CMake build flags.

AmosZJH, 2025-03-21:
Hi @MartyG-RealSense , i have tried  the solution you mentioned https://github.com/IntelRealSense/realsense_mipi_platform_driver, but i failed over and over again, any other ways that i can build the SDK?

MartyG-RealSense, 2025-03-21:
Hi @AmosZJH  There is no need to use the MIPI driver instructions if you have the D435 camera model, as it only helps with RealSense camera models that have a motion-tracking IMU or a GMSL connector on them.  D435 has neither of these.

If you have attempted installing on Jetson from DKMS packages or source code using the methods mentioned at https://github.com/IntelRealSense/librealsense/issues/13857#issuecomment-2728800786 what failures did you experience, please?

Other than these two methods, the only other approach that would be suitable for multiple cameras is to attempt to install the Jetson version of the librealsense DKMS packages using the **installLibrealsense** build script provided by the JetsonHacks website at the link below.

https://github.com/JetsonHacksNano/installLibrealsense