realsenseai/librealsense #13859: Realsense network device post version 2.53

ArenaGrenade, 2025-03-17:
In the release notes of 2.54.1, it is specified that realsense net and rs server and associated components were removed in favour of a new system in the future:

> The network-device functionality has been removed in preparation for a replacement in the next release

None of the newer versions hint about this. @MartyG-RealSense are there any updates regarding this replacement? Is the only way to do this now a downgrade to librealsense 2.53.x?

MartyG-RealSense, 2025-03-17:
Hi @ArenaGrenade   There is nothing to announce at this time regarding networking support.  A downgrade to 2.53.1 is currently required to use the rs-server networking system.  

There is a Python-based alternative networking tool called **EtherSense** that remains part of the RealSense SDK though.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

ArenaGrenade, 2025-04-05:
Hi Sorry for the delay. I will test out EtherSense. Thanks for the update!

closed 2025-04-05 by ArenaGrenade