realsenseai/librealsense #13863: Realsense-viewer does not detect camera L515

nyevik, 2025-03-19:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Please provide a description of the problem
New out of the box camera(just been in the box for several years) L515. 
Decided to finally try it.
RealSence-viewer does not detect it, neither on Windows 11, not Ubuntu, nor Debian.

Please advise.

MartyG-RealSense, 2025-03-19:
Hi @nyevik  The L515 is now a retired camera model that is no longer supported in the RealSense SDK.  To use an L515 with the RealSense SDK or the RealSense Viewer tool, please install version **2.50.0**.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

MartyG-RealSense, 2025-03-31:
Hi @nyevik  Do you require further assistance with this case, please?  Thanks!

nyevik, 2025-03-31:
No, thank you.

MartyG-RealSense, 2025-03-31:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2025-03-31 by MartyG-RealSense