realsenseai/librealsense #13865: realsense-viewer: command not found

yslbit, 2025-03-19:
Hi,
I am trying to launch realsense-viewer in vanilla ubuntu 22.04, but it tell me: realsense-viewer: command not found

I installed RealSense SDK 2.0 following the instructions as described in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages 

best wish

MartyG-RealSense, 2025-03-19:
Hi @yslbit  The realsense-viewer tool should be on your computer if you have installed the **librealsense2-utils** DKMS package.

Can you confirm that you installed the librealsense2-utils package, please?

Can you also confirm if you have installed the packages on a PC or an Nvidia Jetson single-board computer?

MartyG-RealSense, 2025-03-31:
Hi @yslbit  Do you require further assistance with this case, please?  Thanks!

yslbit, 2025-03-31:
> Hi [@yslbit](https://github.com/yslbit) Do you require further assistance with this case, please? Thanks!

I am sry to reply u so late. I have been pay attention to others. Thanks for u patience. That is absolutely  great help. Thanks again

MartyG-RealSense, 2025-03-31:
It's no trouble at all.  You are very welcome.  Thanks very much for the update!

closed 2025-04-10 by MartyG-RealSense