realsenseai/librealsense #13867: synch realsense D415 with external devices

egleorl, 2025-03-19:
Hello! I haven't been able to find much information on this topic, but I was wondering what the best solutions are for synchronizing a RealSense D415 camera with the streams from other devices. Specifically, I am using Lab Streaming Layer to record different streams synchronously from multiple devices, such as EEG and ECG. At the same time, I am recording the scene using the depth camera on another PC. Now, I need to synchronize the EEG data with the RealSense data. How can I achieve this? Thank you!

MartyG-RealSense, 2025-03-19:
I believe that they solved the issue to their personal satisfaction using the **TIME_OF_ARRIVAL** metadata timestamp, though did not share their programming solution for doing so.

I am not aware of anyone else who used this method for syncing with a non-RealSense camera.

Using **ROS** may be an easier way to sync, as you can publish topics from different types of devices at the same time (for example, a RealSense camera and a LIDAR camera).  RealSense has a ROS compatibility wrapper for ROS1 and ROS2 that publishes ROS topics of the RealSense depth, RGB, infrared and IMU streams.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

**ROS2**
https://github.com/IntelRealSense/realsense-ros

egleorl, 2025-03-19:
Ok, I'll try, thanks!

MartyG-RealSense, 2025-03-31:
Hi @egleorl  Do you have an update about this case that you can provide, please?  Thanks!

egleorl, 2025-04-01:
Hi!
I'm currently experimenting with a new setup, recording directly using ambient cameras via LSL.

That said, my original plan with the RealSense was to send specific events or markers to the various data streams, events that would also be visible in the RealSense frames. Then, I would use the metadata — specifically the Time of Arrival — to align those events on the RealSense timeline with the other data streams timelines.