realsenseai/librealsense #13869: rtabmap on xavier nx with d435i

bluejade115, 2025-03-20:
my machine is Xavier nx noetic with JetPack5.1.12. For d435i, the firmware is 05.13.00.50，librealsense verison 
 is 2.50.0and realsense-ros wrapper version is 2.3.2。
When launch file is running, the error of 
**"Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set.
/rgbd_sync subscribed to (approx sync):
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info**"  occurs. I can not solve this error. 

Here is the log:
... logging to /home/ysc/.ros/log/bd6433dc-0572-11f0-a60b-48b02d87cdeb/roslaunch-lite-9598.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.103:35655/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 640
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 480
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: d435
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 159 lines cut ...]
    <include  file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="align_depth" value="true"/>
        <arg name="enable_pointcloud"  value="false"/>
        <arg name="enable_sync"  value="true"/>
        <arg name="enable_depth"  value="true"/>
        <arg name="depth_height"  value="480"/>
        <arg name="depth_width"  value="640"/>
        <arg name="color_width"  value="640"/>
        <arg name="color_height"  value="480"/>
        <arg name="color_fps"  value="30"/>
        <arg name="depth_fps"  value="30"/>
        <arg name="color_format"  value="RGB8"/>
        <arg name="serial_no" value="244222075300"/>
        <arg name="device_type" value="d435"/>
</include>

    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_sync/rgbd_sync camera_nodelet_manager" output="screen">
        <param name="sync_queue_size"  value="200"/>
        <param name="approx_sync"  value="true"/>
        <remap from="rgb/camera_info"  to="/camera/color/camera_info"/> 
        <remap from="depth/image"      to="/camera/aligned_depth_to_color/image_raw"/>
        <remap from="rgb/image"      to="/camera/color/image_raw"/>
        <param name="compressed_rate"  value="5"/>
    </node>  
</launch>

MartyG-RealSense, 2025-03-20:
Hi @bluejade115  May I first ask which Ubuntu version you are using.  JetPack 5 should be used with Ubuntu 20.04.  

Ubuntu 18.04 uses JP4, whilst Ubuntu 22.04 and 24.04 are suited to JP6.

bluejade115, 2025-03-21:
> Hi [@bluejade115](https://github.com/bluejade115) May I first ask which Ubuntu version you are using. JetPack 5 should be used with Ubuntu 20.04.
> 
> Ubuntu 18.04 uses JP4, whilst Ubuntu 22.04 and 24.04 are suited to JP6.

Thank to your rapid reply. I use ubuntu 20.04. when rtabmap is running, I also find CPU 100%.

MartyG-RealSense, 2025-03-21:
What happens if you use the **opensource_tracking.launch** RealSense ROS1 launch file, which should be able to be used with rtabmap without launching a separate rtabmap launch file after the RealSense one.

`roslaunch realsense2_camera opensource_tracking.launch`

Alternatively, when rtabmap is installed it has its own RealSense launch file called **rs_rtabmap.launch** that can be used to launch a RealSense camera instead of rs_camera.launch.  The contents of this launch file can be found at https://github.com/IntelRealSense/realsense-ros/issues/1653

`roslaunch realsense2_camera rs_rtabmap.launch`

bluejade115, 2025-03-21:
> What happens if you use the **opensource_tracking.launch** RealSense ROS1 launch file, which should be able to be used with rtabmap without launching a separate rtabmap launch file after the RealSense one.
> 
> `roslaunch realsense2_camera opensource_tracking.launch`
> 
> Alternatively, when rtabmap is installed it has its own RealSense launch file called **rs_rtabmap.launch** that can be used to launch a RealSense camera instead of rs_camera.launch. The contents of this launch file can be found at [IntelRealSense/realsense-ros#1653](https://github.com/IntelRealSense/realsense-ros/issues/1653)
> 
> `roslaunch realsense2_camera rs_rtabmap.launch`

I have used the opensource_tracking.launch and rs_rtabmap.launch before, but the same error occur. The error is `Did not receive data since 5 seconds! `. When this error occur, I check the relative topic, such as /camera/color/image_raw, the result is no message published from  realsense camera's node. And CPU usage always is 100%. Today, I add `export OMP_WAIT_POLICY=passive` to `~/.bashrc`. After that, I succeefully run the same code about rtabmap/sync https://github.com/IntelRealSense/librealsense/issues/13869#issue-2935581966  above. But I try to use rs_rtabmap.launch, the same erro r`Did not receive data since 5 seconds! ` occur  again.

MartyG-RealSense, 2025-03-22:
Does the problem still occur if you set **align_depth** to false?  Nvidia Jetsons have been known to have problems with the RealSense ROS1 wrapper when pointcloud or depth-color alignment is enabled.

A solution that has worked for some users is to launch with the **rs_rgbd.launch** launch file instead of rs_camera.  You can try this using the following procedure:

1.  Install support for an RGBD launch on ROS1 Noetic with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Use this launch instruction:

`roslaunch realsense2_camera rs_rgbd.launch`

With the rs_rgbd launch file, align_depth is already set to true by default.

bluejade115, 2025-03-22:
> Does the problem still occur if you set **align_depth** to false? Nvidia Jetsons have been known to have problems with the RealSense ROS1 wrapper when pointcloud or depth-color alignment is enabled.
> 
> A solution that has worked for some Jetson users when using pointclouds or alignment in the ROS1 wrapper is to launch with the **rs_rgbd.launch** launch file instead of rs_camera. You can try this using the following procedure:
> 
> 1. Install support for an RGBD launch on ROS1 Noetic with the command below.
> 
> `sudo apt-get install ros-noetic-rgbd-launch`
> 
> 2. Use this launch instruction:
> 
> `roslaunch realsense2_camera rs_rgbd.launch`
> 
> With the rs_rgbd launch file, align_depth is already set to true by default.

Thanks to your reply. When I set `align_depth=false`, it works. But I encounter a new problem that  is `The time difference between rgb and depth frames is high`. I set color_image(640,480) with 30fps, depth_image(640,480) with 15 fps. I have checked that camera is using USB3.2. I also try to use rtabmap_sync/rgbd_sync to decrease the time difference, but it do not work.

MartyG-RealSense, 2025-03-22:
When using rs_camera.launch, if you add **enable_sync:=true** to the launch instruction then the wrapper will attempt to send depth and color with the same timetag.

enable_sync is set to true by default in the rs_rgbd.launch file.

MartyG-RealSense, 2025-03-24:
It does not sound as though there is a sync problem between depth and color if the FPS of the streams is almost what you defined before rtabmap starts.

It is not unusual for RealSense ROS streams to be slow when used with rtabmap.  https://github.com/introlab/rtabmap_ros/issues/358 is an example of this.

Subscribing to compressed topics instead of raw topics to reduce bandwidth might help to improve performance, as described for rs_camera.launch and rtabmap.launch at the link below.

http://official-rtab-map-forum.206.s1.nabble.com/How-to-use-rtabmap-ros-data-throttle-for-Realsense-camera-tp5198p5213.html

Another user achieved improved performance if they set decimation to 2 instead of 1.

bluejade115, 2025-03-28:
> It does not sound as though there is a sync problem between depth and color if the FPS of the streams is almost what you defined before rtabmap starts.
> 
> It is not unusual for RealSense ROS streams to be slow when used with rtabmap. [introlab/rtabmap_ros#358](https://github.com/introlab/rtabmap_ros/issues/358) is an example of this.
> 
> Subscribing to compressed topics instead of raw topics to reduce bandwidth might help to improve performance, as described for rs_camera.launch and rtabmap.launch at the link below.
> 
> http://official-rtab-map-forum.206.s1.nabble.com/How-to-use-rtabmap-ros-data-throttle-for-Realsense-camera-tp5198p5213.html
> 
> Another user achieved improved performance if they set decimation to 2 instead of 1.

Thanks to your help, I solved those problems!!

MartyG-RealSense, 2025-03-28:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update!

closed 2025-04-10 by MartyG-RealSense