realsenseai/librealsense #13871: librealsense-2.38.1 | pyrealsense2 import fixing help

Mekan-Geldiyev, 2025-03-20:
I was following this guide: https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

and then on the step 5, it gives me this error like the source directory "/home/nvidia/librealsense-2.38.1/build/DPYTHON_EXECUTABLE=/usr/bin/python3.6" does not exist.

This is strange because /usr/bin/python3.6 DOES exist. I need help what do i do from here I just need to be able to import pyrealsense2

MartyG-RealSense, 2025-03-21:
Hi @Mekan-Geldiyev  You can confirm if the path that you are using on your particular computer is correct by inputting the command **which python3** into your terminal to get the path that should be used.

You do not need to use a librealsense version as old as 2.38.1, as the guide uses it only because it was written over four years ago and that was the SDK version being used by the guide's author at the time.  It would be fine to use librealsense **2.55.1**.

Mekan-Geldiyev, 2025-03-21:
I used this https://github.com/jetsonhacks/installRealSenseSDK/blob/master/buildLibrealsense.sh
to try and do it and it installed like the latest version which is 2.56.3. My device is detected  but it says the port is like busy or something. On top of that when i try some basic code I get the basic runtime error but none of the fixes that I've researched that you've recommended work. I need some serious help. I'm going to install and build librealsense 2.53.1 instead now but I need some serious help. I've tried deleting the build folder and rebuilding the version of librealsense that was built in to my Jetson Nano, then I tried to use the 2.38.1 version and build it but that also didn't work. This is the code that doesn't work:

class Camera:
    # Class handles Camera object instantiation and data requests.
    def __init__(self):
        self.pipeline = rs.pipeline()  # Initialize RealSense pipeline
        self.config = rs.config()
        # Enable depth stream at 640x480 in z16 encoding at 30fps
        self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        # Enable color stream at 640x480 in rgb8 encoding at 30fps
        self.config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

    def start(self):
        self.profile = self.pipeline.start(self.config)  # Start the pipeline
        # Obtain depth sensor and calculate depth scale
        depth_sensor = self.profile.get_device().first_depth_sensor()
        self.depth_scale = depth_sensor.get_depth_scale()

    def get_frames(self):
        return self.pipeline.wait_for_frames()  # Wait and fetch frames from the pipeline

    def stop(self):
        self.pipeline.stop()  # Stop the pipeline when finished

This line over here is what gets the problem: self.profile = self.pipeline.start(self.config)  # Start the pipeline

[image]
[image]

MartyG-RealSense, 2025-03-22:
Updating to CMake 3.13.0 is fine.  That should not be the cause of your problems.

In the realsense-viewer log I note that the top line says "Make sure to remove redundancies!". 

[image]

If this message is present then you likely have a `Multiple realsense udev-rules were found!` warning.  This can cause problems for the camera's operation, as the **udev rules** are related to handling of the camera device.  The multiple udev rules issue can occur if librealsense has been installed from both DKMS packages and source code on the same computer.

The solution for this issue is simple though.  The message should disappear if the rule below is deleted from the computer.

`/etc/udev/rules.d/99-realsense-libusb.rules`

The other type of rule that is installed by librealsense is `/lib/udev/rules.d/60-librealsense2-udev-rules.rules` - this rule should **not** be deleted.

closed 2025-04-10 by MartyG-RealSense