realsenseai/librealsense #13881: Unexpected behaviour when using multi cam with harware sync mode 2

jeferal, 2025-03-25:
I have 5 cameras D455 running in Ubuntu with a real time kernel.

Firmware version: FW 16.0.1
Librealsense version: 2.55.1

When running the official realsense-ros wrapper with 5 cameras using one camera as master (inter cam sync mode 1) and 4 cameras as slaves (inter cam sync mode 2) everything works well (I am monitoring the frame counters and checking that the difference between frames is always 1).

The configuration has Autoexposure ON, Autoexposure limit 8333 and Emitter always ON. 15 FPS for stereo and infra.

However, when I start another application that interacts with an Ethercat Bus at ~1Khz, I am seeing many frame dropps (frame diffs up to 40 frames), broken frames, frames with emitter off, etc.
The driver also reports these errors:
```
Hardware Notification:Right MIPI error,1.74289e+12,Error,Hardware Error
Hardware Notification:REC error,1.74289e+12,Error,Hardware Error
Asic Temperature value is not valid!
```

On the other hand, if I use intra sync mode 4 (Gen-lock if I am not wrong), the problem that I am reporting does not happen when the second Ethercat application is running. I can only see broken frames every now and then and some dropped frames and the driver does not report the above hardware notifications.

**I know that the gen lock mode is not supported, but I want that since these two modes (inter cam sync mode 2 and inter cam sync mode 4) do not seem to be so different at least from an user perspective, why do I find this enormous difference in their behaviours.**

Thank you very much in advance for your help.

MartyG-RealSense, 2025-03-25:
Hi @jeferal  Genlock mode is no longer supported with updates or technical support but it is still accessible in the RealSense SDK.

The key difference between modes 2 and 4 is that with mode 2 (Slave), the SDK listens for a sync trigger pulse for a certain amount of time on each frame and then if it does not receive a trigger within that time frame, it proceeds with capturing the frame without syncing it.

With mode 4, it listens for a trigger pulse indefinitely and will not proceed with capture until the trigger is received.

Does the problem still occur if you use mode 3?  This is a Genlock mode that works like mode 2 (i.e it does not wait indefinitely like mode 4 does) and is called Full Slave.

I note that you are using a real time (RT) kernel.  In the past this has caused complications on the rare occasions when they have been used with RealSense.  This is referred to at https://github.com/IntelRealSense/librealsense/issues/1490 and https://github.com/IntelRealSense/librealsense/issues/2395#issuecomment-426536906

jeferal, 2025-03-25:
Thank you very much @MartyG-RealSense. I also tried with inter cam sync mode 3 and I notice that there are no USB disconnections when I check dmesg, but in the infra stream I notice the following similar to when I use sync mode 2:
- Broken frames
- Frames with wrong exposure
- Frame drops
- Very bright frames
- Very dark frames

Has the autoexposure algorithm changed recently when using a sync mode? As in, is the autoexposure algorithm running in the camera or in the computer and the command is sent in each step?

I will check the comments on the Real Time Kernel, thank you for the information.

MartyG-RealSense, 2025-03-25:
The only recent changes to auto-exposure were in SDK version 2.55.1 (the one that you are using), where the **Auto Exposure Limit** and **Auto Gain Limit** functions were introduced back into the SDK after being missing since version 2.53.1.

[image]

If the camera is being used in a location with a consistent level of artificial lighting that does not change much then you could try disabling auto-exposure so that a fixed exposure value is used.

jeferal, 2025-03-25:
Thank you again @MartyG-RealSense for your help. Unfortunately disabling auto-exposure is not an option. The requirements for my use case are:
- Emitter always On
- Cameras with hardware sync
- Autoexposure enabled

Are there any past issues that relate broken frames, frame drops, very bright frames, very dark frames when using inter cam sync mode 2 or 3?
What could be potential reasons that could cause problems like these?

jeferal, 2025-03-25:
Thank you for the explanation on the emitter always on.

I tried hardware sync 2 and 3 with combinations of emitter always on enabled and disabled and the problem still happens. Where I notice frame drops, broken frames, very bright/dark frames. The problem also happens with autoexposure disabled.

jeferal, 2025-03-26:
Thank you @MartyG-RealSense for the response.

That is correct:
I do not transfer data to the cameras over Ethercat, but the Ethercat application seems to have a negative effect on the cameras simply by being active.

Yes, that is correct, the cameras are able to sync normally when mode 4 is used and also when I do not use any hardware synchronization mode. Without hardware sync, they work perfectly.

If mode 4 seems to work well together with the ethercat application and also the no hardware sync mode, why am I experiencing USB disconnections with hardware sync mode 2 in particular and not with the others? Is there anything different in this particular hardware sync2?

MartyG-RealSense, 2025-03-26:
Mode 2 will likely be capturing frames at a much more rapid rate than Mode 4, as it performs frame capture even if a trigger signal is not received, while Mode 4 will only capture when it recognizes a trigger.  So Mode 4's slower frequency of capture may provide better stability in the face of however the Ethercat bus is negatively affecting the RealSense cameras.

There is not enough information about use of Ethercat with RealSense to form a solid theory about why it is occurring.

jeferal, 2025-03-26:
With mode 2, I set the working frequencies of the slave cameras at 15fps, the same frequency as the master camera. When the ethercat application is not running, I obtain 15fps that are stable from each camera. I tested that the sync works by pointing two cameras to a watch.

If the sync works and the 15 fps are stable for both slaves and master cameras, does this mean that the cameras are only producing frames when the trigger is received? There should not be frames that belong to no trigger signal, because the trigger is always received on time with respect to the working frequency of the camera, is this correct?

My point is that, from a user perspective, I do not see the overhead that inter cam sync mode 2 is adding with respect to mode 4, since both seem to be working at 15FPS.

MartyG-RealSense, 2025-03-27:
In mode 2, the slave camera will produce frames continuously no matter what (in this case, at a rate of 15 frames per second from each of the five D455 cameras simultaneously).  Whether those frames are synced to the master depends on whether the trigger was recognized in time (because mode 2 listens for the trigger for a certain time period on each frame and then gives up listening and performs an unsynchronized capture of that particular frame if the trigger wasn't received in time).

While you can observe over the course of a number of minutes whether the slave timestamps are syncing to the master, I am not aware of a mechanism to tell whether a specific frame synced to the trigger or not.

MartyG-RealSense, 2025-03-31:
The cause of the Right MIPI Error is very difficult to diagnose and its appearance is often equivalent to a vague "something went wrong message".

The REC error, which is typically associated with an issue with USB data transmission, has been resolved by some RealSense users in the past by changing the FPS speed or the resolution.

In case you have tried having different FPS speeds for master and slave on mode 2, it is not necessary to do so as that is only required with genlock mode 4.  With mode 2, both master and slave should have the same FPS.

jeferal, 2025-04-01:
Thank you for the information. Today I will then try with different combinations on FPS and resolution with same FPS for both master and slave in mode 2 as you mention, thank you.

On the other hand, I am carrying a sanity check on the signal generated by the master camera. I set a camera to be the master and I measured the signal that arrives to a slave camera. I would like to understand if the noise that is measured in the signal is enough to consider this signal as corrupted: e.g. drop frames or start capturing when there is no trigger. Could you please send me the link to the specifications of how the trigger signal should be?

[image]

[image]

MartyG-RealSense, 2025-04-01:
In mode 2 a capture will take place no matter what, even if a trigger is missed.

In mode 4 a capture will not take place until a trigger is received.  

When using a trigger from a master camera that is set to mode 1, the trigger pulse's width should be 50 us (microseconds).  The voltage should be 1.8 V (the voltage from pin 5, the Sync Pin).

A case at https://github.com/IntelRealSense/librealsense/issues/13407#issuecomment-2401861878 that discusses possible low voltage on their trigger pulse eventually achieved the master camera-generated trigger pulse of 50 us and 1.73 V shown below.

[image]

jeferal, 2025-04-02:
Thank you for the information on the trigger signal. I have a few questions now:
* What are the differences between using Inter cam sync mode 1 and using output trigger enabled 1? I do not find this explanation enough in this comment https://github.com/IntelRealSense/librealsense/issues/9810#issuecomment-932152876. I want to know for both options exactly in what event the trigger is supposed to happen.
* What is the timeout in hardware sync 2 (the amount of time that a camera would wait for a trigger signal)? Is this information in any official documentation?

Thank you very much for your help.

MartyG-RealSense, 2025-04-10:
Hi @jeferal  Do you require further assistance with this case, please?  Thanks!

jeferal, 2025-04-16:
Yes. Thank you.

In a configuration where there is no noise in the synchronization signal and I am using the supported syncronization mode (**Hardware sync 2**) with another camera as the master camera (Hardware sync 1) I have found a broken frame in the infrarred stream.

It is critical to my application not to have broken frames like this.

[image]

**Why do these broken frames appear in the hardware sync mode 2?**

MartyG-RealSense, 2025-04-16:
In single-camera applications whose pipeline uses the wait_for_frames() instruction, if a bad frame occurs then the pipeline jumps back to the last known good frame, repeats it and continues on from that point.  So in theory, broken frames should not occur in applications that are using wait_for_frames().

In multiple camera applications whose pipeline uses the poll_for_frames() instruction instead, it does not have this auto-recovery jump back mechanism.

Also, frame generation is a bit more unpredictable in mode 2 than it is in mode 4 in regards to how mode 2 will initiate a capture on every frame no matter what, whilst mode 4 will wait indefinitely for a trigger and so has more 'time to think', which may reduce the chance of a bad frame occurring.

As you are using the ROS wrapper, you could try disabling the **Global Time** camera time synchronization mechanism to see if broken frames reduce in mode 2, as ROS is not as dependent on it as librealsense is because ROS has its own built-in timing mechanism.  You can test the disabling of Global Time by adding **global_time_enabled:=false** to your ROS launch instruction.

Alternatively, you could try reducing the FPS in mode 2 from 15 to 5 to see if broken frames cease.

jeferal, 2025-04-17:
Thank you very much for the information on broken frames,  today I will try using Global Time and reducing the FPS.

I some have questions regarding the broken frames:
1. Where is implemented in the librealsense SDK the 'auto-recovery' when using wait_for_frames()? Can you send me links to the code?.
2. Why is the auto-recovery not implemented in multiple camera applications which use the poll_for_frames()?
3. Is the broken frame generated inside the camera? What exactly causes it?

I sadly found more likely to have more broken frames when I am using a gen lock mode (inter cam sync 4) and that's why I moved to a "supported" mode such as inter cam sync mode 2, expecting not to have problems like these.

MartyG-RealSense, 2025-04-17:
1.  The 'last good frame' jump-back mechanism is first referred to in an old comment at https://github.com/IntelRealSense/librealsense/issues/1686#issuecomment-388606174 but there is not information available regarding where and how it is implemented within the SDK source code, unfortunately.

2.  https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709 explains that the 'block until there is a complete frame' mechanism of wait_for_frames() would be problematic for a multiple camera setup, which is why poll_for_frames() - which processes frames instantly without any waiting to check if it is a complete frame - is recommended for multicam applications instead.

3.  It is highly unusual to experience an infrared frame like the one in the image above.  Typically, that kind of effect (where there is an incorrect section off to the side of the image) is more likely to occur on the depth stream if there is a problem when the camera hardware is combining the pair of left and right infrared frames into a single depth frame.

Are all of the 5 cameras attached to the same computer and active simultaneously?  If they are, do you have a computer with a specification that can handle that many active cameras?  For 4 cameras or more, a computer with an Intel Core i7 or i9 processor or equivalent capability is recommended, because each active camera that is added to the computer consumes a proportion of its resources.

Do broken frames still occur if the Ethercat bus is not being used?

closed 2025-05-01 by MartyG-RealSense