realsenseai/librealsense #13883: Camera detected but not working

axel-kaliff, 2025-03-25:
Camera Model =D400
Firmware Version =5.16.0.1
Operating System & Version 	Linux (Debian 12 Bookworm)
Kernel Version (Linux Only) 	6.1.21-v8+
Librealsense SDK Version 	{ 2.56.1 }
USB Type = 3.2
Advanced Mode = YES

Hello, 

I'm trying to set up my  D400 RGBD camera on my headless raspberry pi (4b) running debian 12 bookworm.

I installed the SDK according to the RS-backend method found in [this](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692) issue.

The camera is detected, `rs-enumerate-devices` runs as normal, but when I run ` rs-record -f ./test1.bag -t 60` I get:

> ERROR [547882724256] (rs.cpp:255) [rs2_pipeline_start_with_config_and_callback_cpp( pipe:0x55868e4850, config:0x55868e4960, callback:0x55868f9820 ) Invalid Value] hwmon command 0x2c( 9 0 0 0 ) failed (response-9= No expected user action)
>  ERROR [547882724256] (rs.cpp:255) [rs2_delete_pipeline_profile( profile:nullptr ) UNKNOWN] null pointer passed for argument "profile"
> RealSense error calling rs2_pipeline_start_with_config_and_callback_cpp(pipe:0x55868e4850, config:0x55868e4960, callback:0x55868f9820):
>     hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

Kernel log:
>  uvcvideo 2-1:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71

I've been wrecking my brain with this problem for days now, trying every installation method I can find. 

Help would be much appreciated!

axel-kaliff, 2025-03-25:
Note: the libuvc method gave the same results

MartyG-RealSense, 2025-03-25:
Hi @axel-kaliff  Does it make a difference if you simply use the command **rs-record** so that it saves the file with the default name of 'test.bag' with a default recording length of 10 seconds?

MartyG-RealSense, 2025-03-25:
I note that you used the RSUSB installation method at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

There can be problems on a headless system if the CMake build flag **-DBUILD_GRAPHICAL_EXAMPLES=true** is used, as this flag installs not only graphics-based tools and examples but also support for graphics.  But on a headless system this can cause errors to occur because a display is not present, so -DBUILD_GRAPHICAL_EXAMPLES should be set to **false** in the CMake build instruction so that graphics support is not included in the SDK build.  

Plain-text tools such as rs-enumerate-devices will still be installed because **DBUILD_EXAMPLES=true**

axel-kaliff, 2025-03-26:
Same results, after doing the following:

I removed and remade the build directory, I ran: ` cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false`

I then ran the `make` command according to the instructions.

Don't know the relevance, but I got this warning after the `make` command:

> 
> CMake Warning at cmake_uninstall.cmake:2 (MESSAGE):
>   Cannot find install manifest:
>   "/home/cloud/librealsense-2.56.1/build/install_manifest.txt"

axel-kaliff, 2025-03-26:
Could it have something to do with previous installation attempts, like old udev rules that might be causing trouble?

axel-kaliff, 2025-03-26:
This solved it; now rs-record works properly. Many thanks, @MartyG-RealSense !!

MartyG-RealSense, 2025-03-26:
That's great news, @axel-kaliff - thanks very much for the update about your success!

closed 2025-04-25 by MartyG-RealSense