realsenseai/librealsense #13886: RealSense D450 Stuck in DFU Mode After Firmware Update

davidtr99, 2025-03-26:
We have recently purchased a RealSense D450 module (K83121-101), an Intel D4 vision board, and a flex cable for connection.

Initially, we installed the latest version of the SDK, but the RealSense Viewer was unable to detect the camera. On Windows 11, when connecting the device via a USB 3.2 port, a pop-up appeared stating that the device was not recognized properly. We tested another RealSense camera (D435) using the same port, USB cable, and software, and it worked without issues. So, we assumed that the problem was with this particular device.

Following the advice from issue #11621, we disconnected the camera module and updated the firmware of the vision board using the FW update utility. The firmware was updated from version 5.12.7.150 to 5.16.0.1, the latest available for the D450 series. The previous version (5.12.7) is marked as unsupported for D45x devices in the [firmware-releases-d400](https://dev.intelrealsense.com/docs/firmware-releases-d400) documentation, which likely explains why the device was not being recognized.

After successfully installing the new firmware, the device was detected by both the PC and the RealSense Viewer. However, the viewer identifies it as "Intel RealSense D4XX Recovery" and prompts for a firmware update to the same version already installed (5.16.0.1).

[image]

When attempting to update the firmware with the camera module attached—whether via the Viewer or the FW updater utility—the process gets stuck after installation, stating that the device is in DFU mode and is unlocked.

[image]

Could you provide guidance on how to resolve this issue? Are there any available workarounds?

MartyG-RealSense, 2025-03-26:
Hi @davidtr99  You can attempt to recover a camera in recovery mode in the rs-fw-update firmware update tool by adding the -r flag to the tool's launch instruction.  For example for the 5.16.0.1 firmware .bin file:

`rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin`

davidtr99, 2025-03-26:
Hi @MartyG-RealSense! Thanks for your prompt reply.
After the update, the process gets stuck and shows "timed out!" after a few seconds.

[image]

MartyG-RealSense, 2025-03-26:
It looks as though the firmware update in recovery mode was successfully completed.  What name is the camera now identified as in the RealSense Viewer, please?

davidtr99, 2025-03-27:
When opening the RealSense Viewer, the device still appears as "Intel RealSense D4XX Recovery". Nothing has changed.

[image]

Additionally, when listing connected devices in the FW updater utility, it also appears to still be in recovery.

[image]

MartyG-RealSense, 2025-03-27:
Can you confirm which version of the Vision Processor D4 board you are using, please?  The D450 module will only work with the V3 and V5 versions of the D4 board.  Connecting D450 to V3 turns it into a D455, whilst connecting it to V5 turns it into a D457.

davidtr99, 2025-03-27:
I believe the exact version of the vision board is simply "D4 Board". You can see the specific details in the following images:

[image]

[image]

So, there is a compatibility issue between the D450 module and this version of the vision board?

davidtr99, 2025-03-28:
Thanks for the clarification. Now it's clear what the problem was. I appreciate your help in resolving this. I'll go ahead and close this issue.

closed 2025-03-28 by davidtr99