realsenseai/librealsense #13891: Error type: Access Violation!

Vaishali-1602, 2025-03-28:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2025-03-28:
Hi @Vaishali-1602  Are you trying to open a bag file in the RealSense Viewer tool when this error occurs?  

If you are not opening a bag file in the Viewer then please check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Vaishali-1602, 2025-04-10:
> Hi @Vaishali-1602<https://github.com/Vaishali-1602> Do you require further assistance with this case, please? Thanks!
>
>
> Hi @Vaishali-1602<https://github.com/Vaishali-1602> Do you require further assistance with this case, please? Thanks!

Hi ,

Thanks for reaching out. No assistance required at this time.

Thanks

Company Use

MartyG-RealSense, 2025-04-10:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance at this time, I will close this case.  Thanks again!

closed 2025-04-10 by MartyG-RealSense