realsenseai/librealsense #13892: Realsense device is not being detected in Nvidia Jetson Orin Nano Dev kit

Anto-Merary, 2025-03-29:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|

Please provide a description of the problem
I was trying to connect the intel realsense D435i I tried everything like givin all permission changin udev rules and nothing working

closed 2025-04-17 by MartyG-RealSense

MartyG-RealSense, 2025-06-21:
Hi @copumpkin  JetPack 6 removed a component called **hiddraw** that the RealSense cameras equipped with an IMU depended upon (such cameras are called HID devices).  Non-IMU camera models such as D415, D435 / D435f and D405 are not affected.

If you have an Orin then the only alternative to compiling the libuvc backend (or downgrading JetPack from JP6 to JP 5.1.3 or using a non-IMU camera model) is a user-installable driver that was designed for RealSense D457 cameras on a non-USB (GMSL) cable connection but also provides support to IMU-equipped RealSense cameras attached by USB to Jetson Xavier or Orin using JetPack 6.  It is complicated to install though.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

copumpkin, 2025-06-21:
Thanks @MartyG-RealSense! That's unfortunate but thank you for the added information! It seems like such a change would be inadvertent given how common it is for people to pair RealSense cameras with Jetson hardware. Do you know if NVidia has an open issue to add that `hiddraw` component back to the next JetPack release, to avoid headaches for their users in future? If not, I might try to open one and see if the removal was deliberate or if they simply didn't realize that it would cause downstream headaches like this.

In the mean time I'll try the driver you suggested, thank you!

MartyG-RealSense, 2025-09-24:
Hi @Perzival07  Any USB cable will be fine as long as it meets the following requirements:

-  It is USB Type C / A to C (a full size USB connector on one end and a micro-sized connector on the other end)
- The cable is a high quality one designed for data transfer and not an inexpensive cable intended only for device charging

Examples of RealSense-suitable cables can be found at the link below.

https://newnex.com/realsense-3d-camera-connectivity.php

By default a D457 uses a **GMSL / FAKRA** cable and will not work with a USB cable.  To set the camera to be detected on a USB cable you need to move a hidden switch sidewards using these instructions:

https://support.realsenseai.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

The camera will then be detected as a **D455**.

Perzival07, 2025-09-24:
Thank you very much. The camera worked briefly, but then it prompted me for a D455 firmware update. I went ahead with the update, but since then the D457 is no longer being detected. Could you please advise me on what steps to take next?