realsenseai/librealsense #13896: Segmentation Fault Issue in rs2::align::process

sinuku, 2025-03-31:
The environment in which I am running is as follows:
Ubuntu 22.04
C++20, g++11
librealsense 2.56.3

A segmentation fault intermittently occurs when performing rs2::align::process().
Below is the gdb backtrace of the crash:
```
(gdb) bt
#0  0x00007eddf604c6d9 in void std::vector<std::shared_ptr<rsutils::lazy<rs2_extrinsics> >, std::allocator<std::shared_ptr<rsutils::lazy<rs2_extrinsics> > > >::_M_realloc_insert<std::shared_ptr<rsutils::lazy<rs2_extrinsics> > const&>(__gnu_cxx::__normal_iterator<std::shared_ptr<rsutils::lazy<rs2_extrinsics> >*, std::vector<std::shared_ptr<rsutils::lazy<rs2_extrinsics> >, std::allocator<std::shared_ptr<rsutils::lazy<rs2_extrinsics> > > > >, std::shared_ptr<rsutils::lazy<rs2_extrinsics> > const&) () from /usr/local/lib/librealsense2.so.2.56
#1  0x00007eddf6264e6a in librealsense::extrinsics_graph::register_extrinsics(librealsense::stream_interface const&, librealsense::stream_interface const&, rs2_extrinsics) () from /usr/local/lib/librealsense2.so.2.56
#2  0x00007eddf628d9dc in rs2_register_extrinsics () from /usr/local/lib/librealsense2.so.2.56
#3  0x00007eddf60c498e in librealsense::align::create_aligned_profile(rs2::video_stream_profile&, rs2::video_stream_profile&) () from /usr/local/lib/librealsense2.so.2.56
#4  0x00007eddf60c4ecd in librealsense::align::allocate_aligned_frame(rs2::frame_source const&, rs2::video_frame const&, rs2::video_frame const&) () from /usr/local/lib/librealsense2.so.2.56
#5  0x00007eddf60c5637 in librealsense::align::process_frame(rs2::frame_source const&, rs2::frame const&) () from /usr/local/lib/librealsense2.so.2.56
#6  0x00007eddf60de2e6 in rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::{lambda(rs2::frame, rs2::frame_source const&)#1}>::on_frame(rs2_frame*, rs2_source*) () from /usr/local/lib/librealsense2.so.2.56
#7  0x00007eddf60d991c in librealsense::processing_block::invoke(librealsense::frame_holder) () from /usr/local/lib/librealsense2.so.2.56
#8  0x00007eddf6292a06 in rs2_process_frame () from /usr/local/lib/librealsense2.so.2.56
#9  0x000058b660244e9e in rs2::processing_block::invoke (this=0x58b6608db828, f=...)
    at /usr/local/include/librealsense2/hpp/rs_processing.hpp:291
#10 0x000058b660245090 in rs2::filter::process (this=0x58b6608db828, frame=...)
    at /usr/local/include/librealsense2/hpp/rs_processing.hpp:372
#11 0x000058b6603260a6 in rs2::align::process (this=0x58b6608db828, frames=...)
    at /usr/local/include/librealsense2/hpp/rs_processing.hpp:732
#12 0x000058b66031ea29 in grabber::base::RealSense::getData (this=0x58b6608db818, frameset=..., color=..., depth=..., 
    point_cloud=std::shared_ptr<open3d::geometry::PointCloud> (empty) = {...})
    at /home/rvsuser/rvs/src/grabber/base/rs_sensor.cpp:89
```

grabber::base::RealSense::getData is a function I defined, and it calls rs2::align::process() inside.
Here is the implementation of getData:
```
bool RealSense::getData(const rs2::frameset& frameset, cv::Mat& color, cv::Mat& depth,
                        shared_ptr<geometry::PointCloud>& point_cloud) {
    try {
        auto aligned = m_align.process(frameset);

        auto color_frame = aligned.get_color_frame();
        auto depth_frame = aligned.get_depth_frame();

        ...
    } catch (const std::exception& e) {
        LOG_ERROR("Exception during getData: {}", e.what());
        return false;
    }
}
```

The member variable m_align is defined as follows:
`rs2::align m_align{RS2_STREAM_COLOR};`

The overall logic, including capture, looks like this:
```
rs2::frameset fs;
Mat color, depth;
shared_ptr<geometry::PointCloud> pcd;
if (cam.capture(fs, 2*FRAME_TIME_MS)) {
    success = cam.getData(fs, color, depth, pcd);
}

bool RealSense::capture(rs2::frameset& frameset, int timeout_ms) const {
    return m_pipe.try_wait_for_frames(&frameset, timeout_ms);
}
```

What could be the problem?
I have added exception handling inside the getData function as shown below, but the condition is not being met:
```
if (!frameset) {
    LOG_ERROR("Invalid frameset provided to process");
    return false;
}

if (frameset.size() == 0) {
    LOG_ERROR("Empty frameset provided to process");
    return false;
}
```

MartyG-RealSense, 2025-03-31:
Hi @sinuku  This appears to be a very rarely occurring issue with align_process.

Your problem sounds similar to a C++ segfault case at https://github.com/IntelRealSense/librealsense/issues/4469

Some users in this case thought that it might be related to running their application in Debug mode, whilst another worked around it by using C code in their C++ application.

In another C++ alignment segfault case at https://github.com/IntelRealSense/librealsense/issues/2567 where one user fixed it by using BGRA* or RGBA8 as their color stream format, whilst that fix did not work for another user and they found that the issue seemed to be occurring if their application linked to a non-RealSense library.

Others found that the problem occurred in a specific librealsense version and using an older version worked for them.

So there is not a single, clear solution in these linked-to cases, unfortunately.  Do any of the suggested causes seem relevant to your own project, please?

sinuku, 2025-04-01:
Hi @MartyG-RealSense  Thank you for your response.

The color stream is using RS2_FORMAT_BGR8.
OpenVINO is not being used.
CMake was built with CMAKE_BUILD_TYPE=Debug, and the compile options are as follows:
```
target_compile_options(${PROJECT_NAME} PRIVATE
    -march=native
    $<$<CONFIG:Debug>:-g -O0>
)
```

Looking at the `librealsense` source code, it seems that a segmentation fault occurs at `_external_extrinsics.push_back(lazy_extr);`.
Do you think adding `std::lock_guard<std::mutex> lock(_mutex);` inside the `extrinsics_graph::register_extrinsics` function would resolve the issue?
If so, it seems that `_mutex` should be changed from `std::mutex` to `std::recursive_mutex`.
```
void extrinsics_graph::register_extrinsics(const stream_interface & from, const stream_interface & to, rs2_extrinsics extr)
{
    auto lazy_extr = std::make_shared< rsutils::lazy< rs2_extrinsics > >( [=]() { return extr; } );
    _external_extrinsics.push_back(lazy_extr);
    register_extrinsics(from, to, lazy_extr);
}
```

MartyG-RealSense, 2025-04-01:
Would it be possible for you to try setting  CMAKE_BUILD_TYPE to **Release** instead of Debug?

I do not have knowledge of mutex programming and so do not have advice to offer for that particular question.  I do apologize.

sinuku, 2025-04-04:
@MartyG-RealSense 
I'm using the librealsense library built with `-DCMAKE_BUILD_TYPE=Release`

The build type I mentioned earlier referred to my own project, which is currently under development.

I believe that this issue should not occur regardless of the build type of my project.

MartyG-RealSense, 2025-04-04:
If the problem is not being caught by **try_wait_for_frames()** (which should return a 'false' state instead of allowing a program to exit) or by the try-catch exception catching mechanism then this suggests that the segmentation error is being caused by an issue that is not an exception error or a time-out of the pipeline due to frames not arriving.

Does the problem still occur if you add the **keep()** instruction below the 'auto' lines to store frames in the computer's memory?

```
auto color_frame = aligned.get_color_frame();
auto depth_frame = aligned.get_depth_frame();

color_frame.keep();
depth_frame.keep();
```

Also, does a segmentation error occur if you run the SDK's **rs-align** C++ example alignment program, if you included the SDK examples in the librealsense build when you installed it?

MartyG-RealSense, 2025-04-10:
Hi @sinuku  Do you require further assistance with this case, please?  Thanks!

sinuku, 2025-04-15:
@MartyG-RealSense 
Both methods you suggested result in the same issue.
It doesn't seem to solve the problem :(

MartyG-RealSense, 2025-04-15:
I do not have further suggestions to offer unfortunately, other than your own thought at https://github.com/IntelRealSense/librealsense/issues/13896#issuecomment-2767692630 about testing whether modifying the mutex instruction of the SDK source code at **/src/environment.cpp** and then building the SDK from the modified source coderesolves the segmentation fault.

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/src/environment.cpp#L75

sinuku, 2025-04-28:
@MartyG-RealSense 
I am currently editing and testing it myself. I will let you know if anything comes up. Thanks!

MartyG-RealSense, 2025-05-17:
Hi @sinuku  Have you been able to make progress with your issue after editing and testing, please?  Thanks!

closed 2025-05-30 by MartyG-RealSense