realsenseai/librealsense #13898: Can not take Y12I image for OEM calibration with pyrealsense2

bwolleb, 2025-03-31:
Hi,

I am trying to capture a frame with a Realsense D455 in the Y12I format using pyrealsense2 but it fails with a `RuntimeError: Couldn't resolve requests` error:

```
import pyrealsense2
config = pyrealsense2.config()
#config.enable_stream(pyrealsense2.stream.color, 1280, 800, pyrealsense2.format.bgr8, 15) # Works
config.enable_stream(pyrealsense2.stream.color, 1280, 800, pyrealsense2.format.y12i, 15) # Doesn't work
pipeline = pyrealsense2.pipeline()
pipeline.start(config)
```

I need that format in order to use the offline OEM calibration tool, but I am not sure if it's on the `color` stream (I am trying to create the `trackingColorImagex.bin`) or some other as the [documentation](https://www.intelrealsense.com/wp-content/uploads/2020/06/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf) seems to indicate that it's on the `infrared` stream (I also tried it and it fails the same). This format is not reported either when running `rs-enumerate-devices`.

Do you have any pointer on what I should try next?

System:
OS: Debian
Arch: x86_64
Python: 3.8.10
pyrealsense2: 2.55.1.6486 (Pypi's latest)
Sensor: RealSense D455 (USB 3.2)

MartyG-RealSense, 2025-03-31:
Hi @bwolleb  The undistorted RealSense format used for camera calibration is infrared Y16.

[image]

`config.enable_stream(pyrealsense2.stream.infrared, 1280, 800, pyrealsense2.format.y16, 15)`

bwolleb, 2025-03-31:
Ok thanks, this definitely works! To my inderstanding, the output of this can create the `trackingColorImagex.bin` used by the tool, right (a bit misleading to use the `infrared` stream)? Then what configuration is used for the `trackingImagex.bin` ?

MartyG-RealSense, 2025-03-31:
If you mean the OEM Calibration Tool supplied with Intel's $1500 USD OEM Calibration Target Board product then I do not have any information about the workings of the OEM version of the calibration tool or the mentioned .bin files, unfortunately.

bwolleb, 2025-03-31:
Yes I am using that target board, sorry this was a bit out of scope of this issue. Thanks a lot!

closed 2025-03-31 by bwolleb