realsenseai/librealsense #13899: RuntimeError: No device connected while running python file, but realsense-viewer works

Ischavank, 2025-03-31:
I'm running ubuntu 20.04.06 LTS using WSL

I'm using a D435 model and it seems like it is working because the realsense-viewer shows working live stereo and rgb feeds.
The problem is when I run a simple python script, I get this message:
  File "main.py", line 3, in <module>                                                                       profile = pipe.start()                                                                              RuntimeError: No device connected  

I already used sudo, like `sudo python3 main.py` while changing the line `import pyrealsense2 as rs` to `import pyrealsense2.pyrealsense2 as rs` but this also didn't work.

I'm not that accustomed with IntelRealSense, but I installed the library pyrealsense2 using pip install which seems ok, but I still have this error so all help with this would be greatly appreciated!

Ischavank, 2025-04-01:
Thank you for the quick response! I'm afraid that all three of these options didn't work, but I thought about removing the parts connected to the pyrealsense2 library and doing that part of the installation again, while hopefully keeping the realsense-viewer working, could you give me a hand with what I should remove and what steps I should take to do this?

Another problem could be this: 

[image]

I get these two separate errors, which don't effect the live feed, but could cause the python code to not be able to connect, would you have any idea how to solve this and if it matters?

MartyG-RealSense, 2025-04-01:
The `Multiple udev rules were found` message can cause problems with the camera's option.  These rules govern the handling of RealSense devices.  It typically happens if librealsense has been installed from both DKMS packages and source code on the same computer.  It is an easy error to fix though.  If the **/etc** version of the rule located at `/etc/udev/rules,d/` is deleted then the error message should disappear.  

The /lib version of the rule should not be deleted.

*************

The **Auto Gain Limit** error typically occurs if an incorrect firmware driver version for a particular librealsense version is installed in the camera.  For example, librealsense versions 2.55 and 2.56 are recommended to use firmware **5.16.0.1**.  If an older firmware such as 5.15.1.0 is used with these librealsense versions then the Auto Gain Limit error can occur.

Which librealsense version are you using, please?

***************

I am not aware of a proven method to safely uninstall the pyrealsense2 wrapper whilst leaving realsense-viewer working.  The pyrealsense2 components being left in place should not affect realsense-viewer though.

Ischavank, 2025-04-01:
Thank you for already giving the answer for the first problem, it was indeed a quick fix!

Because I had indeed installed the librealsense in 2 ways, I think I have two versions, but one of them is 2.50 (as seen in the picture below), the other one I couldn't find. My own firmware version is 5.16.0.1, so that should be good.

[image]

MartyG-RealSense, 2025-04-01:
You are very welcome!

The **grep Firmware** command's response that you have 5.16.01 installed but the recommended firmware is installed suggests that it believes that 2.50.0 is your librealsense version since 5.13.0.50 is the recommended firmware for 2.50.0. 

What happens now when you run a Python script after the /etc udev rule has been removed?

Ischavank, 2025-04-01:
When doing the realsense-viewer, I don't get the "Multiple realsense udev-rules were found:" error anymore, and the viewer still works fine, but the other pop-up in the viewer still persists and the python scripts gives the same error: "RuntimError: No device connected".

Maybe a dumb question, is it (easily) possible to use the other firmware version? And would you recommend for me to downgrade to the recommende firmware version 5.13.0.50?

MartyG-RealSense, 2025-04-01:
Yes, it's no problem to downgrade your firmware to version 5.13.0.50 if you have the D435 camera model.  It is a relatively straightforward process if you do the downgrade in realsense-viewer.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/12599#issuecomment-1902680303

closed 2025-04-17 by MartyG-RealSense