realsenseai/librealsense #13901: Randomly losing camera connections till power cycle

eric-fulfil, 2025-03-31:
Hi,

We have a strange issue lately where the more we restart a service which connects to one of 5 D457's the more likely we "permanently" lose one or more cameras until a hard power cycle.

We have a Jetson Xavier AGX with 5 D457's connected directly over FAKRA cables.  Once we lose all the cameras (rs-fw-update -l shows no cameras), we physically power off and on the board for maybe a 50/50 chance that all camera connections are restored.  On a softer software reboot, there's a tiny chance we might get some cameras back.  The following graph shows a lot of unsuccessful soft reboots followed by successful hard reboots.  We're wondering what this started happening someone suddenly, maybe I/O pressure from competing programs?

Also noticed that once after updating Ubuntu packages we went from zero to 5 cameras, what could have "reset" this connection and is there a way that we could "hard reset" the driver to simulate cutting the power entirely so we can once again safely restart services that connect to these cameras?  Are we missing proper SIGTERM / closing of cameras on app exit that should be more graceful perhaps?  We have to figure out what makes these cameras semi-permanently disappear from the realsense driver's view.

tonynajjar, 2025-05-13:
We've been painstakingly experiencing the same issue on devices in a production environment. 

 **System**

- librealsense-2.56.3
- realsense-ros 4.56.3 (ROS Jazzy)
- Realsense firmware 5.16.0.1
- Board: [Jetson Orin-NX from ConnectTech](https://connecttech.com/product/rudi-nx-embedded-system-with-nvidia-jetson-orin-nx/)
- 2 D457 realsense connected with Fakra Cables

**What happens**

With no clear pattern, we lose the camera streams and the cameras stop being detected by `rs-enumerate-devices`.

The realsense-ros node reports:

```
[realsense2_camera_node-4]  13/05 11:15:48,499 ERROR [281472040296896] (global_timestamp_reader.cpp:253) Error during time_diff_keeper polling: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[realsense2_camera_node-4]  13/05 11:15:49,771 ERROR [281472040296896] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
```

I believe the issue is lower-level than the ROS node because the syslog reports:

```
kernel: [  563.769457] d4xx 48-001b: ds5_read(): i2c read failed -121, 0x030c
kernel: [  563.779996] d4xx 48-001b: ds5_dfu_device_release(): no communication with d4xx
kernel: [  563.802894] d4xx 48-001b: ds5_read(): i2c read failed -121, 0x030c
kernel: [  563.836363] d4xx 48-001b: ds5_read(): i2c read failed -121, 0x030c
kernel: [  563.869735] d4xx 48-001b: ds5_read(): i2c read failed -121, 0x030c
kernel: [  563.880270] d4xx 48-001b: ds5_dfu_device_release(): no communication with d4xx
```

If you provide a private channel I can send you the full logs.

We discussed in depth with the board manufacturer ConnectTech and they believe the issue is on intel's side, perhaps in the realsense firmware.

@MartyG-RealSense I would very much appreciate a timely response as this is greatly affecting our production operations right now.

One useful finding from ConnectTech is that the issue is reproduceable by running `i2cdetect -y -r <bus>`.

tonynajjar, 2025-05-13:
Okay thanks I just did