realsenseai/librealsense #13903: Intel realsense d435i detetcted as usb 2.1 instead of usb 3.1

MELVINELIASMATHEW, 2025-04-02:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|239122073837|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|247523060860|
|**Firmware Update Id**|247523060860|
|**Dfu Device Path**||

my intel RealSense d435i is being detected as usb 2.1 instead of usb 3.1. I am using the latest version of the sdk and the firmware. I tried with another laptop and different cables but still the issue is present.

MartyG-RealSense, 2025-04-02:
Hi @MELVINELIASMATHEW  Please provide information about the questions below.

1.  Have you tried it with the official 1 meter cable supplied with the camera, or are all of your cables ones that you purchased yourself?

2.  Is the camera being plugged directly into the computer's USB port or into a USB hub?

3.  Does it make a difference if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it?

4. Is the camera detected as 3.1 if you insert the large end of the USB cable into the USB port of the computer or hub with a quick, firm motion instead of inserting it slowly?

MELVINELIASMATHEW, 2025-04-02:
No I haven't tried the official cable . All the cables I tried was the one I purchased myself.
The camera was plugged directly to the computer using the c to c cable and the output haven't made any change after turning the connector or firmly plugging in the connector to the port

MELVINELIASMATHEW, 2025-04-07:
Thankyou all for your support. Now the issue is solved as I tried the official cable that came with the camera . Now everything is working great

closed 2025-04-07 by MELVINELIASMATHEW

MartyG-RealSense, 2025-04-07:
You are very welcome.  It's great to hear that it is working now with the official cable.  Thanks very much for the update!