realsenseai/librealsense #13905: Good starting point for using realsense with gstreamer

prarthanasigedar, 2025-04-03:
Hello Team,
I am trying to use g-streamer for real-sense . I want to to get feed from real-sense, and send it to other computer on same network for visualization with as much as low latency. I am planning to use G streamer for this. 
I tried one repo but was not at all successful in running the real-sense plugin.
[](https://github.com/WKDSMRT/realsense-gstreamer)
Any guidance to this? Or is there a better method anyone can suggest for transferring live video from one computer to the other using real-sense with as low latency as possible. It's mainly for teleop across different rooms.

MartyG-RealSense, 2025-04-03:
Hi @prarthanasigedar Was the repo that you tried the plugin at the link below, please?

https://github.com/WKDSMRT/realsense-gstreamer

If you have tried that plugin, there is another one available:

https://www.aivero.com/2020/03/gstreamer-elements-realsense-open-sourced/

https://gitlab.com/aivero/legacy/public/gstreamer/gst-realsense

There is also a guide provided by Intel about running GStreamer with a RealSense camera in a Docker container.

https://amrdocs.intel.com/docs/2023.1.0/dev_guide/files/gstreamer-video-pipeline-using-realsense-plugin.html

prarthanasigedar, 2025-04-07:
thank-you very much!

MartyG-RealSense, 2025-04-10:
Sincere apologies for the delay in responding further.

Was your question for myself or @Zauzolkov  please?  Thanks!

closed 2025-04-24 by MartyG-RealSense