realsenseai/librealsense #13909: How to use hole_filling_filter offline to fill holes for the depth map

pgswag, 2025-04-04:
"I have a depth image captured a few months ago using a RealSense D435 camera. Now, I want to apply the hole_filling_filter function to this depth image offline. How can I do that? Could you provide a MATLAB program? If not, Python would also be acceptable." Thanks！

MartyG-RealSense, 2025-04-04:
Hi @pgswag  https://github.com/IntelRealSense/librealsense/issues/4400#issuecomment-511197098 has a MATLAB script that demonstrates the implementation and activation of the **hole_filling_filter** post-processing filter.

[image]

**********

And the link below has an example of doing so in Python with a bag file.

https://support.intelrealsense.com/hc/en-us/community/posts/15869493073811/comments/15929158191635

[image]

pgswag, 2025-04-04:
So, you mean that there is no way to call the hole filling function to process my depth map in PNG format anymore?

MartyG-RealSense, 2025-04-04:
Generally, if you need to perform a process on a saved image that is not supported built-in by the RealSense SDK then using OpenCV code (which is fully supported by the SDK) is a good way to do it.  The link below has a tutorial for OpenCV hole-filling.

https://learnopencv.com/filling-holes-in-an-image-using-opencv-python-c/

pgswag, 2025-04-04:
Thank you very much for your help. I will try the method you recommended. Thank you again