realsenseai/librealsense #13912: Realsense-viewer cant find the D455  on jetpack 6.2

N0ZA, 2025-04-05:
The realsense-viewer in the orin used to identify the RGB and Depth camera, but this time it didnt work. i tried checking the connections with lsusb but it seems like only the depth is being detected but not the RGB camera. rebooting the jetson did nothing either

MartyG-RealSense, 2025-04-05:
Hi @N0ZA  Have you updated your JetPack version since the last time that RGB was working in realsense-viewer, please?

If RGB is not working then you may be able to access an alternative RGB mode called 'RGB from left infrared sensor'.  This can be activated by enabling the Infrared stream in the Stereo Module of realsense-viewer and setting the Infrared stream's format as RGB8 instead of the Y8 format that Infrared normally uses.

N0ZA, 2025-04-05:
i reinstalled realsense SDK and the realsense viewer works now. Now the current issue is that my python script is unable to access through pyrealsense2 but it does work through the cv. i need to use the depth camera as well so i need to access it through the pyrealsense2. it says Realsense not found when i run it through pyrealsense

MartyG-RealSense, 2025-04-05:
Sometimes the camera can be successfully used in realsense-viewer but not be detected in pyrealsense2 even when a number of different possible solutions are attempted.

Is the camera detected if you add **sudo** in front of your Python script's launch command?  For example: `sudo python3 test.py `

N0ZA, 2025-04-05:
no

MartyG-RealSense, 2025-04-05:
Thank you.  Next, please try reloading the **udev** device handling rules.

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
```

N0ZA, 2025-04-05:
didnt work. just to let you know, i use a jetpack 6.2

MartyG-RealSense, 2025-04-05:
RealSense camera models equipped with an IMU, such as D455, can sometimes not be detected on Jetsons using JetPack 6, because JP6 removed an instruction called **hiddraw** that the IMU-equipped models depended upon.
 
However, since your camera works in realsense-viewer under JetPack 6, I suspect that the JP6 issue is not the cause of the problem in this particular case,  

For some users, downgrading their camera firmware driver to version **5.13.0.50** has helped.  The firmware update can be performed in realsense-viewer with the procedure below.

********

1.  Please visit the firmware releases page at the link below and scroll down the list of firmware versions until you reach 5.13.0.50.

https://dev.intelrealsense.com/docs/firmware-releases-d400

2.  Download the firmware **.bin** file for 5.13.0.50 to your computer.

3.  Open to the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** option from the drop-down menu.

4.  A file selection box will pop up.  Navigate to the location on your computer that you downloaded the firmware .bin file to and select the file.  The firmware downgrade to 5.13.0.50 should now automatically begin.

MartyG-RealSense, 2025-04-10:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update.  As you have resolved the issue, I will close the case.  Thanks again!

closed 2025-04-10 by MartyG-RealSense