realsenseai/librealsense #13916: Cannot re-enable capturing of streams without re-plugging or explicit hardware reset.

truschival, 2025-04-07:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux (6.8.0-57-generic #59~22.04.1-Ubuntu) |
|**Library version**|  2.55.1-0~realsense.12474 |
|**Name**|Intel RealSense D435I|
|**Serial Number**|033322070017|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|032523050994|
|**Firmware Update Id**|032523050994|
|**Dfu Device Path**||

**Problem:** cannot re-enable capturing of streams without re-plugging or explicit hardware reset. 

It started some weeks ago, with pyrealsense2. Now I took time to investigate and found it is not related to python but also happens reproducibly with binaries like `realsense-viewer`.

Steps to reproduce with `realsense-viewer`:

1. Enable RGB module (or stereo module) with toggle button -> Stream arrives
2. disable stream
3. re-enable color stream -> Error "Frame did not arrive in 5s" 

Strangely I can  do the same afterwards for the Stereo Module (or RGB-Module), but also only once. 

Opening and closing the software has no effect. Also other software such as the hdr demo (`rs-hdr`) don't work until I choose `hardware reset` from the menu or unplug and plug again.

MartyG-RealSense, 2025-04-07:
Hi @truschival  Which method did you use to install the librealsense SDK, please?

Official support for kernel 6.8 was not added to librealsense until version 2.56.2 onwards.  If librealsense is built from source code with the **libuvc backend** or **RSUSB backend** installation methods though then the kernel version does not matter because these installation methods bypass the kernel.

Do you still need to unplug-replug if you unplug the micro-sized end of the USB cable from the side of the D435i, turn the connector around the opposite way and re-insert it into the side of the camera (USB Type-C cameras are two-way insertion at the micro-sized end).

truschival, 2025-04-07:
@MartyG-RealSense - thanks for the hint. The kernel version could be the issue. The issue started after a power outage, probably the kernel was updated and became effective after this forced reboot.

I didn't install it from source but using the debian packages adding `https://librealsense.intel.com/Debian/apt-repo jammy main` to the apt sources list.

Is there a librealsense 2.56.2 Debian package available? According to the release notes 2.56.2 is only beta.
I will check building the library from source.

truschival, 2025-04-07:
As a quick investigation I tried kernel version 5.15 on Ubuntu 22.04 and had no improvement, the symptoms are the same.

MartyG-RealSense, 2025-04-07:
Thanks very much for the additional information.  Debian packages are not published to the librealsense repository for the 2.56.1 and 2.56.2 beta versions but in the **2.56.3** beta version they can be downloaded locally as an extractable zip file in the 'Assets' file list at the bottom of the 2.56.3 release page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.3

As you are using 22.04, the **Jammy** version of the zip file would be the appropriate one to download and extract.

[image]

As a hardware reset resolves the issue, you could try incorporating the reset into your pyrealsense2 script by adding the lines below before your pipeline start line.

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

truschival, 2025-04-08:
Unfortunately the issue still remains. 
* I downloaded and installed `librealsense2_jammy_x86_debians_beta.zip`
* checking the version of the packages I get with `dpkg -l \*realsense\* `: ` 2.56.3-0~realsense.14839` for the libraries and `1.3.28-0ubuntu1` for the dkms package
* The firmware of my camera is `5.16.0.1` 
* The camera is plugged in using a 1m USB 3 A->USB C cable (I have tried others but it doesn;t change)

I am no longer using my python code but use `realsense-viewer` to avoid any self-made side effects.

What I find really curious is that the first time enabling streams it works. Only at the second time the stream does not work anymore. 
This goes even for the stereo module. I can enable "Infrared" , get a stream, disable it, reenable it, it does not work. but i stil can enable the "depth" and get a stream (once) without hardware reset. I also can enable rgb once.

MartyG-RealSense, 2025-04-08:
It has the symptoms of a 'start after stop' issue, where the streams work the first time but after they are stopped and then started a second time, issues like 'No frames received' messages occur.  

An example of this is the case at https://github.com/IntelRealSense/librealsense/issues/6899 where it was thought at https://github.com/IntelRealSense/librealsense/issues/6899#issuecomment-671984909 that the cause might be an issue with the USB controllers on the user's particular computer.

Each USB port has a USB controller handling it, and there are different brands of controller that the manufacturer of a computer or a USB hub device can choose.  Some controllers work better with RealSense than others.  Sometimes when the camera is attached directly to the computer, plugging it into a USB port instead can help as the hub's ports will likely be using a different controller brand to the one on the computer's ports.

In the linked case, they apparently did not change to a USB hub but instead found that rebinding the **xhci_hcd** USB drivers worked every time.

truschival, 2025-04-09:
Many thanks for the hint!
Trying the same software on another computer didn't show the issues. So it really seems hardware (host) related.

truschival, 2025-04-10:
I have the exact same USB controller as in #6899 
`08:00.0 USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller (prog-if 30 [XHCI])` 

While this does not solve the problem, it at least explains it. I am still not sure why it hasn't appeared earlier.

MartyG-RealSense, 2025-04-10:
Thanks very much for the update!

You cannot change the USB controller that a particular computer uses.  As mentioned above, you could try adding a USB hub device to the computer, as the hub will likely use a different USB controller brand.  The **Amazon Basics** own-brand hub that Amazon sell works very well with RealSense cameras.

closed 2025-04-24 by MartyG-RealSense