realsenseai/librealsense #13917: realsense-viewer:Found 0 RealSense devices, with D455

hawrk, 2025-04-07:
i have Jetson AGX Orin Developer Kit and the OS is Ubuntu 22.04 LTS (Linux ubuntu 5.15.148-tegra), and i have installed  realsense SDK following the steps:[https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide](url)

and then use usb connect to jetson orin,  the lsusb command it shows : 
Bus 002 Device 007: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455 
Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0951:16e5 Kingston Technology HyperX Alloy Origins
Bus 001 Device 006: ID 0461:574a Primax Electronics, Ltd HP 125 USB Optical Mouse
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

it seens the device have been detected, but i use realsense-viewer, it has no anything like this:

[image]

and the log is:

[image]

 and how can i fixed this problem? 

please help!

MartyG-RealSense, 2025-04-07:
Hi @hawrk  Is the D455 camera detected if you launch realsense-viewer in sudo admin permissions mode with **sudo realsense-viewer**

hawrk, 2025-04-08:
@MartyG-RealSense  it just the same result, Did I miss anything during the install compilation, or I was missing some configuration?

My installation steps are:
step 1:   git clone  https://github.com/IntelRealSense/librealsense.git
step2 :  mkdir build && cmake && make && sudo make install,    all these are successful
step3:  ./scripts/setup_udev_rules.sh,   in /etc/udev/rules.d has 99-realsense-libusb.rules files

there is no errors in realsense-viewer in console, I'd like to see more detailed logs, but I don't know where the logs are

hawrk, 2025-04-08:
some other details:
 when i execute:
Ubuntu 20/22 (focal/jammy) with LTS kernel 5.13, 5.15
./scripts/patch-realsense-ubuntu-lts-hwe.sh

the script shows : 

[image]

Does this have an impact, or how can I fix it ?

MartyG-RealSense, 2025-04-08:
Which JetPack version is your Jetson using, please?  Is it JetPack 6?

Usually the CMake **make** instruction contains a list of configuration flags, like the build command below.

`mkdir build && cd build && cmake .. -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install`

I note that in your own build instruction, you used **mkdir build** instead of **mkdir build && cd build**, suggesting that after the build folder was created, it did not change directory into the build folder and then start building librealsense inside the build folder.

MartyG-RealSense, 2025-04-08:
`patch-realsense-ubuntu-lts-hwe.sh` is for PC computers.  Jetson boards should use the `patch-realsense-ubuntu-L4T.sh` patch script.

It is not necessary to use the patch script if you change the `-DFORCE_RSUSB_BACKEND` build flag from false to true, as an RSUSB = true build of librealsense bypasses the kernel.

hawrk, 2025-04-09:
@MartyG-RealSense thx, i try to add  cmake options  -DFORCE_RSUSB_BACKEND=true  -DCMAKE_BUILD_TYPE=release ,
and clear all the old source and rebuild, and it works

MartyG-RealSense, 2025-04-09:
Hi @hawrk  It's great to hear that you were successful.  Thanks very much for the update!

closed 2025-04-24 by MartyG-RealSense