realsenseai/librealsense #13918: Jetson Dosent detect Intel D435i camera

charan-xrc, 2025-04-07:
I've been trying to connect the D435i camera to my Jetson, but neither the SDK nor the ROS node is detecting the camera. I've tried various installation methods, but in the end, the camera still isn't being recognized by `librealsense`, however the camera can be seen as a USB device.

Device: Jetson orin 62GB developerkit
OS: Ubuntu 20.04
ROS: Humble

[image]
[image]

MartyG-RealSense, 2025-04-08:
Hi @charan-xrc  Are you using JetPack 6 on your Jetson, please?  If you are then RealSense cameras equipped with an IMU component, such as D435i, may not be detected by the SDK or the ROS wrapper.  This is because JetPAck 6 removed a function called **hiddraw** that IMU-equipped RealSense cameras depended upon.

If your problem is JetPack related then there are a couple of possible workarounds.

1.  Install the SDK using the **libuvc backend** installation method and its 'libuvc_installation.sh' build script.  If the camera is still not detected, a further step of downgrading the camera firmware driver version to **5.13.0.50** may be needed.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

Or: 

2.  Install the **RealSense MIPI driver**, which adds support for IMU on a Jetson Orin with JetPack 6 and a camera that is on a USB cable connection.  This method is complicated to perform though.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

MartyG-RealSense, 2025-04-08:
JetPack 5.1.3 should be okay.

charan-xrc, 2025-04-08:
Thank you for that, I'll try downgrading the jetpack.

MartyG-RealSense, 2025-04-17:
Hi @charan-xrc  Do you require further assistance with this case, please?  Thanks!

charan-xrc, 2025-04-17:
I tried downgrading the JP to` 5.1.3 `and both SDK and the ROS2 package works fine. The only backside is I had to step down my ROS2 version from humble to foxy. Thank you for your help.

MartyG-RealSense, 2025-04-17:
You are very welcome.  I'm pleased to hear that you were able to achieve a working setup.  Thanks very much for the update!

closed 2025-04-24 by MartyG-RealSense