realsenseai/librealsense #13920: D435f reported as D435 - Cannot differentiate cameras with/without IR pass filters

mjc6fe, 2025-04-09:
I am currently testing the RealSense cameras with IR pass filters (e.g., the D435f and the D435if) to evaluate whether they perform better than the unfiltered counterpart (D435i). To my surprise, the D435f product name is returned as `D435`. I think this could be a bug that has been overlooked. It would help applications to know exactly the characteristics of the cameras, especially if you run a system with mixed set of cameras.

### Questions
1. How can the user differentiate between a camera with and without a IR pass filter? I would expect that `RS2_CAMERA_INFO_NAME` provides the actual commercial name of the camera (e.g., Intel RealSense D435f, if you bought a D435f camera, and similarly with D435if).
2. I do not have a D435if. What is the output for this camera? I presume it will not be D435if, meaning you will not be able to know which exact variant it is.

### Test setup
Ubuntu 22.04 with librealsense v2.55.1 and v2.56.3, with FW version 5.16.0.1. Both return the same.

MartyG-RealSense, 2025-04-09:
Hi @mjc6fe  Thanks very much for your questions.

It is normal for a D435f to be detected as a D435 in applications such as realsense-viewer and rs-enumerate-devices.

[image]

The list of supported product ID (PID) numbers can be found at the link below.  There are not dedicated PIDs defined for D435f and D435if.

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/src/ds/d400/d400-private.h#L6

You can tell if a camera is a D435f or D435if by looking at a sticker on the bottom of the camera.  The model name (D435f) is there, though it is in very small text and so needs a magnifying glass or magnifying eyewear to read easily.

If the detected camera name is D435 then **RS2_CAMERA_INFO_NAME** will only be able to recognize a D435f or D435if by that name.  Likewise, only the PIDs of D435 and D435i would be recognized.

Once you have identified from the sticker on the base whether the camera is a filter-equipped model then you could attach a small sticker to it on the side to differentiate it at a glance from a non-filtered one.  In regards to differentiating them in a program, you could note the unique serial numbers of the cameras so you know which serial numbers relate to a filtered camera.

mjc6fe, 2025-04-09:
Thanks @MartyG-RealSense for the quick reply.

One remark: the problem is identifying which cameras have filters programmatically. Yes, the label will identify which camera it is -or you could see if there is a filter on top or not- but this is not an automatic detection, which is the point here.

This is exactly what I was afraid of. So, in summary, if the user wants to differentiate between D435f, D435i and D435if, the client has to write its own logic. For example, if the PID is the one of D435, then you know that it has a filter and cannot therefore be the D435i. If on top of that you need to know whether it has an IMU or not (to differentiate between D435f and D435if), then you need to check if the sensor supports an IMU (e.g., `rs2::sensor::is()` - https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1sensor.html#a38e2cbbf9dcb16c15d0339e7429b7332). Right?

MartyG-RealSense, 2025-04-24:
Hi @mjc6fe  Do you require further assistance with this case, please?  Thanks!

mjc6fe, 2025-04-24:
@MartyG-RealSense not for now. This can be closed. Thank you!

MartyG-RealSense, 2025-04-24:
Thanks very much, @mjc6fe - I'm pleased that I could help!