realsenseai/librealsense #13922: Reading frame by frame from .bag file

swishswish123, 2025-04-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   |  | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |     | 
| Platform                                 |   |
| SDK Version                            |                      | 
| Language                            |  Python                       | 
| Segment			|              | 

### Issue Description

I have a project where I have two cameras- a realsense camera and an RGB endoscope, and I record frames from both. I was initially saving each individual frame from the realsense into an individual .bag file using `save_single_frameset`. However, when I go back to reading these while post-processing, it is extremely slow and it's not really real-time as I have to restart the pipeline every time I want to read the next bag file that contains a single frame of data. 

I’m trying to enhance my program by saving all frames that are read into a single .bag file, which when I read from it, it reads in real-time as expected. However, when I read this recorded .bag file, it is not reading frame-by-frame, which means it is not in sync with the recorded video stream from an endoscope. 

When I read individual .bag files per frame, everything syncs correctly. However, when I use a single large .bag file, the RealSense playback seems to run out of sync. 

From what I understand from previous issues I have seen (#1579 #3121 #3850), I am meant to set the real time to False when initialising the reading of the .bag file:

```
self.config.enable_device_from_file(bag_file, repeat_playback=False)
self.pipeline_profile = self.pipeline.start(self.config)
self.playback = self.pipeline_profile.get_device().as_playback()
self.playback.set_real_time(False)
self.playback.pause()
self.streaming = True
```

However, this doesn't seem to work for me. I also tried `self.frames = self.pipeline.poll_for_frames()` as suggested when reading the frames with no success and also tried to pause and resume the playback when reading frames:

```
self.playback.resume()
self.frames = self.pipeline.wait_for_frames()
self.playback.pause()
```

I apologise if the issue has been resolved in the past but I'm not sure if maybe I'm not implementing the suggestions correctly? Is there a basic example script where someone managed to get the frame-by-frame reading to work?

Thanks so much for the help!

MartyG-RealSense, 2025-04-10:
Hi @swishswish123  Rather than restarting the pipeline for each bag file, a few RealSense users in the past have approached the problem of reading multiple bag files automatically by running their bag reading code in a loop. 
 https://github.com/IntelRealSense/librealsense/issues/6887 has a Python discussion about this, whilst https://github.com/IntelRealSense/librealsense/issues/2693 has a C++ one.

If you would like to continue using a single large bag file, can you test please whether performance improves if you add a **keep()** instruction immediately after the wait_for_frames() line so that frames are stored in the computer's memory. This can help to alleviate processing bottlenecks that cause lag.

```
self.frames = self.pipeline.wait_for_frames()
self.frames.keep()
```

swishswish123, 2025-04-11:
Thank you so much @MartyG-RealSense for the quick reply!

I am not too sure what you mean by reading multiple bag readings in one loop? When reading individual .bag files, each time I read a .bag file I call this method:

```
    def load_next_bag(self):
        self.current_bag_index += 1
        if self.current_bag_index >= len(self.bag_files):
            self.streaming = False
            return False 

        self.pipeline = rs.pipeline()
        self.config.enable_device_from_file(self.bag_files[self.current_bag_index], repeat_playback=False)
        self.pipeline.start(self.config)
        return True
```

However, as I reset the pipeline (doesn't work if I don't reset the pipeline) then the rate at which the frames are read is extremely slow (maybe every frame takes about 3-5 seconds to arrive which is why I really wanted to try using a single large bag file instead of saving each individual frames in a separate bag file. 

I tried with the .keep() and it doesn't seem to change the fact that the endoscope stream and realsense stream are still out of synch unfortunatey, with the endoscope running much quicker.

MartyG-RealSense, 2025-04-11:
My understanding is that reading bag files in a loop means that once the final frame of one bag is reached, the program automatically starts playing back the frames of the next bag in a set from its first frame, going through the collection of bag files sequentially in this way.

You mentioned that you are post-processing whilst reading the bag.  Because post-processing is calculated on the computer's CPU chip and not on the camera hardware, the filters can be expensive in terms of the processing burden that they impose on the CPU.  If you are applying filters to the data, does performance improve if you comment out the **.process** lines of the filters in your script so that the filters are not applied?

If you need to use the filters, and one of them is a Spatial filter, then I recommend commenting it out to disable it as that filter can have the largest processing burden for only a small amount of benefit.

If it is not the post-processing that is causing lag when using a single large bag then the size of the file could also influence lag if loading it into the computer's memory significantly reduces the computer's available memory capacity due to the large size of bag files.

swishswish123, 2025-04-11:
ah wait I think I may not have explained myself well, I see how you're understanding from my previous message that using a single .bag file is too slow. 

Actually it's not slow at all! Just out of sync. I'll try and explain the two scenarios a little better:

## 1) Saving and reading single framesets:

**Saving**
I save these using the saver method:
```
self.saver = rs.save_single_frameset(f'{self.save_pth}/') 
self.saver.process(self.frames)
```
Then right after that I save the endoscopic frame into a video writer. 

So I have X .bag files, each containing a single frame. 

**Reading**
For reading the frames, I load their paths in the _ _ init_ _:
```            
self.bag_files = load_bag_files_ordered(self.record_file)
```

Then when I loop through each frame by index, I load each endoscope frame from the .mp4 player using opencv's cap.read() function, and right after that I load the next bag file as 
```
    def load_next_bag(self):
        self.current_bag_index += 1
        if self.current_bag_index >= len(self.bag_files):
            self.streaming = False
            return False 

        self.pipeline = rs.pipeline()
        self.config.enable_device_from_file(self.bag_files[self.current_bag_index], repeat_playback=False)
        self.pipeline.start(self.config)
        return True
```

and read that frame. I haven't introduced any post-processing yet, so all the filters you see in the _ _ init _ _ of the class aren't currently being used as I commented them out:
```
        if len(self.record_file)>0:
            if not self.single_bag:
                if not self.load_next_bag():

        self.frames = self.pipeline.wait_for_frames()
        self.frames.keep()
        color_frame = self.frames.get_color_frame()
        depth_frame = self.frames.get_depth_frame()
        .......
```

This approach, when reading the frames they are perfectly in sync, but they are very slow. See below video:

https://github.com/user-attachments/assets/b9560e7a-f5e6-4e67-bc79-e9d33eeab0dd

## 1) Saving and reading ALL frames to a single .bag file:

**Saving**
In this approach, I enable reading to a .bag file in the _ _ init _ _ method of the Realsense camera:
```
            if len(save_pth) > 0 and single_bag:
                self.config.enable_record_to_file(f'{self.save_pth}/full_stream.bag')
```

So then I assume that every time the .wait_for_frames() is called, it automatically saves the frame to the .bag file. Therefore, what I do is I read the realsense frame, and if that is successful I then also save the endoscope frame in the same way that I did before to an mp4 video writer. 

**Reading**
To read these, I initialise the stream with:
```
        self.config.enable_device_from_file(bag_file, repeat_playback=False)
        self.pipeline_profile = self.pipeline.start(self.config)
        self.playback = self.pipeline_profile.get_device().as_playback()
        self.playback.set_real_time(False)
        self.streaming = True
```
where `bag_file` is the path to the full_stream.bag file.

Then the reading is the same:
```
        self.frames = self.pipeline.wait_for_frames()
        self.frames.keep()
        color_frame = self.frames.get_color_frame()
        depth_frame = self.frames.get_depth_frame()
        .......
```

The problem with this is the opposite as the previous approach. The actual timing is great, it seems like the same speed as it was taken, but the endoscope runs much quicker as it likely has less frames saved. That is- even though I set both the video writer and realsense to save at 30fps, it is more likely half that speed due to all the saving to files etc. The thing is that then when I read the files, the endoscope stream reads frame by frame whereas the realsense seems like it is I guess reading the frames so that it seems like they were read at 30fps. 

I'm not sure I explained this very well but hopefully with the video it makes more sense:

https://github.com/user-attachments/assets/fd3c2451-040f-4d55-8137-9f2e707dcd84

MartyG-RealSense, 2025-04-11:
wait_for_frames() blocks processing of frames until a complete frame is received. poll_for_frames() processes frames immediately without any blocking, as described at https://github.com/IntelRealSense/librealsense/issues/2422

Would RealSense and the endoscope be more equal if you used 60 FPS at 848x480 for depth nstead of 30?

It is possible to use camera metadata to sync RealSense depth data with RGB from a non-RealSense camera, as discussed at https://github.com/IntelRealSense/librealsense/issues/2186

Another approach is to use hardware sync, connected by a trigger pulse signal, and make the non-RealSense RGB camera master over the RealSense camera.

swishswish123, 2025-04-11:
I tried also with poll_for_frames which still doesn't work unfortunately. 

Also very strange but it doesn't seem to let me set depth to 848x480, and gives me:

```
Failed to resolve the request: 
        Format: Z16, width: 848, height: 480
        Format: RGB8, width: 848, height: 480

Into:
        Formats: 
         YUYV
         Z16
```

On a separate note- is there a way to check how many frames are in the .bag file? I wanted to check if I am reading the same amount of frames as the endoscope stream in the first place.

MartyG-RealSense, 2025-04-11:
The easiest way to check the number of frames is to drag and drop a bag file into the central panel of the realsense-viewer tool to initiate automatic playback.  The frame count is at the top of the panel, and you can move the timeline slider to the end of the file to see the final frame count.

I recall now that you can only play the bag back at the resolution that it was originally recorded at.

swishswish123, 2025-04-14:
I am not sure what part says the frame count, I only see the timestamp I believe?

MartyG-RealSense, 2025-04-14:
When you click the 'i' info button o the top of the stream panel - not the Info button in the side-panel or the 'Show frame metadata' button - the frame number is shown beside the word 'Frame'.

[image]

swishswish123, 2025-04-17:
ah okay found it! 
And I see the issue now- it seems like there are many more frames being saved with this method of using just one large .bag file (I assume actually 30fps). 

Is there a way to control when a frame is saved? I initially thought that it would only save when I call the wait_for_frames() function but I guess it just saves at 30fps no matter what.

swishswish123, 2025-04-17:
ah right but I can't save a frame on command then, so I can't save a frame right when I have just saved an endoscopic image frame? Just like I do when I save a frameset?

MartyG-RealSense, 2025-04-17:
The **rs-save-to-disk** C++ example program, which captures a PNG image upon a keypress, might suit your needs.  If you installed the tools and examples when you installed librealsense then there should be a pre-built executable version of rs-save-to-disk on your computer.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

If you would prefer to continue using Python scripting, https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225 has a Python script that saves the RGB color image to a PNG file when run.

swishswish123, 2025-04-17:
ah okay, unfortunately I pretty much have to use Python scripts as it's part of a wider project. In that case, you'd recommend to save an RGB image for the color and a np array of depth as in the comment?

The reason I wanted to use the .bag file is because I later on use other methods with pyrealsense2 such as generating a point cloud, which if I understand correctly can only be done with the frame class type and not just the RGB/depth, right?

MartyG-RealSense, 2025-04-17:
If you need to generate a pointcloud in pyrealsense2 with an offline file then it will be necessary to use a bag file, yes.

Given your preference for using a single large bag file, setting set_real_time to **true** and using a **playback.seek** operation to find a particular frame in the bag may suit your needs. https://github.com/IntelRealSense/librealsense/issues/8601#issuecomment-802899662 has information about doing this in Python.

swishswish123, 2025-04-18:
Thanks for the suggestion! This is probably what I would need if I can get it to work...

I have added the following in my read method:

```
                timestamp_to_seek = self.timestamps[self.frame_count]
                self.playback.seek(timestamp_to_seek)
                self.frame_count += 1

                self.playback.resume()  # resume playback
                self.frames = self.pipeline.wait_for_frames() # wait for frames
                self.playback.pause() # pause playback
```

and I don't get any errors or anything, but it didn't seem to be making a difference. 
I checked for example one of the frames and I get the following:

```
>>self.frames
<pyrealsense2.frameset RGB8 Z16 #2 @1744935873778.220215>
>>>timestamp_to_seek
1744935874011.7004
```
Meaning that the timestamp of the stream (at 1744935873778.220215) does not match the timestamp I'm seeking (1744935874011.7004).

Am I using this wrong? I saw one of the comments said to go to the beginning then seek to the frame you want but not sure what they mean- do they mean essentially reset the whole pipeline? Because that would again defeat the purpose as it would be too slow.

MartyG-RealSense, 2025-04-18:
I wonder if once you have set the frame that you want to seek and resumed playback, there is a time period inbetween resumption and pause when the frame count has moved on a bit.  You could think of it like pressing the Play button on a music player and then immediately pressing pause.  No matter how fast your finger is and how firmly the pause button is pressed, the playback is likely to have moved on a small amount before the pause takes effect.

What happens if you pause before you seek?

```
timestamp_to_seek = self.timestamps[self.frame_count]

self.playback.pause() # pause playback

self.playback.seek(timestamp_to_seek)

self.playback.resume()  # resume playback
self.frames = self.pipeline.wait_for_frames() # wait for frames
```

swishswish123, 2025-04-18:
So I changed it to what you said and it still gives the wrong timestamp but interestingly another thing is that it just seems to be freezing after about 3 frames on the `.pause()`... very strange!

swishswish123, 2025-04-18:
Actually.. I found a different solution!

In my solution I am still using individual .bag files for each frame but when reading them, instead of re-starting the pipeline with `rs.pipeline()` every time, I just stop the pipeline with .stop() and seems like I don't have to reset it. 

For future reference, my final code for reading from bag files is:

Inside main read method:
``` 
def read_frame(self)
           .......
            # loading from individual .bag files- loading next bag file to read
            if self.read_mode and not self.single_bag:
                if not self.load_next_bag():
                    print("No more bag files to play.")
                    return (False, *(None,) * (5 if self.stereo else 2))

            self.frames = self.pipeline.wait_for_frames()
            self.frames.keep()
            .......
```

where the load_next_bag() method is:
```
    def load_next_bag(self):
        if self.current_bag_idx >= len(self.bag_files):
            self.streaming = False
            return False  # No more bags
        self.current_bag_idx += 1
        self.pipeline.stop()
        self.config.enable_device_from_file(self.bag_files[self.current_bag_idx], repeat_playback=False)
        self.pipeline.start(self.config)
        return True
```

MartyG-RealSense, 2025-04-18:
You are very welcome, @swishswish123  - I'm very pleased that you achieved a solution.  Thanks so much too for sharing your method!

closed 2025-04-24 by MartyG-RealSense