realsenseai/librealsense #13929: realsense-viewer:Found 0 RealSense devices, with L515 on Jetson Orin AGX

orgicus, 2025-04-12:
Hi,

I've attempted to use a L515 with a Jetson Orin AGX and unfortunately 0 devices are found.

Context:
- I've use apt to install librealsense
- lsusb lists the device correctly and guvcview displays the IR stream just fine
- realsense-viewer displays 0 devices found

I've also tried building from source with both UVC and `FORCE_RSUSB_BACKEND` options however I'm seeing the same error.

It's worth noting ./scripts/patch-realsense-ubuntu-L4T.sh displays an error regarding jetpack version:

```bash
Jetson Board (proc/device-tree/model): Jetson AGX Orin Developer Kit
Jetson L4T version: 35.4.1
Unsupported JetPack revision 35.4.1 aborting script
```

full jetson details:

```bash
Software part of jetson-stats 4.3.0 - (c) 2024, Raffaello Bonghi
Model: Jetson AGX Orin Developer Kit - Jetpack 5.1.2 [L4T 35.4.1]
NV Power Mode[3]: MODE_50W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: p3701-0005
 - Module: NVIDIA Jetson AGX Orin (64GB ram)
Platform:
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.120-tegra
jtop:
 - Version: 4.3.0
 - Service: Active
Libraries:
 - CUDA: 11.4.315
 - cuDNN: 8.6.0.166
 - TensorRT: 8.5.2.2
 - VPI: 2.3.9
 - Vulkan: 1.3.204
 - OpenCV: 4.5.4 - with CUDA: NO
```

I'm also seeing this notification in realsense-viewer:
```bash
Timestamp: 1744413944448.410400
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2:/lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!
```
I'm attaching relevant images:
[image]
[image]
[image]
[image]

How can I use the L515 with Jetson Orin AGX running Jetson L4T version: 35.4.1 ?

Thank you for your time,
George

MartyG-RealSense, 2025-04-12:
Hi @orgicus  The RealSense SDK and RealSense Viewer tool no longer support L515 unfortunately as it is a retired camera model.  If you use version **2.50.0**, the final version that is validated as supporting L515, then your camera should be detected.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

orgicus, 2025-04-12:
That worked like charm!

Thank you so much @MartyG-RealSense !

closed 2025-04-12 by orgicus

MartyG-RealSense, 2025-04-13:
You are very welcome, @orgicus - I'm pleased that I could help.  Thanks very much for the update!