realsenseai/librealsense #13930: Incorrect Z-axis Pose Estimation for ArUco Markers When Switching Resolutions/Streams with Intel D405

lzy-zj, 2025-04-12:
When using D405 camera to capture the same ArUco marker at different resolutions (e.g., 1280x720 vs. 640x480), the estimated pose from cv2.aruco.estimatePoseSingleMarkers shows inconsistent results:

The X and Y components of the translation vector (tvec) are nearly identical.
The Z component (depth) varies significantly across resolutions/streams.
1280x720 is right

[image]

[image]

MartyG-RealSense, 2025-04-12:
Hi @lzy-zj  Depth accuracy reduces as the resolution is reduced.

lzy-zj, 2025-04-14:
> Hi [@lzy-zj](https://github.com/lzy-zj) Depth accuracy reduces as the resolution is reduced.

I didn't use depth frames, only color frames. So should I avoid using this resolution(640*480) in the future?

MartyG-RealSense, 2025-04-14:
As you have a D405, a mode called 'RGB from left IR sensor' is supported, where you can obtain an RGB image from the left infrared sensor.  You can call this mode in your script by changing the color cfg line to the one below to see whether it improves your results and allows 1280x720 resolution to be used.

`config.enable_stream(rs.stream.infrared, width, height, rs.format.bgr8, 30)`

On D405 the usual RGB color image is also provided by the left IR sensor (as this camera model does not have a dedicated RGB sensor), with a key difference that it is passed through an Image Signal Processor (ISP) chip inside the camera to generate the final RGB image.  The 'RGB from Left IR' RGB image will not be ISP-adjusted.

There are some differences in how the image is colored between the RGB and 'RGB from Left IR' images, but the Left IR image might be sufficient for your needs if it can work in your script at 1280x720 or 640x480.

closed 2025-05-01 by MartyG-RealSense