realsenseai/librealsense #13931: Says 'aMalloc failed status' everywhere else but works in pyrealsense

RikisuT, 2025-04-12:
Device : Jetson Nano 4gb reComputer j101 board
Camera : Realsense d435i

Error when I run realsense viewer and realsense ros
Realsense viewer (only when rgb camera is run):
``` 15:29:46 [Error] /tmp/sandbox/src/proc/synthetic-stream.cpp:58 - Exception was thrown during callback: aMalloc failed status: ```
Realsense ros (even if rgb is disabled) like this:
``` bash
 ros2 launch realsense2_camera rs_camera.launch.py   enable_rgb:=false   enable_depth:=false   enable_infra1:=true   enable_infra2:=true   infra_module.profile:=640x480x15   enable_gyro:=false   enable_accel:=false 
``` 
``` [realsense2_camera_node-1]  08/04 15:21:40,206 ERROR [547243907488] (synthetic-stream.cpp:58) Exception was thrown during callback: aMalloc failed status: ```

Interestingly when I open it directly using pyrealsense rgb camera runs ok:
```python
import pyrealsense2 as rs
import numpy as np
import cv2

# Initialize pipeline
pipeline = rs.pipeline()

# Configure streams
config = rs.config()
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:
        # Wait for a coherent frame
        frames = pipeline.wait_for_frames()

        # Get color frame
        color_frame = frames.get_color_frame()

        # Convert color frame to numpy array
        color_image = np.asanyarray(color_frame.get_data())

        # Show the image using OpenCV
        cv2.imshow('RealSense RGB', color_image)

        # Press 'q' to quit
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
finally:
    # Stop streaming
    pipeline.stop()

    # Close OpenCV window
    cv2.destroyAllWindows()

```
Thank you

MartyG-RealSense, 2025-04-12:
Hi @RikisuT  Which RealSense Viewer and RealSense ROS2 wrapper versions are you using, please?

RikisuT, 2025-04-12:
Device Name: Intel RealSense D435I
Device FW version: 5.16.0.1
Ubuntu 20.04
ROS2 Foxy

MartyG-RealSense, 2025-04-12:
Thank you very much for the information.

Does RGB still fail in the Viewer if you set it to BGR8 format like you do in your Python script?

The best ROS wrapper version match for your use of Ubuntu 20.04 and Foxy would be 4.51.1.  This is because support for Foxy was removed in wrapper 4.54.1 onwards.

If you need a newer wrapper version than 4.51.1 that supports Foxy, the link below has a "forked" version of the 4.54.1 wrapper that a RealSense ROS user created shortly before Foxy support was removed, so it should work for Foxy if you build it from source code.

https://github.com/MichaelKerley/realsense-ros

RikisuT, 2025-04-12:
I tried running in realsense viewer even at 320 x 180 it still says the same

MartyG-RealSense, 2025-04-12:
The specifications for the J101 board say that it has a USB 2.0 port and a USB 3.0 port.  This is not ideal, as a RealSense camera will work best with USB 2.1 and USB 3.1.  USB 3.0 was only minimally supported even when the 400 Series camera range launched at the end of 2017, and the official documentation for using the camera with USB 2.0 is equally old.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

When using industrial boards with an 'exotic' hardware configuration (i.e different from a typical consumer PC), it is recommendable to build the librealsense SDK from source code using the 'libuvc backend' or the 'RSUSB backend' installation methods.  Which method did you use to install librealsense on your J101 please?

RikisuT, 2025-04-12:
I used the RSUSB backend method by modifying the JetsonHacks's buildLibrealsense.sh script

MartyG-RealSense, 2025-04-12:
Thank you.  And trying the camera in different USB ports on the J101 makes no difference?

Also, are you using the camera with the official 1 meter USB cable supplied with it or with a different cable of your own choice?

RikisuT, 2025-04-12:
Yes I'm using the 1m cable from the box.
There is only one 3.0 port and the other ports also show the same output
One more problem I noticed now is that it says motion module failure when I enable it from Realsense-Viewer

MartyG-RealSense, 2025-04-13:
Often when a Motion Module Failure occurs in the RealSense Viewer, the Motion Module works normally when enabled a second time and so is more of an annoyance than a serious problem in such cases.  Do the gyro and accel streams start normally if you attempt to enable the Motion Module a second time after clicking on the Motion Module Failure pop-up error box to close it?

I see that advice about your question was offered yesterday (12 April 2025) at the link below.

https://nevsky.vercel.app/?question=git-1744448832545&update=1744416000026

RikisuT, 2025-04-14:
Temporarily switched to using PyRealSense for accessing the camera, then publishing the data to ROS. Will investigate the original issue later.
Also the answer in the link provided looks generated. Its pasted twice 😅

MartyG-RealSense, 2025-04-14:
Thanks very much for the update.  Please do let me know if you make further progress with the issue.  Good luck!

RikisuT, 2025-06-09:
Sorry couldn't respond on time
The issue was in the power delivered.
Thank you for the help.

MartyG-RealSense, 2025-06-09:
You are very welcome.   It's no trouble at all.  I'm pleased to hear that you found the cause.  Thanks very much for the update!