realsenseai/librealsense #13932: No Frames Received D435i

anouseconder, 2025-04-13:
Camera Model:D435I
Firmware Version:5.16.0.1
Platform:jetson orin nano devekopoer kit 8g
SDK Version:v2.56.3
USB type:3.2
Issue Description:
I deployed my own YOLOv8 model on the Jetson Orin Nano and used the D435I for image recognition. After correctly installing the relevant libraries on this Nano platform, I successfully completed the model deployment and was able to use the D435I for recognition. However, when I tried to use my model again after some time, the code calling the D435I encountered the following error:
[image]
Then, after I opened the RealSense Viewer and enabled several options, this situation occurred: only the RGB Camera image was not being received.
[image]
Other relevant information and error messages are as follows:

[image]
[image]
[image]
[image]
But strangely, I can use the D435I normally and compile the same program on the Windows 11 system, so it's not a problem with the USB cable.
Also, before this problem occurred, I had installed the ROS 2 system and some related dependencies. I wonder if these installations caused the issue, since everything was working fine initially.
 I wonder how I can find out where the problem lies.

MartyG-RealSense, 2025-04-13:
Hi @anouseconder  I note that you have enabled Depth, RGB and Motion Module.  When these three streams are simultaneously enabled in the Viewer, the RGB or Depth stream can become No Frames Received for some RealSense users, like in your image. above.  

There is no solution for that issue in the Viewer other than to only enable depth + RGB or the Motion Module and depth or RGB (but not all three). The enabling of the Motion Module is the trigger for the problem, so changing the order that the streams are enabled in makes no difference.

In pyrealsense2 scripts that enable all three streams simultaneously, a workaround for the problem is to create two separate pipelines, with Depth + RGB on one pipeline and the IMU on its own on the other pipeline, as demonstrated by a script at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

Does your pyrealsense2 program use three streams simultaneously like in your Viewer image, please?

anouseconder, 2025-04-14:
Hi @MartyG-RealSense ：Even when I enable the RGB stream alone, I encounter the same error (see the figure below), and my code only uses the RGB stream and the depth stream. Before this error occurred, I was able to use all three streams normally.

[image]

MartyG-RealSense, 2025-04-14:
As you appear to be able to enable the depth stream on its own, please next try using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to reset the calibration of your camera to its factory-new values in the realsense-viewer.  Doing so can correct a problem with a corrupted 'calibration table' inside the computer that can prevent the camera from working correctly if the table corrupts.

anouseconder, 2025-04-16:
Hi @MartyG-RealSense ：I have followed your instructions, but the RGB image still cannot be received.

MartyG-RealSense, 2025-04-16:
A Jetson Orin Nano user in a case at the link below experienced some of these errors a couple of months ago.  The solution that worked for them was to reinstall librealsense using the **libuvc backend** installation method.

https://support.intelrealsense.com/hc/en-us/community/posts/37949799858195/comments/38485838652307

This installation method is described at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

It provides a pre-made installation script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

When using the script with a Jetson, I recommend editing line 46 to replace the CMake build instruction there with the one below.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release  -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on`

Once librealsense and the realsense-viewer has been installed and is working correctly, you could subsequently reinstall the ROS2 wrapper by building it from source code with the **colcon build** procedure.

anouseconder, 2025-04-20:
Hi @MartyG-RealSense ：I followed your method to reinstall librealsense, but now not only can I not receive RGB images, but the motion module also cannot work properly.

[image]

After enabling stereo and RGB, there is depth information, but the image reception issue persists. Moreover, the motion module now fails to display properly when activated.

[image]
[image]

MartyG-RealSense, 2025-04-20:
Do you get an RGB image if you do not enable the Motion Module?  There is a known issue where enabling Depth, RGB and Motion Module in the RealSense Viewer can cause one of the streams (most often RGB) to display No Frames Received.  

The trigger for this issue is the enabling of the Motion Module, so it does not make a difference if the order that the streams are enabled in is changed.  There is no fix in the Viewer for this issue with enabling depth, RGB and Motion Module simultaneously, except to work around it by only enabling Motion Module and one other stream (Motion + Depth, or Motion + RGB), or only enabling Motion Module on its own.

anouseconder, 2025-04-20:
Hi @MartyG-RealSense ：No, it's not working even if the motion module is not enabled.

[image]

MartyG-RealSense, 2025-04-20:
Do you have the option to wipe your Jetson Orin Nano clean and reinstall everything fresh from the beginning, including Ubuntu?  Sometimes hidden glitches will develop in a system that are removed after a complete wipe and reinstall and the problems suddenly disappear because of that clean-up.

closed 2025-05-11 by MartyG-RealSense