realsenseai/librealsense #13933: realsen camera measure box size 3d

tunqLam158, 2025-04-14:
Hello, I am doing my graduation project, I use camera d415 to measure the length, width and height of the box on the conveyor belt. I have determined the pointcloud of the box but I can't get close to the dimensions. Can everyone give me some suggestions to measure the size of the 3D box? Thank you.

MartyG-RealSense, 2025-04-14:
Hi @tunqLam158  The RealSense SDK has a pre-made box volume measuring project in the Python language called **box-dimensioner_multicam** that, respite its name, can be used with a single camera.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

It calculates a pointcloud and generates a green bounding-box around the box that is overlaid on an RGB image of the scene, along with text of the volume measurements.

[image]

MartyG-RealSense, 2025-04-24:
Hi @tunqLam158  Do you require further assistance with this case, please?  Thanks!

tunqLam158, 2025-04-24:
@MartyG-RealSense How to manually provide camera internal calibration values?

tunqLam158, 2025-04-24:
@MartyG-RealSense  How to manually provide camera internal calibration values? thankyou

MartyG-RealSense, 2025-04-24:
Only one RealSense user attempted to implement this after being advised of the possibility to do so at https://github.com/IntelRealSense/librealsense/issues/9951#issuecomment-966039933

The user documented their progress at https://github.com/IntelRealSense/librealsense/issues/10054 and https://github.com/IntelRealSense/librealsense/issues/10100

At https://github.com/IntelRealSense/librealsense/issues/4061 another user demonstrated setting custom values for the intrinsics with Python code by creating an **intrinsics object**.

closed 2025-05-11 by MartyG-RealSense