realsenseai/librealsense #13935: Arch | cannot detect D455 in realsense-viewer

bnly-xxx, 2025-04-14:
### Env:
Arch linux 6.14.1-arch1-1
librealsense 	2.55.1-1
camera  D455
I install librealsense from aur, with PKGBUILD file:
```

   4   │ _RS4XX_VER=5.16.0.1
   5   │ _SR300_VER=3.26.1.0
   6   │ _L51X_VER=1.5.8.1
   7   │ _L53X_VER=3.5.5.1
   8   │
   9   │ pkgname=librealsense
  10   │ pkgver=2.55.1
  11   │ pkgrel=1
  12   │ pkgdesc="Intel® RealSense™ SDK 2.0 is a cross-platform library for Intel® RealSense™ depth cam
       │ eras (D400 & L500 series and the SR300)."
  13   │ arch=('x86_64')
  14   │ url="https://github.com/IntelRealSense/librealsense"
  15   │ license=('Apache')
  16   │ makedepends=('cmake')
  17   │ depends=('glfw' 'glu' 'gtk3' 'libusb')
  18   │ source=("https://github.com/IntelRealSense/librealsense/archive/refs/tags/v${pkgver}.tar.gz"
  19   │     "https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-${_RS4XX_VER}.bin"
  20   │     "https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-${_SR300_VER}.bin"
  21   │     "https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-${_L51X_VER}.bin"
  22   │     "https://librealsense.intel.com/Releases/L5xx/FW/L53X_FW_Image-${_L53X_VER}.bin"
  23   │     "realsense-viewer.desktop")
  24   │ sha256sums=('54546d834ff5d8b35d9955319ad2e428f6d9ae4c61b932d1bd716ed81ad135f7'
  25   │             'a481376ac2d072de1d057fe73d74fcc00ab5da17aa63fa92c18bb8f65adf909c'
  26   │             'c4ac2144df13c3a64fca9d16c175595c903e6e45f02f0f238630a223b07c14d1'
  27   │             '87a9a91b613d9d807b2bfc424abe9cac63cad75dfc04718592c44777cb0b4452'
  28   │             'b837b2cff2b270b89eed3c0b212ab4108389a20b6e07c19dd5957918ff9ce7e0'
  29   │             '59281f91e7d471a7dde1cf7207eddd8624e05218cc4301ee52e4c453a0c8ab21')
  30   │
  31   │ prepare(){
  32   │   cd "${srcdir}/${pkgname}-${pkgver}"
  33   │   sed -i 's|, GROUP:="plugdev"||g' "config/99-realsense-libusb.rules"
  34   │   sed -i 's|, GROUP="plugdev"||g' "config/99-realsense-libusb.rules"
  35   │
  36   │   mkdir -p build/common/fw/
  37   │   cp "../D4XX_FW_Image-${_RS4XX_VER}.bin" build/common/fw/
  38   │   cp "../SR3XX_FW_Image-${_SR300_VER}.bin" build/common/fw/
  39   │   cp "../L51X_FW_Image-${_L51X_VER}.bin" build/common/fw/
  40   │   cp "../L53X_FW_Image-${_L53X_VER}.bin" build/common/fw/
  41   │ }
  42   │
  43   │ build() {
  44   │   cd "${srcdir}/${pkgname}-${pkgver}"
  45   │   mkdir -p build && cd build
  46   │   CFLAGS="${CFLAGS} -Wformat -pthread" \
  47   │   CXXFLAGS="${CXXFLAGS} -Wformat -pthread" \
  48   │   unset HOME
  49   │   cmake .. \
  50   │     -DCMAKE_INSTALL_PREFIX=/usr \
  51   │     -DCMAKE_INSTALL_LIBDIR=lib \
  52   │     -DCMAKE_INSTALL_SBINDIR=bin \
  53   │     -DCMAKE_INSTALL_SYSCONFDIR=/etc \
  54   │     -DCMAKE_BUILD_TYPE=Release \
  55   │     -DBUILD_SHARED_LIBS=on \
  56   │     -DBUILD_WITH_STATIC_CRT=off \
  57   │     -DBUILD_WITH_OPENMP=on \
  58   │     -DBUILD_EXAMPLES=true \
  59   │     -DCHECK_FOR_UPDATES=off
  60   │   make
  61   │ }
  62   │
  63   │ package() {
  64   │   cd "${srcdir}/${pkgname}-${pkgver}/build"
  65   │   DESTDIR="${pkgdir}" make install
[... 66 lines cut ...]
[166671.830399] usb 6-1: new SuperSpeed USB device number 3 using xhci_hcd
[166671.842508] usb 6-1: New USB device found, idVendor=8086, idProduct=0b5c, bcdDevice=50.f1
[166671.842522] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[166671.842530] usb 6-1: Product: Intel(R) RealSense(TM) Depth Camera 455
[166671.842535] usb 6-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455
[166671.842541] usb 6-1: SerialNumber: 147623060600
[166671.855677] uvcvideo 6-1:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[166671.855821] usb 6-1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[166671.862189] uvcvideo 6-1:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[166671.862207] usb 6-1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[166671.869757] hid-generic 0003:8086:0B5C.000A: hiddev97,hidraw4: USB HID v1.10 Device [Intel(R) RealSense(TM) Depth Camera 455  Intel(R) RealSense(TM) Depth Camera 455 ] on usb-0000:66:00.3-1/input5
[166730.969644] usb 6-1: USB disconnect, device number 3
[166736.904603] pcieport 0000:00:08.3: PME: Spurious native interrupt!
[166737.530548] usb 6-1: new SuperSpeed USB device number 4 using xhci_hcd
[166737.542090] usb 6-1: New USB device found, idVendor=8086, idProduct=0b5c, bcdDevice=50.f1
[166737.542117] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[166737.542125] usb 6-1: Product: Intel(R) RealSense(TM) Depth Camera 455
[166737.542132] usb 6-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455
[166737.542138] usb 6-1: SerialNumber: 147623060600
[166737.556569] uvcvideo 6-1:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[166737.556714] usb 6-1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[166737.563190] uvcvideo 6-1:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[166737.563208] usb 6-1: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[166737.570934] hid-generic 0003:8086:0B5C.000B: hiddev97,hidraw4: USB HID v1.10 Device [Intel(R) RealSense(TM) Depth Camera 455  Intel(R) RealSense(TM) Depth Camera 455 ] on usb-0000:66:00.3-1/input5
[166907.917184] amdgpu 0000:64:00.0: amdgpu: VM memory stats for proc (0) task (0) is non-zero when fini```

MartyG-RealSense, 2025-04-14:
Hi @bnly-xxx  I recommend first trying to launch realsense-viewer in **sudo** admin permissions mode with the command `sudo realsense-viewer` if you have not done so already, as this can get around possible problems with the udev device handling rules.

**********

Installing the librealsense SDK from aur with PKGBUILD is not a method that is usually used, so there are no past references to compare your situation to.  Instead, to install for non-Ubuntu flavors of Linux, it is recommendable to build the SDK from source code.

https://github.com/IntelRealSense/librealsense/issues/4271 has an example of a user-contributed installation guide for Arch Linux.

*********

Another method that can be used for non-Ubuntu Linux branches is the **libuvc backend** method of source code installation at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc method provides a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

Line 46 of the script would need to be edited to add the CMake build flags on lines 50-59 of your aur script above.

bnly-xxx, 2025-04-15:
Hey, @MartyG-RealSense . I successfully install **libuvc** backend by following modified script.
```
#!/bin/bash -xe

#Locally suppress stderr to avoid raising not relevant messages
exec 3>&2
exec 2> /dev/null
con_dev=$(ls /dev/video* | wc -l)
exec 2>&3

if [ $con_dev -ne 0 ];
then
	echo -e "\e[32m"
	read -p "Remove all RealSense cameras attached. Hit any key when ready"
	echo -e "\e[0m"
fi

lsb_release -a
echo "Kernel version $(uname -r)"
cd ~/
sudo rm -rf ./librealsense_build
mkdir librealsense_build && cd librealsense_build

if [ $(sudo swapon --show | wc -l) -eq 0 ];
then
	echo "No swapon - setting up 1Gb swap file"
	sudo fallocate -l 2G /swapfile
	sudo chmod 600 /swapfile
	sudo mkswap /swapfile
	sudo swapon /swapfile
	sudo swapon --show
fi

echo Installing Librealsense-required dev packages
sudo pacman -S --needed --noconfirm git cmake openssl freeglut libusb pkgconf gtk3 unzip
rm -f ./master.zip

wget https://github.com/IntelRealSense/librealsense/archive/master.zip
unzip ./master.zip -d .
cd ./librealsense-master

echo Install udev-rules
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo cp config/99-realsense-d4xx-mipi-dfu.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger 
mkdir build && cd build
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release \
-DCMAKE_INSTALL_PREFIX=/usr \
      -DCMAKE_POLICY_VERSION_MINIMUM=3.5 \
      -DCMAKE_INSTALL_LIBDIR=lib \
      -DCMAKE_INSTALL_SBINDIR=bin \
      -DCMAKE_INSTALL_SYSCONFDIR=/etc \
      -DCMAKE_BUILD_TYPE=Release \
      -DBUILD_SHARED_LIBS=on \
      -DBUILD_WITH_STATIC_CRT=off \
      -DBUILD_WITH_OPENMP=on \
      -DBUILD_EXAMPLES=true \
      -DCHECK_FOR_UPDATES=off \
	#   -DBUILD_PYTHON_BINDINGS:bool=true 
	  
make -j2
sudo make install
echo -e "\e[92m\n\e[1mLibrealsense script completed.\n\e[0m"
```
I only modified `apt-get` part and cmake option.

But when I try to install python library by enable `   -DBUILD_PYTHON_BINDINGS:bool=true `, it appear
```
[  0%] Creating directories for 'pybind11'
[  0%] Building C object third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
[  0%] No download step for 'pybind11'
[  1%] No update step for 'pybind11'
[  1%] No patch step for 'pybind11'
[  1%] Performing configure step for 'pybind11'
CMake Error at CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
make[2]: *** [CMakeFiles/pybind11.dir/build.make:93：external-projects/pybind11/src/pybind11-stamp/pybind11-configure] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1089：CMakeFiles/pybind11.dir/all] 错误 2
```
I think I could fix it by adding `    -DCMAKE_POLICY_VERSION_MINIMUM=3.5 `, But i'm a newbie on Cmake stuff.

closed 2025-04-15 by bnly-xxx

MartyG-RealSense, 2025-04-16:
I'm pleased to hear that you achieved a solution, @bnly-xxx - thanks so much for sharing the details of it!