realsenseai/librealsense #13942: unexcepted depth image in pyrealsense while realsense-viewer is ok

migra1315, 2025-04-15:
## abstract
I tried to save depth image in python by pyrealsense, but get unexcepted image like

[image]

Meanwhile in realsense-viewer, the depth is like 

[image]

## code
```
import pyrealsense2 as rs
import numpy as np
import cv2
import json
import matplotlib.pyplot as plt

class realsense_camera():
    def __init__(self):
        self.pipeline = rs.pipeline()  #定义流程pipeline
        self.config = rs.config()   #定义配置config
        self.config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 15)  #配置depth流
        self.config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 15)   #配置color流
        self.profile = self.pipeline.start(self.config)  #流程开始
        self.align_to = rs.stream.color  #与color流对齐
        self.align = rs.align(self.align_to)

    def get_aligned_images(self):
        frames = self.pipeline.wait_for_frames()  #等待获取图像帧
        aligned_frames = self.align.process(frames)  #获取对齐帧
        aligned_depth_frame = aligned_frames.get_depth_frame()  #获取对齐帧中的depth帧
        color_frame = aligned_frames.get_color_frame()   #获取对齐帧中的color帧

        ############### 相机参数的获取 #######################
        intr = color_frame.profile.as_video_stream_profile().intrinsics   #获取相机内参
        depth_intrin = aligned_depth_frame.profile.as_video_stream_profile().intrinsics  #获取深度参数（像素坐标系转相机坐标系会用到）
        camera_parameters = {'fx': intr.fx, 'fy': intr.fy,
                            'ppx': intr.ppx, 'ppy': intr.ppy,
                            'height': intr.height, 'width': intr.width,
                            'depth_scale': self.profile.get_device().first_depth_sensor().get_depth_scale()
                            }
        # 保存内参到本地
        # with open('./intrinsics.json', 'w') as fp:
        #     json.dump(camera_parameters, fp)
        #######################################################
        
        depth_image = np.asanyarray(aligned_depth_frame.get_data())  #深度图（默认16位）
        color_image = np.asanyarray(color_frame.get_data())  # RGB图
        
        #返回相机内参、深度参数、彩色图、深度图、齐帧中的depth帧
        return intr, depth_intrin, color_image, depth_image, aligned_depth_frame
    
    def get_image(self):
        intr, depth_intrin, rgb, depth, aligned_depth_frame = self.get_aligned_images() #获取对齐的图像与相机内参
        cv2.imwrite("outputs/color.png",rgb)
        cv2.imwrite("outputs/depth.png",depth)

        plt.imshow(depth)
        plt.axis('off')
        plt.savefig(
            f"outputs/depth_color.png",
            bbox_inches="tight", dpi=300, pad_inches=0.0
        )
        plt.show()

if __name__ == '__main__':
    foo = realsense_camera()
    foo.get_image()

```

migra1315, 2025-04-15:
addtion info
the camera is realsense d435i
the pc is ubuntu 22.04

migra1315, 2025-04-15:
I've found solution:
When I use pyrealsense, I just turn on the sensor and get the first frame of the image. At this point, it is unstable and requires reading a few more frames

closed 2025-04-15 by migra1315

MartyG-RealSense, 2025-04-15:
Thanks very much @migra1315  for the updates.  I'm pleased to hear that you achieved a solution!