realsenseai/librealsense #13944: C++ Rosbag: Failed to create ros reader: Bag unindexed

BasTossings, 2025-04-16:
Creating a new issue because @RealSenseCustomerSupport keeps closing previous issues addressing the same problem without actually fixing this bug.

Previous issues that have **NOT** actually been resolved:

- #3044
- #3029
- #4762
- #2835

There is currently **NO** way to use `librealsense` to save a **WORKING** recording to a BAG file without either forcefully terminating the process in which the recording is happening in a very specific way (which is not acceptable in many cases, including mine) or repairing the files manually afterwards. Both of these 'workarounds' are **NOT** acceptable solutions to this problem.

BAG files created using `librealsense`'s `rs2_config_enable_record_to_file` cause a `C++ Rosbag: Failed to create ros reader: Bag unindexed` error when loading them using `librealsense`.

It seems the recording does not stop synchronously when `rs2_pipeline_stop` is called. Even when calling `rs2_delete_pipeline` and `rs2_delete_config` afterwards, which causes it to produce a corrupt BAG file.

@RealSenseCustomerSupport please provide a fix for this bug.

MartyG-RealSense, 2025-04-16:
Hi @BasTossings  A bag file recording should become indexed when the pipeline is properly closed.  An improper closure (such as terminating the program) could cause the recording to be unindexed and therefore unable to be played back.

Does this problem occur if you run the librealsense C++ bag recording and playback example program **rs-record-playback**

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp

You should be able to launch a pre-made executable version of this program if you installed the RealSense tools and example programs when installing librealsense.  On Linux, if the examples are installed, it should be in the **usr/local/bin** directory.

Bag files can also be recorded and played back in the **realsense-viewer** tool as an alternative to using program code.

BasTossings, 2025-04-16:
> Hi [@BasTossings](https://github.com/BasTossings) A bag file recording should become indexed when the pipeline is properly closed

The pipeline is properly closed using `rs2_pipeline_stop` and then `rs2_delete_pipeline`.

I have dissected [rs-record-playback.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp) and something very weird is going on with shared pointers in relation to the `record_device` class.

When the recording is stopped, line 139 of that file does the following:

``` c++
device = pipe->get_active_profile().get_device();
```

this causes the shared pointer to release the last reference to the original value of `device`, causing `rs2_delete_device` to be called, who in turn does the following:

``` c++
void rs2_delete_device(rs2_device* device) BEGIN_API_CALL
{
    VALIDATE_NOT_NULL(device);
    delete device;
}
```

This triggers the destructor of the `record_device` class, where the recording is properly terminated:

``` c++
librealsense::record_device::~record_device()
{
    for (auto&& s : m_sensors)
    {
        s->disable_recording();
    }
    if ((*m_write_thread)->flush() == false)
    {
        LOG_ERROR("Error - timeout waiting for flush, possible deadlock detected");
    }
    (*m_write_thread)->stop();
    //Just in case someone still holds a reference to the sensors,
    // we make sure that they will not try to record anything
    m_sensors.clear();
}
```

Now, when I recreate the code in `rs-record-playback.cpp` by straight up calling the native library methods from c#, emulating the shared pointer behavior:

``` c#
var PrevDevice = Device;
Device = NativeMethods.rs2_pipeline_profile_get_device ( Profile, out error );
NativeMethods.rs2_delete_device ( PrevDevice );
```

the `rs2_delete_device` still does the `delete device;`, **however**, this time it does _not_ trigger the `record_device` destructor and the recording is not properly stopped, in term leading to the corrupted BAG file.

BasTossings, 2025-04-17:
The `rs2_device` pointer looks okay from what I can see:

[image]

MartyG-RealSense, 2025-04-17:
If you are able to make use of C# then there is C# bag recording scripting at https://github.com/IntelRealSense/librealsense/issues/3142 that you could test if you have not seen this discussion already.

https://github.com/IntelRealSense/librealsense/issues/3537 has another example of C# bag recording code provided by a RealSense user, whilst the link below has the SDK's official C# bag recording example in its 'C# Cookbook' reference.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#recording

BasTossings, 2025-04-17:
@MartyG-RealSense ok, I believe I have it working now. The following c++ code works:

``` c++
void realSenseRecorder()
	{
	std::cout << "Starting recording\n";
	const std::string recordingFile = "C:\\Data\\Temp\\realsense\\a.bag";

	rs2_error* error;
	auto ApiVersion = rs2_get_api_version(&error);
	auto Context = rs2_create_context(ApiVersion, &error);

	auto Pipe = rs2_create_pipeline(Context, &error);
	auto Config = rs2_create_config(&error);
	rs2_config_enable_record_to_file(Config, recordingFile.c_str (), &error);

	auto Profile = rs2_pipeline_start_with_config(Pipe, Config, &error);

	// Recording loop

	std::cout << "Now recording\n";

	for ( int i = 0; i < 100; i++ )
	{
		auto FrameSet = rs2_pipeline_wait_for_frames(Pipe, 5000u, &error);
	
		rs2_release_frame(FrameSet);
	
		if ( i % 20 == 0 )
			std::cout << ".";
	}

	std::cout << "\n";

	// End

	std::cout << "Stopping recording...\n";
	
	rs2_pipeline_stop(Pipe, &error);
	rs2_delete_pipeline_profile(Profile);
	rs2_delete_pipeline(Pipe);
	rs2_delete_context(Context);
	rs2_delete_config(Config);

	std::cout << "Done. Press any key to exit...\n";
	getch();
	}
```

The key to this working is:

- Saving the Profile pointer that is returned from rs2_pipeline_start_with_config and deleting it when done recording.
- Deleting the Config pointer that is returned from rs2_create_config when done recording.

The two points above are not very intuitive and the documentation does not mention this as far as I can tell ([rs2_pipeline_start_with_config](https://docs.ros.org/en/ros2_packages/humble/api/librealsense2/generated/function_rs__pipeline_8h_1a1ca97b5a3d907689738cedf89fa08dad.html) and [rs2_create_config](https://docs.ros.org/en/ros2_packages/humble/api/librealsense2/generated/function_rs__config_8h_1acb431269721b211e94266119d156f9b8.html))

BasTossings, 2025-04-17:
> That's great news, [@BasTossings](https://github.com/BasTossings) - thanks so much for sharing and explaining your C++ solution!

Happy to help. The sample code in [rs-record-playback.cpp](https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp) uses the C++ API and not the raw API calls like many other bindings appear to do. The C++ smartpointer magic automatically releases all resources, even when the results are never stored by the user. So porting that code directly to another binding implementation (like python or c#) is a recipe for at best memory/handle leaks or, at worst, hard to debug bugs like in this case.

For anyone else having trouble to get it to work using the c# wrappers, the following code should work:

``` c#
using var Context = new Context ();
using var Pipeline = new Pipeline ( Context );
using var Config = new Config ();

Config.EnableRecordToFile ( "path goes here" );
using var Profile = Pipeline.Start ( Config );

for ( int i = 0; i < 100; i++ )
	{
	using var FrameSet = Pipeline.WaitForFrames ();

	if ( i % 20 == 0 )
		Console.Write ( '.' );
	}

Pipeline.Stop ();
```

closed 2025-04-17 by BasTossings