realsenseai/librealsense #13945: D450 (K83121-101) + D4 Board (J32139-120) Stuck in Recovery Mode after Firmware Update 5.16.0.1

deepGio, 2025-04-16:
### Environment Details:

* **Camera Module:** D450 (Part Number: K83121-101)
* **Board Unit:** D4 Board (Part Number: J32139-120)
* **Firmware Version Attempted:** 5.16.0.1
* **Operating System:**Ubuntu 22.04**
* **Librealsense SDK Version:** **2.55.1**
* **Host:** **Jetson Orin Nano 8GB**

### Problem Description:

I have a setup consisting of a D450 camera module (K83121-101) mounted on a D4 board unit (J32139-120). I attempted to update the firmware to version 5.16.0.1. After the process, the device is detected by the system but only in Recovery Mode.

### Steps to Reproduce:

1.  Followed "Method 1" as described in issue **https://github.com/IntelRealSense/librealsense/issues/11621**.
    * Disconnected the D450 module (K83121-101) from the D4 board (J32139-120).
    * Connected the D4 board (J32139-120) alone to the host computer.
    * Used the firmware update tool (e.g., `realsense-viewer` or `rs-firmware-update`) to install firmware version 5.16.0.1 onto the D4 board. The update reported success.
    * Disconnected the D4 board from the host.
    * Re-attached the D450 module (K83121-101) to the D4 board.
2.  Connected the complete assembly (D4 board + D450 module) to the host computer.
3.  Launched `realsense-viewer` or ran `rs-enumerate-devices`.

### Observed Behavior:

* `realsense-viewer` detects the device but indicates it is in "Recovery Mode" (see attached screenshot).

[image]

* The output of `rs-enumerate-devices -s` confirms the recovery state:
    ```text
    Device info: 
        Name                          : 	Intel RealSense D4XX Recovery
        Firmware Version              : 	5.16.0.1
        Physical Port                 : 	2-1.3-4 
        Product Id                    : 	0ADB
        Product Line                  : 	D400
        Firmware Update Id            : 	210423021847 
    ```

### Expected Behavior:

The device should be detected correctly as a functional D450 camera, not in Recovery Mode, and allow streaming from `realsense-viewer` and other applications using the SDK.

### Additional Context / Attempts to Resolve:

* Method 1 described in issue **https://github.com/IntelRealSense/librealsense/issues/11621** did not successfully update the camera module component, leaving the device in recovery. I understand other users have reported similar issues with this method for this specific hardware combination (K83121-101 module).
* Method 2, suggested in the same issue, involves using an older module (like K83121-100) to perform the initial firmware update on the board before swapping back to the K83121-101 module. Unfortunately, I do not possess an older K83121-100 module.
* **Question:** Are there any alternative procedures or recovery steps available to correctly flash the firmware onto the K83121-101 module when paired with the D4 J32139-120 board, without requiring access to an older module?

I would appreciate any assistance or guidance on how to resolve this issue and get the camera operational. Thank you.

MartyG-RealSense, 2025-04-16:
Hi @deepGio  The rs-fw-update has a recovery mode that can be attempted.  The procedure for doing so is below:

1.  Download a firmware **.bin** file for firmware version 5.16.0.1.  It sounds as though you have likely done this step already if you attempted a firmware update in rs-fw-update.

2.  Run rs-fw-update with the -r recovery flag included in its launch command.

rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin

*********

However, it appears that the problem is likely because your Vision Processor D4 board J32139-120 is an incorrect match for the D450 module.  This is the V1 version of the D4 V4 board, which is incompatible with the D450 depth module.  D450 needs to be matched with the V3 version (to make a D455) or the V5 version (to make a D457).

[image]

The correct D4 V4 board can be purchased from Intel authorized retailers, such as Rutronik24.

**V3**
https://www.rutronik24.com/product/intel/99a2nx/14278151.html

**V5**
https://www.rutronik24.com/product/intel/99c2m4/20357915.html

deepGio, 2025-04-17:
Thank you for the quick follow up @MartyG-RealSense . We purchased the correct D4 Board (V3). Once I receive the package, I'm going to test it and come back here for a short feedback.

MartyG-RealSense, 2025-04-17:
Great, I look forward to your next report.  Good luck!

MartyG-RealSense, 2025-05-30:
That's great to hear.  As the board is working without problems, I will close this case.  Thanks very much for the update!

closed 2025-05-30 by MartyG-RealSense