realsenseai/librealsense #13946: Inconsistent feed when using Full Slave Mode (mode 3)

Jrmay1, 2025-04-16:
# Inconsistent feed when using Full Slave Mode (mode 3)

## System setup:
* Camera Model: D455 
* Librealsense version: v2.54.1
* Firmware Version: 5.15.0.2
* Resolution: 848x480
* Sync Mode: 3
* External Genlock: 25fps

## The Issue:
* Sometimes, when using an external sync at 25 fps the camera feeds (left and right depth) is outputted inconsistently. 
* Often, when I restart the cambar, this issue does go away, however not always.

Please can you advise on how to prevent this issue?

MartyG-RealSense, 2025-04-16:
Hi @Jrmay1  If you are not using RGB color then Inter Cam Sync Mode 2 (Slave mode) would be the preferable choice instead of 3 (Full Slave).

When using Depth, there is only one feed.  There will only be a left and right feed when using Infrared.

25 FPS is not supported for depth in Mode 2 and 3.  On a D455 in those modes at 848x480 resolution, the supported FPS speeds will be 6, 15, 30 or 60.

25 FPS is only supported for the Y16 Infrared stream.

********

If your external sync pulse needs to be 25 FPS then the configuration that is likely to work best for you is to set the mode to 4 and the slave camera's FPS to 60 (as in mode 4, the trigger frequency has to be approximately half the speed of the slave camera's FPS).

It is much easier to get an external sync pulse working with mode 4 than it is with mode 2 and 3, as 2 & 3 require the sync pulse to be matched very precisely with the camera's FPS frequency, whilst mode 4 is much more flexible.

Jrmay1, 2025-04-16:
Hi @MartyG-RealSense, thank you for your response!

Apologies, I said depth, but I think I meant the IR Streams. I need to get these at 25fps, (and genlocked), so would you recommend 4 for this as well?

Thank you again

MartyG-RealSense, 2025-04-16:
Hardware sync is only intended to sync depth with a trigger (and depth + RGB in mode 3).  A RealSense user at https://github.com/IntelRealSense/realsense-ros/issues/2648 had test results though that indicated to them that infrared frames were being synced too.

I discussed it with my Intel RealSense colleagues at the time and they advised that the data that the user was seeing was because infrared and depth frames are always synchronized even when hardware sync is active.  This is because the depth is derived from infrared frames and so there is no way for them to be out of sync with each other.

In regard to your question about whether mode 4 could be used with the IR streams: there has not been a past precedent for this, so it is difficult to give a definite response.  You might have to set mode 4 and enable the depth stream at 30 FPS (since depth is a requirement for hardware sync), and also enable the Infrared and Infrared 2 streams in Y16 format at 25 FPS and see if they sync to the genlocked depth frames - since infrared should perfectly sync with depth - even though the infrared frames themselves are not directly having hardware sync applied to them.

It might be better to use Y8 format for left and right infrared and set them to 30 FPS (and your trigger to 30 FPS too) to ensure that trigger, depth and IR frequencies match up perfectly.

Jrmay1, 2025-04-16:
Thank you for your response.

I just realized that I misread the code, and we were actually using mode 4.

I've been reading back through previous issues and there is mention that mode 4 is experimental/not supported, but this was from a couple of years ago. Is mode 4 supported?

Cheers,

MartyG-RealSense, 2025-04-16:
Mode 4 continues to be selectable in the SDK, though does not otherwise receive support.  So it will forever be in the state that it was a few years ago when Intel ceased updates for genlock sync.

MartyG-RealSense, 2025-04-16:
When using genlock with depth only in modes 4 and higher I would call it complete.  Mode 3 (which synced Depth + RGB streams on the same camera) did not work well and was not improved as official support for genlock ceased.

I am not aware of obvious limitations with it, though when using modes 5 and above (called 'burst' modes that generated additional frames on each frame depending upon the mode number) it became more complicated.

closed 2025-05-01 by MartyG-RealSense